Koenkk/zigbee2mqtt #979: Flashing firmware issue (debugger not recognized on pc)

OmerZfira, 2019-01-30:
Hi, I am having trouble flashing the stick and would appreciate help!

I'm using windows 10 , installed the driver from texas instruments, but the cc debugger LED stays red.
I think the problem is not with the driver, because when connecting the sniffer it appears on "Device Manager" under "Cebal".

From what I understand from TI's guide there probably is an error with the firmware, so I tried to update/install the firmware with their tool "SmartRF studio 7", but the debugger is not appearing there as well.

[On another guide](http://www.ti.com/lit/ug/swru197h/swru197h.pdf) (Section 8.3 Forced boot recovery mode) they showed how to "short circuit to initiate a fresh install/update of firmware, but my debugger doesn't have the pins from figure 7..
This is my debugger:
https://imgur.com/a/vsur9uK

Has anyone got any advice how to proceed?
Thanks and a good day.

P.S.
I also tried the installation method on linux but since the LED stayed red and there aren't any troubleshooting guides for linux it was a dead end for me.
Also I don't have anarduino for trying the alternative installation option.

Turiok, 2019-01-30:
Hello OmerZfira,

I had the same problem tonight but with a CC debugger (not the same as yours). My PCB is red :).
Windows 10 doesn't recognized and after several tests. I change my USB cable and it works. 
Another possibility,  I think your driver is different if you doesn't have the same hardware that in the documentation.
 
I hope it'll help you.

Sincerely,
Didier

OmerZfira, 2019-01-31:
Thanks, it was the USB cable!

closed 2019-01-31 by OmerZfira