Koenkk/zigbee2mqtt #980: Support for Eurotronic SPIRIT Zigbee Thermostat

DoraDeu, 2019-01-30:
Hi,

there is a very interesting zigbee thermostat: "Eurotronic SPIRIT Zigbee Thermostat".
I have it already at home and it would be great if it could be supported.
I would be happy to help with testing or additional information. Just let me know.
The deivce connects to the zigbee adapter in iobroker with the following IDs:

SPZB0001
00158d0001922e34

Thanks and best regards,
Dora

JLFN, 2019-01-31:
Hey
You need to install zigbee2mqtt and then go to the path and run ``` .bash DEBUG=* npm start. ```
Then pair the device and past the log on hastebin. and pust the link here.

JLFN, 2019-02-04:
Hey its has now bee added to dev branch.
https://github.com/Koenkk/zigbee-shepherd-converters/commit/a3b5aba53f7629f41d71425ff0ad8d8a334ba257

MuZ-homekid, 2019-02-06:
hi all ... 

quick newbie question.

can anyone explain that to me please ... 

i already updated Zigbee to 0.9.1 
also take off and plug in again the usb device.
will be great if get it solved 

[image]

JLFN, 2019-02-06:
@MuZ-homekid MuZ-homekid) Lastest version is 1.1.1 ,like i said its supported in the devbranch.

MuZ-homekid, 2019-02-07:
@JLFN  sorry for that .. 
but it´s a problem with the converter ... ;) not with the zigbee adapter himself
so what you think I'm still in the wrong place .. 

or is that area only for requests for the converter .. 
if thats true .. sorry for that again .. 

cheers paule

JLFN, 2019-02-07:
its in the dev you manual add the changes maybe.

willew, 2019-04-02:
I successfully linked the thermostat and I can see the status in HomeAssistant. Could not find how to control it from HomeAssistant.  Could you help in this.

willew, 2019-04-22:
[This](http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html) pointed me to the correct direction. The answer to the above is yes. E.g. you can set the current heating point to the desinred one by publishing the following message to the topic `zigbee2mqtt/<device_id>/set` 
```
{
  "current_heating_setpoint": 19
}
```

9600, 2019-05-10:
So just to be clear, is it not possible to simply open/close the valve on command? Or in fact perhaps actually you could achieve this by simply configuring setpoint to 99 or 0.

The reason I am asking is that the radiator will be covered and therefore I do not want to use its temperature sensor and to use one located elsewhere instead.

closed 2019-09-14 by stale[bot]

alexair777, 2019-11-03:
Is there a possibility to change the local temperature? Like an offset calibration? I can't find local_temperature_calibration in the zigbee2mqtt topic. Thank you so much! Best regards

sti0, 2019-11-17:
It's possible. See my comment:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/734#issuecomment-551956806

To set a local_temperature_calibration just use the set topic and place a number between the quotes instead.