Koenkk/zigbee2mqtt #981: How is the zigbee network

MattL0, 2019-01-31:
Simple question for the zigbeetomqtt solution: 

are all ''routers', device contributing to the zigbee network alltogether? I mean, the more you add the better the network will be ?..

or is the larger network have subsets of networks?

ptvoinfo, 2019-01-31:
@MattL0 The number of routers should be enough. Routers should cover all necessary areas with all your end-devices. More routers will just occupy slots on the coordinator (keep in mind the limit of devices).

Bruceforce, 2019-01-31:
> @MattL0 More routers will just occupy slots on the coordinator (keep in mind the limit of devices).

I thought routers would connect to other routers as well. Otherwise the network would be"full" if I just have 20 Zigbee bulbs. If this is really the case it should be documented in https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/information/zigbee_network.md

ptvoinfo, 2019-01-31:
Yes, If the router cannot connect to a coordinator directly, it will connect to another router.

closed 2019-04-08 by stale[bot]