Koenkk/zigbee2mqtt #987: sudo chown -> invalid user

ekoehler87, 2019-01-31:
Hi friendly people

just started to install zigbee2mqtt and i cant get over the first steps :D 

when i write

sudo chown -R pi:pi /opt/zigbee2mqtt

i get the result "invalid user" and because of that im not able to do the next steps.
Im sure its pretty easy - but please, can you help? 

Thanks a lot, Eric

Epuipment:

OpenHABian @ Rasperry Pi 3B
CC2531 Stick
Firmware: CC2531ZNP-Prod_20190109.hex

khzd, 2019-01-31:
Pi is your user ?

ekoehler87, 2019-01-31:
:D no - but i tried different combinations and non worked

so i tested just "openhabian" (without ":" ) and i think it worked. Sorry for that

But thats what happened after that:

root@openHABianPi:~ # sudo chown -R openhabian /opt/zigbee2mqtt
root@openHABianPi:~ # cd /opt/zigbee2mqtt
root@openHABianPi:/opt/zigbee2mqtt # npm install

> serialport@6.2.2 install /opt/zigbee2mqtt/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.15.0 runtime=node arch=arm platform=linux)
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules/serialport/build'
gyp ERR! System Linux 4.14.79-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/serialport
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/openhabian/.npm/_logs/2019-01-31T20_44_37_347Z-debug.log

khzd, 2019-01-31:
Pi:pi is user:group see unix syntaxe before going further
Cheers

closed 2019-04-08 by stale[bot]