Koenkk/zigbee2mqtt #988: Failed to configure ('undefined')

sylarevan, 2019-01-31:
Hi there,

With last version (1.1.0), I get a "failed to configure" error message at startup concerning some of my hue dimmer switches. Restarting something shows the same error for another dimmer switch which was previously successfully configured. Importantly, this does not prevent all the switches to work correctly apparently.

Here is what I get:
```
1/31/2019, 6:50:44 PM - info: Zigbee: allowing new devices to join.
1/31/2019, 6:50:44 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
1/31/2019, 6:50:47 PM - info: Succesfully configured hue-switch-sejour (0x0017880110488a0b)
1/31/2019, 6:51:34 PM - error: Failed to configure hue-switch-chambre (0x00178801021b66c2) ('undefined')
1/31/2019, 6:51:34 PM - error: Failed to configure hue-switch-leane (0x00178801020f1283) ('undefined')
1/31/2019, 6:51:34 PM - error: Failed to configure hue-switch-valentine (0x00178801021b66b0) ('undefined')
```
Not really a bug, I guess, but worth mentioning.

mikep1998, 2019-02-01:
I'm seeing a similar issue. 

Jan 31 18:29:34 raspberrypi npm[18289]:   zigbee2mqtt:info 1/31/2019, 6:29:34 PM Switching log level to 'debug'
Jan 31 18:29:34 raspberrypi npm[18289]:   zigbee2mqtt:info 1/31/2019, 6:29:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
Jan 31 18:30:28 raspberrypi npm[18289]:   zigbee2mqtt:debug 1/31/2019, 6:30:28 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
Jan 31 18:30:28 raspberrypi npm[18289]:   zigbee2mqtt:info 1/31/2019, 6:30:28 PM Zigbee: allowing new devices to join.
Jan 31 18:30:28 raspberrypi npm[18289]:   zigbee2mqtt:info 1/31/2019, 6:30:28 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'

Jan 31 18:31:20 raspberrypi npm[18289]:   zigbee2mqtt:debug 1/31/2019, 6:31:20 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_magnet.aq2","appVersion":3}}' of device 'undefined' (0x00158d000274fcef)

Jan 31 18:31:20 raspberrypi npm[18289]:   zigbee2mqtt:warn 1/31/2019, 6:31:20 PM Device with modelID 'undefined' is not supported.

Why is "modelId":"lumi.sensor_magnet.aq2"  undefine?  It is defined in devices.js

thanks

alinelena, 2019-02-10:
@mikep1998 @Koenkk what I have found that helps when dealing with undefined or error node description is to remove the device via mqtt
```
mosquitto_pub -t "zigbee2mqtt/bridge/config/remove" -m
```
wait to see the message for force removal and then repair. seems to work like  a charm

closed 2019-04-18 by stale[bot]