Koenkk/zigbee2mqtt #990: Confusion about CC2531 Firmware

GSanCap, 2019-02-01:
With the new version of Zigbee2MQTT brings new cool features (like groups) but a new CC2531 firmware installed in the sniffer its required. In the [Documentation](https://koenkk.github.io/zigbee2mqtt/getting_started/flashing_the_cc2531.html) the "Download the firmware CC2531ZNP-Prod.hex" link goes to [this page](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20190109.zip) its ok, that the last CC2531 Firmware needed for the last ZigbeeMQTT version, but if you choose the Alternative Flashing Method the [download link](https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex) goes to old version of this firmware.

Can i use the [updated](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20190109.zip) CC2531 for flashing my CC2531 donggle using Arduino based alternative method?

Once reflashed i lost my pairings?

Thanks in advance

bochelork, 2019-02-01:
Reflashed yesterday and lost all my pairings. Maybe there can be some warning, since pairing is not my favorite business.

Koenkk, 2019-02-02:
@bochelrok agree, where should we put this information? (what would be the most clear to you?)

salopette, 2019-02-02:
here

https://koenkk.github.io/zigbee2mqtt/getting_started/flashing_the_cc2531.html

Koenkk, 2019-02-02:
@kirovilya what changes need to be made to the `.hex` in order to flash it via the arduino?

kirovilya, 2019-02-02:
@Koenkk remove the second last line