Koenkk/zigbee2mqtt #992: Lights turn on after restart of zigbee2mqtt - workaround?

h4nc, 2019-02-01:
I have a xiaomi double switch that turns on my lights. 

After a restart zigbee2mqqt publishes all the states again and so my automation in homeassistant gets triggered and turns on my lights. 

Is there a way to bypass this? May turning of publishing the events at restart for certain devices?

My automation looks like this:

```
- alias: Zigbee Doppelschalter - left
  initial_state: 'on'
  trigger:
  - platform: state
    entity_id: sensor.zigbee_doppelschalter_click
  condition:
    - condition: state
      entity_id: sensor.zigbee_doppelschalter_click
      state: 'left'          
  action:
    - service: light.turn_on
      entity_id: light.licht
```

My config in zigbee looks like this:

```
  '0xXXX':
    friendly_name: Zigbee Doppelschalter
    retain: false
```

h4nc, 2019-02-01:
I don't know if this really is the same issue. As my problem is pushing the states (probably saved in state.json) at the start of zigbee2mqtt causes this. 

I also have the same issue with my ikea tradfri dimmer. If I turn of my light with alexa, the dimmer value will still be let's say "150". After a restart this will turn on the light.

This only would be a workarround but I could imagine two solutions:

- zigbee2mqtt side: don't publish the state of the device at the beginning, or clear the state before (for certain devices = push buttons)
- home assistant side: ignore zigbee2mqtt for X seconds after startup, don't know how or if thats possible. This would cause other issues, like door sensors not beeing updated.

Mind that this also could be a security issue. 
For instance if you want to trigger a garage door to open by a push button a restart of zigbee (caused by a power failure) would open the garage door.

h4nc, 2019-02-02:
I don't understand. When home assistant does not run, there is no automation to be triggered, so in this case no lights will turn on. 
Or do you ask if the "start messaged" appear in zigbee2mqtt

Koenkk, 2019-02-02:
When usinh the automation format from the wiki, does this also happen (https://koenkk.github.io/zigbee2mqtt/integration/home_assistant.html - Responding to button clicks)

h4nc, 2019-02-02:
Looks like this works. The reason is that those three values are reportet at the start up:

{"battery":100,"voltage":3035,"linkquality":42}

click is not included, so the automation is not triggered.

So restarting zigbee doesn't trigger that anymore. 

To be totaly sure I still have to test if a ha or a mosquitto restart will trigger this.
Maybe at the start up of mosquitto it fired those messages again. I don't know that. Do you?

Koenkk, 2019-02-02:
No that shouldn't be the case.

Let me know if this can be closed.

h4nc, 2019-02-02:
I edited my comment above. This will still be an issue for devices like the ikea dimmer.

Koenkk, 2019-02-02:
The `brightness` of the Tradfri dimmer is something added by zigbee2mqtt. Stateless attributes like ('click', 'action') shouldn't be reported, but stateful (like `brightness`) should be.

What is the problem that this will be triggered?

Koenkk, 2019-02-02:
@h4nc I see, can you try adding `brightness` to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/deviceReceive.js#L4?

h4nc, 2019-02-02:
So there is no issue with the dimmer, if you only use the dimmer to control a light.

If you use other ways too: voice control, ha UI
than the state of the dimmer will cause issues.

However could you explain a little more what I should do in that code?

h4nc, 2019-02-02:
I don’t know why but it looks like the retain flag was set to true, so mosquito published the information at the restart.
Do you have an explanation why the retain flag went to true?
Just asking, because I’d like to know if this could happen again.

To get rid of this I deleted my mosquito addon and reinstalled it.
Thus probably isn’t the smartest way to do that. But I knew this would delete the database. Is there another way to do this in future?

Koenkk, 2019-02-03:
@h4nc retain flag is set manually in `configuration.yaml`. A retained MQTT messages can be cleared by sending a message to the same topic without the retain flag.

h4nc, 2019-02-03:
Thanks for clearing that. To still it will be easier to delete and reinstall to clear the database, as finding the right topic and publish.

I still don’t get why my left button was retain: true.

I‘m thinking about using one of those Aqara sensors to trigger my garage door.
But before I do this I must be a 100% sure the automation is only triggered when I push the button. And not from a restart of similar things.

Koenkk, 2019-02-03:
@h4nc agree! We are trying find the correct automation in #959. For now, the MQTT trigger can be used.

closed 2019-02-03 by Koenkk