Koenkk/zigbee2mqtt #993: Update the database or restart zigbee2mqtt?

FRFrancisqueti, 2019-02-01:
Good morning friends.
By using Topic: zigbee2mqtt / bridge / config / remove and Payload: "friendly_name" I successfully remove the device:

 zigbee2mqtt:debug 2019-2-1 09:47:10 Saving state to file C:\opt\zigbee2mqtt\data\state.json
  zigbee2mqtt:debug 2019-2-1 09:47:11 Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0x000d6f000e1e407c'
  zigbee2mqtt:info 2019-2-1 09:47:11 Successfully removed 0x000d6f000e1e407c
  zigbee2mqtt:info 2019-2-1 09:47:11 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x000d6f000e1e407c"}'
  zigbee2mqtt:debug 2019-2-1 09:47:12 Saving state to file C:\opt\zigbee2mqtt\data\state.json

The database confirms that the device has been removed:
https://pastebin.com/TXj1bCB1

However after a few seconds, even with the permit_join = false the device is paired again without the device's network button action.
I noticed that if after removing the device and getting the confirmation through the log if the MQTT server is manually restarted this does not occur.

The device is a Door Sensor (HEIMAN-HS1DS-E) and I'm using MQTTnet.

Koen, what can I do to solve this automatically? Would you like to update the database without the need to restart the MQTT server? Or could I send a command to zigbee2mqtt / bridge / status to restart the server after removing the device?
Thank you.

Koenkk, 2019-02-02:
How do you know the device is paired again? Can you provide the log of that?

FRFrancisqueti, 2019-02-04:
Koen, I really need a topic and message structure to restart the server, could you please help me?

closed 2019-04-12 by stale[bot]