Koenkk/zigbee2mqtt #995: zigbee2mqtt stops, only reboot helps for a short time.

salopette, 2019-02-01:
Hello, I use the current Dev branch.
and it comes immeer again to the error.
After a reboot it runs some time and then again not more.

Here's the log:

https://paste.ubuntu.com/p/BTjZvCRfx2/

Koenkk, 2019-02-02:
Can you try with 1.1.1?

salopette, 2019-02-02:
i will test it, and report!

WagnerBeccard, 2019-02-02:
Maybe related - just on 1.1.1 and tryed to pair my 8 hue dimmer switches wich always loose the conectivity to the network. 
Published just to test `zigbee2mqtt/bridge/config/permit_join; true `and got 
```
 zigbee2mqtt:info 2/2/2019, 10:59:30 AM
     {"message":"request unsuccess: 194","stack":"Error: request unsuccess: 194\n    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:45:21\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:258:13)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:360:14)\n    at Immediate._onImmediate (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:336:22)\n    at runCallback (timers.js:705:18)\n    at tryOnImmediate (timers.js:676:5)\n    at processImmediate (timers.js:658:5)"}
  zigbee2mqtt:info 2/2/2019, 10:59:30 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'

```

afterwards getting only 

` Zigbee publish to device '0x00178801026fc152', lightingColorCtrl - moveToColorTemp - {"colortemp":343,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 194 `

Koenkk, 2019-02-02:
@WagnerBeccard can you:
- stop zigbee2mqtt
- replug stick
- press reset button (the one close to the usb stick), green led should go off
- start zigbee2mqtt.

WagnerBeccard, 2019-02-03:
Then more of the Error: AF data request fails, status code: 183. APS no ack.BIt's working.. a litte. A fiew more restarts without replug and I gain back control

salopette, 2019-02-05:
Hallol, there is a time, today may be again mistakes

https://paste.ubuntu.com/p/M55pz2CfvJ/

Koenkk, 2019-02-05:
@salopette error 16 means `ZmemError`, indicating that the CC2531 is out of memory. 

I see you are running the `20181024` firmware, I would recommend upgrading to the `20190109` firmware. The memory of this firmware has been increased.

salopette, 2019-02-05:
Strange, I only trained 12 devices.

I will use the day to flash the stick and will report!

dzungpv, 2019-02-12:
I have the same problem with 1.1.1 and edge version. 8 devices on the network, two CC2530 router

Koenkk, 2019-02-12:
@dzungpv can you post your log when this happens?

dzungpv, 2019-02-14:
@Koenkk : this is the line show error:
```
zigbee2mqtt:error 2019-2-14 16:26:58 Cannot get the Node Descriptor of the Device: 0x00124b000d66xxxx (Error: Timed out after 10000 ms)
```

Koenkk, 2019-02-14:
@dzungpv  Can you post more logging?

dzungpv, 2019-02-14:
@Koenkk may be my problem also cause by this bug https://github.com/Koenkk/zigbee2mqtt/issues/1090.
I have format the card so lost the log when this occurs

salopette, 2019-03-02:
Hello, problem still exists

2019-3-2 16:25:03 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-02.16-25-03'
2019-3-2 16:25:04 - info: Starting zigbee2mqtt version 1.1.1 (commit #92d88b6)
2019-3-2 16:25:04 - info: Starting zigbee-shepherd
2019-3-2 16:25:05 - info: zigbee-shepherd started
2019-3-2 16:25:05 - info: Coordinator firmware version: '20190109'

https://paste.ubuntu.com/p/m2kmqX6CsV/
https://paste.ubuntu.com/p/XPjrFfq3yF/

Koenkk, 2019-03-03:
@salopette Please try with the dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) and latest dev firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin)

WagnerBeccard, 2019-03-03:
Is the Max devices Firmware still a thing? 
To fix 205 No network route and/or 183 APS no ack. it sometimes (!) helps to alow joining and toggle the light manualy. saved me some reboots.

Koenkk, 2019-03-03:
Yes, the max devices firmware is still supported, once z-stack-firmware dev is merged to master these will be available.

closed 2019-05-09 by stale[bot]