Koenkk/zigbee2mqtt #1000: Can't connect my aqara single wired switch QBKG04LM

mayrp2001, 2019-02-02:
Hi,

I have wired up my Aqara single switch (QBKG04LM) and it works manually. But for some reason home assistant doesn't see it. I have tried pairing it numerous times but with no success. 

I have successfully connected various temperature and window sensors through zigbee2mqtt so I am not sure what the problem is. 

I also have the raspberry pi very close to the actual switch, so distance shouldn't be an issue.

Any thoughts would be greatly appreciated.

Cheers

mayrp2001, 2019-02-02:
Hi, 

I have included the startup log from zigbee2mqtt.

[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-02-02T16:40:45: PM2 log: Launching in no daemon mode
2019-02-02T16:40:46: PM2 log: App [npm:0] starting in -fork mode-
2019-02-02T16:40:46: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.0 start /zigbee2mqtt-1.1.0
> node index.js
  zigbee2mqtt:info 2019-2-2 16:40:49 Logging to directory: '/share/zigbee2mqtt/log/2019-02-02.16-40-49'
  zigbee2mqtt:info 2019-2-2 16:40:50 Starting zigbee2mqtt version 1.1.0 (commit #unknown)
  zigbee2mqtt:info 2019-2-2 16:40:51 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-2 16:40:51 zigbee-shepherd started
  zigbee2mqtt:info 2019-2-2 16:40:51 Coordinator firmware version: '20181024'
  zigbee2mqtt:info 2019-2-2 16:40:51 Currently 4 devices are joined:
  zigbee2mqtt:info 2019-2-2 16:40:51 Temperature_Sensor (0x00158d00023d066a): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
  zigbee2mqtt:info 2019-2-2 16:40:51 Front_Bedroom_Lamp (0x01124b001bb0000c): KS-SM001 - Ksentry Electronics [Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008837134660/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm) (Router)
  zigbee2mqtt:info 2019-2-2 16:40:51 Living_Room_Sensor (0x00158d00025eddd7): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2019-2-2 16:40:51 Master_Bedroom_Sensor (0x00158d0002c78c3a): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:warn 2019-2-2 16:40:51 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-2-2 16:40:51 Allowing new devices to join.
  zigbee2mqtt:warn 2019-2-2 16:40:51 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-2-2 16:40:51 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-2-2 16:40:51 Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 2019-2-2 16:40:52 zigbee-shepherd ready
  zigbee2mqtt:info 2019-2-2 16:40:52 Connected to MQTT server
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'zigbee2mqtt/Temperature_Sensor', payload '{"temperature":16.6,"linkquality":65,"humidity":46.63,"battery":100,"voltage":3005}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'zigbee2mqtt/Front_Bedroom_Lamp', payload '{"state":"OFF"}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'zigbee2mqtt/Living_Room_Sensor', payload '{"battery":100,"voltage":3075,"linkquality":102,"contact":true}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'zigbee2mqtt/Master_Bedroom_Sensor', payload '{"contact":true,"linkquality":94,"battery":100,"voltage":3055}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'homeassistant/sensor/0x00158d00023d066a/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/Temperature_Sensor","json_attributes_topic":"zigbee2mqtt/Temperature_Sensor","name":"Temperature_Sensor_temperature","unique_id":"0x00158d00023d066a_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023d066a","name":"Temperature_Sensor","sw_version":"Zigbee2mqtt 1.1.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'homeassistant/sensor/0x00158d00023d066a/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/Temperature_Sensor","json_attributes_topic":"zigbee2mqtt/Temperature_Sensor","name":"Temperature_Sensor_humidity","unique_id":"0x00158d00023d066a_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023d066a","name":"Temperature_Sensor","sw_version":"Zigbee2mqtt 1.1.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'homeassistant/sensor/0x00158d00023d066a/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/Temperature_Sensor","json_attributes_topic":"zigbee2mqtt/Temperature_Sensor","name":"Temperature_Sensor_battery","unique_id":"0x00158d00023d066a_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023d066a","name":"Temperature_Sensor","sw_version":"Zigbee2mqtt 1.1.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'homeassistant/switch/0x01124b001bb0000c/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/Front_Bedroom_Lamp/set","state_topic":"zigbee2mqtt/Front_Bedroom_Lamp","name":"Front_Bedroom_Lamp_switch","unique_id":"0x01124b001bb0000c_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x01124b001bb0000c","name":"Front_Bedroom_Lamp","sw_version":"Zigbee2mqtt 1.1.0","model":"[Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008 [... cut ...]
  zigbee2mqtt:info 2019-2-2 16:40:52 MQTT publish: topic 'homeassistant/sensor/0x00158d0002c78c3a/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/Master_Bedroom_Sensor","json_attributes_topic":"zigbee2mqtt/Master_Bedroom_Sensor","name":"Master_Bedroom_Sensor_battery","unique_id":"0x00158d0002c78c3a_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c78c3a","name":"Master_Bedroom_Sensor","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'

mayrp2001, 2019-02-02:
Also, when I look at my configuartion.yaml within zigbee2mqtt. A strange entity appears and I am not sure what it is or where it has come from.

https://imgur.com/qb3J54m

Koenkk, 2019-02-03:
Can you post your `database.db`? I don't see the QBKG04LM in the startup log.

mayrp2001, 2019-02-03:
Hi. I couldn't get it paired so that's why it wouldn't have been appearing in the startup log. However I did manage to finally get it paired.

I am still however seeing the strange entity in my zigbee2mqtt config yaml as you can see in the link in the above post. I am clueless as to what it is or what it is for. I deleted it from my config but it appeared again after restarting the add on. 

Any ideas?

Koenkk, 2019-02-03:
Can you lookup this entry in `database.db` and post it here?

Koenkk, 2019-02-06:
@mayrp2001 that entity is from zigbee2mqtt itself. Can you post your startup log again? As I can see the QBKG04LM in the `database.db` (`0x00158d0002ec227d`)

closed 2019-04-14 by stale[bot]