Koenkk/zigbee2mqtt #1002: Pinging Xiaomi plugs fails

milakov, 2019-02-02:
Hi, Iэму just moved to zigbee channel 20 from default 11, everything works fine so far, but I immediately started to observe that all pings of both my Xiaomi plugs fail always:
```
  zigbee2mqtt:debug 2019-2-2 21:44:00 Check online 0x00158d00024f2e59 0x00158d00024f2e59
  zigbee2mqtt:debug 2019-2-2 21:44:10 Failed to ping 0x00158d00024f2e59
  zigbee2mqtt:debug 2019-2-2 21:44:10 Check online 0x00158d0002561266 0x00158d0002561266
  zigbee2mqtt:debug 2019-2-2 21:44:20 Failed to ping 0x00158d0002561266
```
It is pinging Gledopto GL-B007Z bulb (which is router as well) successfully:
```
  zigbee2mqtt:debug 2019-2-2 21:55:48 Check online 0x00124b001b50416b 0x00124b001b50416b
  zigbee2mqtt:debug 2019-2-2 21:55:48 Successfully pinged 0x00124b001b50416b
```

But those bulbs work fine, I can switch them on and off remotely, and if I manually press the button on the plug it is immediately reflected in Home Assistant GUI.

Koenkk, 2019-02-03:
Do you have `availability_timeout` enabled?

milakov, 2019-02-03:
It doesn't work in both cases. If it is disabled I am getting only 2 xiaomi plugs pinger (failing). If I set ```availability_timeout: 60``` I will get not only xiaomi plugs pinged but also gledopto bulb, xiaomi failing, gledopto is being pinged successfully.

Koenkk, 2019-02-03:
Yes, my guess is that the Xiaomi power plugs don't respond to the ping requests. With `availability_timeout`, am I right saying that these devices are always shown as unavailable?

milakov, 2019-02-03:
Correct. But, I think, they responded when I used channel 11 (default one).

Koenkk, 2019-02-03:
@milakov that's really strange indeed, I don't there is anything we can do about this.

glentakahashi, 2019-02-04:
Nothing to add, but am seeing this for my xiaomi plugs as well. With/without availability_timeout

```
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:45 AM Ping 0x00158d000237dc52
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:45 AM Ping 0x00124b0014d9d4fa
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:45 AM Successfully pinged 0x00124b0014d9d4fa
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:46 AM Ping 0x00158d000237dc1a
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:46 AM Ping 0x00158d000237dd9b
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:46 AM Ping 0x00158d000237ddb6
zigbee2mqtt      |   zigbee2mqtt:debug 2/4/2019, 6:09:46 AM Ping 0x00158d0002699ec6
zigbee2mqtt      |   zigbee2mqtt:error 2/4/2019, 6:09:55 AM Failed to ping 0x00158d000237dc52
zigbee2mqtt      |   zigbee2mqtt:error 2/4/2019, 6:09:56 AM Failed to ping 0x00158d000237dc1a
zigbee2mqtt      |   zigbee2mqtt:error 2/4/2019, 6:09:56 AM Failed to ping 0x00158d000237dd9b
zigbee2mqtt      |   zigbee2mqtt:error 2/4/2019, 6:09:56 AM Failed to ping 0x00158d000237ddb6
zigbee2mqtt      |   zigbee2mqtt:error 2/4/2019, 6:09:56 AM Failed to ping 0x00158d0002699ec6
```

bizziebis, 2019-02-04:
They seem to have an erratic behavior. I have two Xiaomi plugs which could be pinged just fine for hours. After a while one plug always fails to get pinged. I didn't change anything to the system.

> 2019-2-4 18:18:38 - debug: Ping 0x00124b00016f2bd5
> 2019-2-4 18:18:38 - debug: Successfully pinged 0x00124b00016f2bd5
> 2019-2-4 18:18:38 - debug: Ping 0x00158d0002370efc
> 2019-2-4 18:18:38 - debug: Ping 0xbfb6775ffeffe79d
> 2019-2-4 18:18:39 - debug: Ping 0x00158d00024d8998
> 2019-2-4 18:18:39 - debug: Successfully pinged 0x00158d00024d8998
> 2019-2-4 18:18:39 - debug: Successfully pinged 0xbfb6775ffeffe79d
> 2019-2-4 18:18:48 - debug: Failed to ping 0x00158d0002370efc

The plug does report regularly

> 2019-2-4 18:00:45 - info: MQTT publish: topic 'zigbee2mqtt/Xiaomi PowerPlug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":9.17,"temperature":28,"linkquality":89,"last_seen":"2019-01-23T13:26:14.288Z"}'
> 2019-2-4 18:06:08 - info: MQTT publish: topic 'zigbee2mqtt/Xiaomi PowerPlug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":9.17,"temperature":28,"linkquality":92,"last_seen":"2019-01-23T13:26:14.288Z"}'
> 2019-2-4 18:12:05 - info: MQTT publish: topic 'zigbee2mqtt/Xiaomi PowerPlug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":9.17,"temperature":28,"linkquality":92,"last_seen":"2019-01-23T13:26:14.288Z"}'
> 2019-2-4 18:19:13 - info: MQTT publish: topic 'zigbee2mqtt/Xiaomi PowerPlug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":9.17,"temperature":28,"linkquality":92,"last_seen":"2019-01-23T13:26:14.288Z"}'
> 2019-2-4 18:26:43 - info: MQTT publish: topic 'zigbee2mqtt/Xiaomi PowerPlug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":9.17,"temperature":28,"linkquality":92,"last_seen":"2019-01-23T13:26:14.288Z"}'

This router is only connected through another router to the coordinator. Could that be a reason it's not pingable?

milakov, 2019-02-04:
@bizziebis My Xiaomi plug are connected directly to coordinator and pinging them fails.

KapJI, 2019-02-05:
I think this is related to https://github.com/Koenkk/zigbee2mqtt/commit/3963a8d26f8e79011629a34aac3549e26766bd52
Logging of errors was added there.

alinelena, 2019-02-17:
similar behaviour here... but with a cc2531 router too not only with the xiaomi plug
```
2/17/2019, 10:08:29 AM - error: Failed to ping 0x00158d0002366b72 xiaomi plug
2/17/2019, 10:08:29 AM - error: Failed to ping 0x00124b0018ed19ff cc2531  router
```

both work fine for hours, in the sense they are pingable. then the error. 
However the xiaomi plug works as expected,  the map reports a motion sensor connected to it and that works too, also the map reports it as online 

there is nothing connected to the cc router so no idea if works or not.

milakov, 2019-03-01:
It started working for me after I flashed 20190223 fw and moved to the latest dev version. Don't know if it is going to stay this way. And sniffer shows that both smart swicthes send Node Descriptor Response packets back to coordinator.

timdonovanuk, 2019-03-05:
Still an issue - my plug worked for a bit, but then stopped. On the latest fw and dev server.

alinelena, 2019-03-05:
got the bullet and flashed the latest firmware with the latest dev... big mistake. it seems to be unstable like hell. 
the ping failures showed almost immediately. pairing was a nightmare, worst experience in more than half a year of regular repairing and worst after around 10 hours all the network seems to be down, with no obvious errors, other than  unsuccessful pings.I wonder if this has to do with the size of the network.

artxxx, 2019-03-06:
same issue, i have 2 switches with 2 button and 2 switches with single button.
Sometimes one 2 button switch don't work(
Every time this is the same switch.

stratplayer, 2019-03-11:
I have same issue with the Power Plug see the log below

`[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-11T21:09:25: PM2 log: Launching in no daemon mode
2019-03-11T21:09:25: PM2 log: App [npm:0] starting in -fork mode-
2019-03-11T21:09:25: PM2 log: App [npm:0] online
> zigbee2mqtt@1.2.1 start /zigbee2mqtt-1.2.1
> node index.js
  zigbee2mqtt:info 3/11/2019, 9:09:30 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-11.21-09-30'
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM zigbee-shepherd started
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Currently 6 devices are joined:
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Smart Button (0x00158d0002134ead): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Office Motion (0x00158d000236c231): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Aqara Power Switch (0x00158d000215f0f5): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Living Room Window (0x00158d000204a0ae): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Office Window (0x00158d0002048697): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Back Door (0x00158d000204bb55): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 3/11/2019, 9:09:32 PM zigbee-shepherd ready
  zigbee2mqtt:info 3/11/2019, 9:09:33 PM Connected to MQTT server
  zigbee2mqtt:info 3/11/2019, 9:09:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 3/11/2019, 9:09:33 PM MQTT publish: topic 'zigbee2mqtt/Smart Button', payload '{"battery":100,"voltage":3015,"linkquality":102}'
  zigbee2mqtt:info 3/11/2019, 9:09:33 PM MQTT publish: topic 'zigbee2mqtt/Office Motion', payload '{"illuminance":16,"linkquality":162,"occupancy":true}'
  zigbee2mqtt:info 3/11/2019, 9:09:33 PM MQTT publish: topic 'zigbee2mqtt/Aqara Power Switch', payload '{"state":"ON","power":0,"voltage":234.5,"consumption":0,"temperature":25,"linkquality":70}'
  zigbee2mqtt:info 3/11/2019, 9:10:23 PM MQTT publish: topic 'zigbee2mqtt/Back Door', payload '{"contact":true,"linkquality":44}'
  zigbee2mqtt:error 3/11/2019, 9:10:43 PM Failed to ping 0x00158d000215f0f5`

artxxx, 2019-03-13:
I solved my issue.
When I used my power plug as Router always received "PM Failed to ping ..."
I marked power plug as "EndDevice" and add Gas sensor as Router.
With Gas sensor as router all works perfect!
It's seems that power plug don't work properly as Router.

joseska, 2019-03-18:
> 
> 
> I solved my issue.
> When I used my power plug as Router always received "PM Failed to ping ..."
> I marked power plug as "EndDevice" and add Gas sensor as Router.
> With Gas sensor as router all works perfect!
> It's seems that power plug don't work properly as Router.

same error here.......  how can I mark the device as EndDevice???

Koenkk, 2019-03-18:
when you mark it as end device, by manually editing database.db, it will just stop pinging the device, it doesnt solve the real problem.

artxxx, 2019-03-19:
> same error here....... how can I mark the device as EndDevice???

With Gas sensor as router its happened again(

I marked all my routers as EndDevice and flashed USB stick firmware with 40 devices support.
After 2 days excellent work, half of my house don't work, this devices unavailable.

Now in search of decision..

artxxx, 2019-03-27:
Flashed dev branch, works perfect

Koenkk, 2019-03-28:
Should indeed be fixed due to #1248, feel free to reopen if this is still an issue.

closed 2019-03-28 by Koenkk

omer1141, 2019-04-21:
I've installed versinon 1.3.1 over the last couple of days and the problem with the ping seems to be back. Did you maybe reintroduce the problem again?

Koenkk, 2019-04-21:
Can you provide the debug log?

omer1141, 2019-04-21:
Here you go.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3101403/log.txt)

Koenkk, 2019-04-22:
@omer1141 I'm not sure why it fails in your case, did this also happen before 1.3?

omer1141, 2019-04-22:
@Koenkk yes it failed pre-version 1.3, but with 1.3.0 is was working ok. After installing 1.3.1, the connection problem started again. Hence the question I've put out there. No change in this part of the code for v 1.3.1?

Koenkk, 2019-04-23:
@omer1141 there have been no changes regarding this, so I'm not sure where this comes from.