Koenkk/zigbee2mqtt #1003: sengled E11-G13 bulbs with zigbee2mqtt

bereska, 2019-02-02:
Did you read the FAQ?
https://koenkk.github.io/zigbee2mqtt/information/FAQ.md
yes
Unsupported device?
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
yes
A question?
Read the documentation: https://koenkk.github.io/zigbee2mqtt/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

this is a question not an issue since I could not find a zigbee2mqtt forum
I'm currently running a network of sengled E11-G13 bulbs in Homeassistant docker on rpi 3b+ with HUSBZB-1 stick
the problem is that the bulbs stop responding within hours after restart
it's a known issue with this setup which cannot be resolved for months
https://github.com/zigpy/bellows/issues/124
will I solve this problem if I switch to zigbee2mqtt?
thank you

Koenkk, 2019-02-02:
This device is supported, I dont see why it wont work.

bereska, 2019-02-02:
thank for your prompt response, it gives me a hope)
my CC2531 is on its way from China

kylezimmerman, 2019-02-06:
I use these bulbs and they are working.

One important note is that even though these bulbs have constant power, they are end devices not routers. So if these are the only devices in your network, you'll have a limit of 20 devices and they'll all be connected directly to the controller.

I ended up having to buy additional router devices in order add all of my devices, and improve network quality.

bereska, 2019-02-06:
@kylezimmerman ,thank you for your notes. I have 10 bulbs so far. How about the dongle range? I live in three-story a brick house.

roygbiv856, 2019-02-06:
@bereska are your interior walls brick? I live in a 2100 sq ft 100 year old home with 2 stories and a basement. My interior walls are standard wood studs. I've actually been surprised by the range. I've got my coordinator and 1 router on the ground floor and 1 router upstairs. I've got sengled bulbs on every floor and I've never had a problem with range.

bereska, 2019-02-06:
@kylezimmerman some of them are brick, some are wood studs. Actually my house is similar to yours except I don't have a basement. My current setup is zigbeepy/bellows HUSBZB-1 stick (controller) plugged into pi 3b+ sitting up on the third floor. Sengled lights (10) are all over the house and I am having issues with the ones on the ground floor that average -80-90 rssi. I thought I could get a better range with a CC2531 dongle, it looks like i'll need a router anyways. What kind of router do you have?

bereska, 2019-02-06:
@roygbiv856 sorry, wrong contact: some of them are brick, some are wood studs. Actually my house is similar to yours except I don't have a basement. My current setup is zigbeepy/bellows HUSBZB-1 stick (controller) plugged into pi 3b+ sitting up on the third floor. Sengled lights (10) are all over the house and I am having issues with the ones on the ground floor that average -80-90 rssi. I thought I could get a better range with a CC2531 dongle, it looks like i'll need a router anyways. What kind of router do you have?

kylezimmerman, 2019-02-06:
I live in a 2 bedroom apartment so range hasn't been too much of an issue for me, it's more been about network size (>30 devices).

For router devices I'm currently using a few Ikea Tradfri bulbs and a couple of Ikea Tradfri Wall Outlets. I tried to pick devices that were routers but also useful on their own.

Most Zigbee devices that are constantly powered are routers (Hue bulbs, etc.), these Sengled bulbs seem to be the exception. I got them because of the great price relative to the others, but I guess sometimes you get what you pay for 😄.

roygbiv856, 2019-02-06:
@bereska i use 2 2531s as routers. Cheap and effective. Im not sure how you're incorporating the HUSBZB with z2m? I didnt even know that was possible

@kylezimmerman i know that adding more routers increases the device limit, but is there a ceiling you eventually hit where additional routers no longer increase the device limit? Also, have you noticed devices responding slower (motion sensor for example) with that many devices in your network?

roygbiv856, 2019-02-06:
@kylezimmerman motion activated lights for every room is currently my ultimate goal. xiaomi sensors work so well and are so cheap that i'm afraid i'll hit the device limit sooner rather later. I know that koenkk has described the limit in other discussions, gotta go dig up that info.

bereska, 2019-02-06:
@kylezimmerman thank you, so if I buy a couple of Hue bulbs and and set them up as routers on every floor I can cover all the Sengled bulbs in the whole house, right?
@roygbiv856 thank you, I ordered only one 2531 to be the gateway for zigbee2mqtt network and it takes weeks to have them shipped from China(. My current zha setup is HUSBZB-1 stick that just works with HA out of box as is.

kylezimmerman, 2019-02-06:
@roygbiv856 I've got a little over half of my rooms set up with Xiaomi motion sensors (they are really cheap). It's so cool when it all comes together!

@bereska Yep that should be fine. That's basically my setup except I'm using Tradfri instead. I find the mix to be a nice balance between network quality and $$$.

bereska, 2019-02-06:
@kylezimmerman thank you

roygbiv856, 2019-02-07:
@kylezimmerman hey, buddy you inspired me to get my xiaomi motion sensor up and running. Problem is, it doesn't seem to pair. I've been holding the button for 5 seconds, then hitting it about once a second like I did to all of my other xiaomi devices, but no luck. Remember the process you used?

kylezimmerman, 2019-02-08:
@roygbiv856 That's awesome! You'll love it when its all set up.

I've had trouble pairing these before. I can't tell exactly what works 100% but here are some tips:

- Use something other than your finger to hold the button. Somehow my finger seems to press it but it doesn't hold it down enough or something. I used a toothpick and had much better results.
- Try removing the battery and putting it back in. Or replacing it if its low/dead.
- You already mentioned it but after holding it for 5sec and it starts to blink, keep pressing it about once every second or so.
- It takes a while sometimes. I've given up before and stopped pressing it repeatedly, put it down on the desk and when it sends a motion message it finally gets picked up.
- Restart zigbee2mqtt or home assistant.
- Make sure you haven't hit the device limit

Good luck!

roygbiv856, 2019-02-08:
@kylezimmerman  Really appreciate it! Works awesome. Now just have to order like 10 more, ha. Turns out after adding an Ikea Tradfri plug, z2m stopped accepting new devices. 

@Koenkk I know in another thread (that I can't find) somebody talked about z2m not allowing the pairing of new devices after adding an Ikea Tradfri plug. Just wanted to confirm this happened to me yesterday. I'm pretty new to GH so not really sure how to report a bug.

closed 2019-04-16 by stale[bot]