Koenkk/zigbee2mqtt #996: Settings changed after update to 1.1.1

Bruceforce, 2019-02-02:
Hi,

I updated the docker image (latest-dev) from version 1.1.0 to the new image with version 1.1.1. After the update I noticed, that my configuration.yaml was changed. The format of the network key was changed from hex to int automatically and the representation changed from

oneline hex list

```
advanced:
  network_key: [0x01, 0x02, 0x03 ...]
```

 to multiline int list
```
advanced:
  network_key:
    - 1
    - 2
    - 3 
```

That's only a cosmetic change of course, but what's a bit more concerning. The `permit_join` was set to true, although I'm absolutely sure that it was set to false before the update. In my opinion these settings - especially the `permit_join` should never be changed automatically.

Koenkk, 2019-02-02:
This is because zigbee2mqtt adds devices to the `configuration.yaml` (when a new device joins the network). The internal representation in zigbee2mqtt doesn't contain formatting/comments. When it's written back, the format of the used YAML library is used. This will only be cosmetic changes (different array formatting, hex -> decimal). 

The `permit_join` is not changed automatically, but note that this can also be changed via an MQTT topic.

Bruceforce, 2019-02-02:
Okay thanks for the clarification.

I only changed `permit_join` via MQTT message and changed it back, but I thought after a restart of zigbee2mqtt it should use the value specified in `configuration.yaml` or am I wrong?

I checked this via an automatically generated backup of the `configuration.yaml` and `permit_join` seems to be changed in the same time as the other things (remove of comments and so on).

I will keep an eye on that the next time I update or add devices and will post here again.

marc-gist, 2019-02-17:
any chance we can preserve comments in the configuration.yaml file? handy to have extra comments in there at times? a separate file works, but not as easy to keep track of.

Koenkk, 2019-02-17:
No, the underlying yaml library doesnt support this, I couldnt find a library which does.

Bruceforce, 2019-04-05:
Closing since I have not seen this again. It was most probably in fact my fault.

closed 2019-04-05 by Bruceforce