Koenkk/zigbee2mqtt #997: Ikea Outlet Switch Support (E1743)

Curtisjk, 2019-02-02:
Is there any main reason why the switch that comes bundled with the Ikea Tradfri Outlet (https://www.ikea.com/gb/en/products/lighting/smart-lighting/trådfri-control-outlet-kit-art-10364797/) is not supported yet? I'm going to be buying some of these outlets and was wondering whether to get the switches at the same time as they are not sold separately.

I know the other Ikea Tradfri remote only got support added recently due to it using Zigbee groups and random IDs (#102) - would there be similar issues with this switch, or does it just need the converters writing?

There was a discussion in #607 but no PR was ever created. 

Many thanks!

Koenkk, 2019-02-02:
AFAIK only some simple modifications to zigbee-shepherd are needed (I can help with this). Let me know once you have the device

HamsterHauge, 2019-02-02:
Hi.
I have the device, if I can help.

Koenkk, 2019-02-02:
@HamsterHauge can you pair the device and provide the output?

Curtisjk, 2019-02-02:
I have just found out this is out of stock all over the UK so may not be
able to help get it integrated just yet!

HamsterHauge, 2019-02-02:
I can control the Outlet.

HamsterHauge, 2019-02-02:
Sorry, my english.. It's the switch that are the problem. And I still can't pair it..

antti365, 2019-02-04:
Hello everyone, finally delivered to Finland outlets with switches, bought, ready for testing. :)

antti365, 2019-02-04:
@Koenkk I understand correctly, for the switch need firmware that supports the group? (to use groups, at least firmware version 20181224 is required!)
Because the switch successfully communicates with the outlet, but absolutely does not want to contact the stick, there is silence in the log file ...
Unfortunately, my programmer has not reached me yet ... :(

Koenkk, 2019-02-04:
That shouldnt be needed, can you share your `database.db` with the device paired?

antti365, 2019-02-04:
@Koenkk So the fact of the matter is that the switch does not even begin the process of pairing with the stick. First, I factory reset by pressing the button 4 times, then I wait for it to stop glowing, and hold the pairing button for> 10 seconds and nothing happens at all, and there is nothing in the log file at all ...

HamsterHauge, 2019-02-04:
Absolutely the same here.

antti365, 2019-02-04:
@Koenkk @HamsterHauge I removed the socket from the database, reset the stick and switch and tried again, records appeared in the log file! :)

`Feb 04 22:11:29 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:29 PM Connecting with device...
Feb 04 22:11:29 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:29 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:30 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:30 PM Connecting with device...
Feb 04 22:11:30 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:30 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:34 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:34 PM Connecting with device...
Feb 04 22:11:34 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:36 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:36 PM Connecting with device...
Feb 04 22:11:36 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:36 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:38 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:38 PM Connecting with device...
Feb 04 22:11:38 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:38 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:40 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:40 PM Connecting with device...
Feb 04 22:11:40 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:40 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:44 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:44 PM Connecting with device...
Feb 04 22:11:44 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:44 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:46 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:46 PM Connecting with device...
Feb 04 22:11:46 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:46 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:51 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:51 PM Connecting with device...
Feb 04 22:11:51 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:53 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:53 PM Connecting with device...
Feb 04 22:11:53 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:53 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:57 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:57 PM Connecting with device...
Feb 04 22:11:57 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:57 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM Connecting with device...
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM Device incoming...
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM New device with address 0x000d6ffffeb1d3c5 connected!
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:info 2/4/2019, 10:11:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6ffffeb1d3c5"}'
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:warn 2/4/2019, 10:11:59 PM Device with modelID 'TRADFRI on/off switch' is not supported.
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:warn 2/4/2019, 10:11:59 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:warn 2/4/2019, 10:11:59 PM Device with modelID 'TRADFRI on/off switch' is not supported.
Feb 04 22:11:59 raspberrypi3 npm[20780]:   zigbee2mqtt:warn 2/4/2019, 10:11:59 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`

antti365, 2019-02-04:
@Koenkk 
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/2829698/database.db.zip)

Attila-FIN, 2019-02-05:
I've been also playing with CC2531 and zigbee2mqtt for two days now, I have successfully paired and controlled Tradfri Control outlet E1603 from Node-RED, paired and received messages from the 5 button Tradfri Remote E1524 (however I had to update node packages, especially zigbee-shepherd-converters, with **npm update**).
Unfortunately I could not make the Tradfri ON/OFF Switch E1743 working. I could pair it after re-inserting the CC2531 stick and restarting zigbee2mqtt. After pairing, in Wireshark I could also see ON and OFF commands sent by the switch (btw, there is also move UP and DOWN commands when buttons are kept pushed). In Wireshark I could also see, that group number is always **0x0000** for the switch, unlike in the case of E1524, where it is randomly generated. I guess that means no group.
I have created a new entry in the devices.js if the zigbee-shepherd-converters package:
```
    {
        zigbeeModel: ['TRADFRI on/off switch'],
        model: 'E1743',
        vendor: 'IKEA',
        description: 'Remote with ON/OFF buttons',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
    },
```
However I haven't seen anything in the zigbee2mqtt console. I believe there might be some configure section needed for the new device, but I do not know what would be needed there.

Koenkk, 2019-02-05:
Can you try with:

```js
    {
        zigbeeModel: ['TRADFRI on/off switch'],
        model: 'E1743',
        vendor: 'IKEA',
        description: 'Remote with ON/OFF buttons',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
            ];

            execute(device, actions, callback);
    },
```

In case that doesn't work

```js
    {
        zigbeeModel: ['TRADFRI on/off switch'],
        model: 'E1743',
        vendor: 'IKEA',
        description: 'Remote with ON/OFF buttons',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', 99, cb),
            ];

            execute(device, actions, callback);
    },
```

and with sniffing verify that the remote now uses group 99?

Attila-FIN, 2019-02-05:
Thanks @Koenkk , I tried both with the same results (the "configure: ..." part had to be added).
In Wireshark I can see the bind request is sent to the remote and there is successful response, but I don't get any message like "WARN No converter available for..." in zigbee2mqtt output.
Please see the Wireshark capture file about bind request and ON/OFF commands:
[Tradfri E1743 binding to group 99.zip](https://github.com/Koenkk/zigbee2mqtt/files/2833956/Tradfri.E1743.binding.to.group.99.zip)
Earlier. while trying to reuse configure sections from other devices, I managed to get the battery level reported for every button press, at least that what it said, but values were strange. Perhaps the ON/OFF values were incorrectly parsed.

Koenkk, 2019-02-06:
@Attila-FIN please update to the latest dev branch, the coordinator will now be added to group 99 where it should receive commands of the remote.

Attila-FIN, 2019-02-08:
@Koenkk , I pulled the latest dev and used the following devices.js entry for the E1743 remote:
```
    {
        zigbeeModel: ['TRADFRI ON/OFF Switch'],
        model: 'E1743',
        vendor: 'IKEA',
        description: 'Remote with ON/OFF buttons',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
                const device = shepherd.find(ieeeAddr, 1);
                const actions = [
                        (cb) => device.bind('genOnOff', 99, cb),
                ];

            execute(device, actions, callback);
        },
    },
```
Now commands from remote are shown in the zigbee2mqtt output:
```
No converter available for 'E1743' with cid 'genOnOff', type 'cmdOff' and data '{"cid":"genOnOff","data":{}}'
No converter available for 'E1743' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
No converter available for 'E1743' with cid 'genLevelCtrl', type 'cmdMove' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":83}}'
No converter available for 'E1743' with cid 'genLevelCtrl', type 'cmdStopWithOnOff' and data '{"cid":"genLevelCtrl","data":{}}'
No converter available for 'E1743' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":83}}'
No converter available for 'E1743' with cid 'genLevelCtrl', type 'cmdStopWithOnOff' and data '{"cid":"genLevelCtrl","data":{}}'
```
Now, only the right converters needed to be picked/created.

Koenkk, 2019-02-08:
When are those genLevelCtrl commands send?

Attila-FIN, 2019-02-08:
I played around a bit with converters, I created new ones, which correspond more to the Wireshark interpreted command description. Unfortunately my understanding of the architecture and generally Zigbee is not very deep yet. so certain things look mysterious to me:

- Why holding the button in the "down" direction is a CMD type 'cmdMove', but "up" direction is cmdMoveWithOnOff? This requires two converters (using the comparison is basically unnecessary in my code below)?
- Where the 'brightness' level is coming from, which is always 255?
- Why every CMD is followed with an "idle" click/action, with 120 (max?) "linkquality"?

Anyway, here are the converters, I hope they can be used for something:
```
    E1743_on: {
        cid: 'genOnOff',
        type: 'cmdOn',
        convert: (model, msg, publish, options) => {
            return {click: 'on'};
        },
    },
    E1743_off: {
        cid: 'genOnOff',
        type: 'cmdOff',
        convert: (model, msg, publish, options) => {
            return {click: 'off'};
        },
    },
    E1743_move_down: {
        cid: 'genLevelCtrl',
        type: 'cmdMove',
        convert: (model, msg, publish, options) => {
            const direction = msg.data.data.movemode === 1 ? 'down' : 'up';
            return {action: `move_${direction}`, rate: msg.data.data.rate};
        },
    },
    E1743_move_up: {
        cid: 'genLevelCtrl',
        type: 'cmdMoveWithOnOff',
        convert: (model, msg, publish, options) => {
            const direction = msg.data.data.movemode === 1 ? 'down' : 'up';
            return {action: `move_${direction}`, rate: msg.data.data.rate};
        },
    },
    E1743_move_stop: {
        cid: 'genLevelCtrl',
        type: 'cmdStopWithOnOff',
        convert: (model, msg, publish, options) => {
            return {action: 'move_stop'};
        },
    },
```

Koenkk, 2019-02-08:
@Attila-FIN experimental support is in https://github.com/Koenkk/zigbee-shepherd-converters/tree/E1743 now. 

* Why holding the button in the "down" direction is a CMD type 'cmdMove', but "up" direction is cmdMoveWithOnOff? This requires two converters (using the comparison is basically unnecessary in my code below)?
IKEA probably uses this to distinguish between both, I've seen more strange stuff like his from IKEA. If the converters work, don't worry too much about this.

* Where the 'brightness' level is coming from, which is always 255?
Where do you see this?

* Why every CMD is followed with an "idle" click/action, with 120 (max?) "linkquality"?
This is an experimental artificial event from zigbee2mqtt (will be removed soon). Linkquality is added for every device, in theory the max is 255, the more closer you keep it to the coordinator the bigger it should get.

vke, 2019-02-09:
@Koenkk could you please add both variants of sensor name? `'TRADFRI ON/OFF Switch', 'TRADFRI on/off switch'`, because of my sensor called `"TRADFRI on/off switch"`. 

```
'{"click":"brightness_down","linkquality":105}'
'{"click":"brightness_stop","linkquality":105}'
'{"click":"brightness_up","linkquality":105}'
'{"click":"brightness_stop","linkquality":105}'
'{"click":"on","linkquality":105}'
'{"click":"off","linkquality":105}'
```

As for the rest, it began to work, **thank you so much**!

antti365, 2019-02-09:
Great job! It remains only to add the value of the battery ... :)

Attila-FIN, 2019-02-09:
@vke , sorry that way my mistake. only the label printed on the box is with capital, the correct zigbee model ID is in fact NOT with capital letter, so the correct one is:
**'TRADFRI on/off switch'**

Koenkk, 2019-02-09:
Updated https://github.com/Koenkk/zigbee-shepherd-converters/tree/E1743, please try.
- Added battery support, please confirm, this could take up to one day to be reported.
- Fixed zigbee model ID.

If everything works, we can merge this.

vke, 2019-02-09:
I see also that IKEA devices report their model (like E1743) and battery level both as percents and as voltage:

```ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 99
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 99
    Command: Read Attributes Response (0x01)
    Status Record, Octets: 45:31:37:34:33
        Attribute: Unknown (0x000a)
        Status: Success (0x00)
        Data Type: Octet String (0x41)
        Octet String: 45:31:37:34:33
    Status Record, String: 
        Attribute: Unknown (0x000b)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: 
    Status Record, String: 2.0.020
        Attribute: Software Build Id (0x4000)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: 2.0.020
    Status Record, Uint16: 1
        Attribute: Unknown (0xfffd)
        Status: Success (0x00)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 1 (0x0001)
```

```ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 166
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 166
    Command: Read Attributes Response (0x01)
    Status Record
        Attribute: Battery Voltage (0x0020)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Measured Battery Voltage: 2.9 [V]
    Status Record
        Attribute: Battery Percentage Remaining (0x0021)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Remaining Battery Percentage: 43.5 [%]
    Status Record, Uint16: 1
        Attribute: Unknown (0xfffd)
        Status: Success (0x00)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 1 (0x0001)
```

vke, 2019-02-09:
Oh, 1-minute late, thanks, I will test right now

Koenkk, 2019-02-09:
@vke If you want to do a quick sanity check, change https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html `repInterval.MAX` to `3` (it should report every 3 seconds than).

vke, 2019-02-09:
@Koenkk thank you very much! Everything is working as expected.

```{"battery":87,"linkquality":107,"click":"brightness_down"}
{"battery":87,"linkquality":110,"click":"on"}
{"battery":87,"linkquality":107,"click":"off"}
{"battery":87,"linkquality":110,"click":"brightness_up"}
{"battery":87,"linkquality":110,"click":"brightness_stop"}
```

Attila-FIN, 2019-02-09:
Thanks @Koenkk for the explanations. I have tried the E1743 branch of the zigbee-shepherd-converters  and the latest dev branch of zigbee2mqtt. I reset the remote and re-paired with zigbee2mqtt. 
Commands are coming and translated to MQTT well, however I still see the strange "brightness":255. I must do something wrong.
After restarting zigbee2mqtt , there were also unhandled 'genPowerCfg' messages coming for the first few button presses with strange batteryPercentageRemaining values:
```
  zigbee2mqtt:info 2019-2-10 00:10:37 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":94,"brightness":255,"click":"off"}'
  zigbee2mqtt:info 2019-2-10 00:10:40 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":21,"linkquality":94,"brightness":255}'
  zigbee2mqtt:warn 2019-2-10 00:10:40 No converter available for 'E1743' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":21}}'
  zigbee2mqtt:warn 2019-2-10 00:10:40 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2019-2-10 00:10:40 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":21,"linkquality":94,"brightness":255,"click":"off"}'
  zigbee2mqtt:info 2019-2-10 00:10:44 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":76,"brightness":255}'
  zigbee2mqtt:warn 2019-2-10 00:10:44 No converter available for 'E1743' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":47}}'
  zigbee2mqtt:warn 2019-2-10 00:10:44 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2019-2-10 00:10:44 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":78,"brightness":255,"click":"on"}'
  zigbee2mqtt:info 2019-2-10 00:10:46 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":60,"linkquality":70,"brightness":255}'
  zigbee2mqtt:warn 2019-2-10 00:10:46 No converter available for 'E1743' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":60}}'
  zigbee2mqtt:warn 2019-2-10 00:10:46 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2019-2-10 00:10:46 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":60,"linkquality":76,"brightness":255,"click":"on"}'
  zigbee2mqtt:info 2019-2-10 00:10:47 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x000d6ffffeb2bb92","type":"EndDevice","model":"E1743","friendly_name":"IKEA_ONOFF_Switch_01"}]}'
  zigbee2mqtt:info 2019-2-10 00:10:47 MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":94,"brightness":255}'
  zigbee2mqtt:warn 2019-2-10 00:10:47 No converter available for 'E1743' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":47}}'
  zigbee2mqtt:warn 2019-2-10 00:10:47 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
```

Afterwards, the translated MQTT messages looked good (except the "brightness" part):
```
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":92,"brightness":255,"click":"off"}'
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":99,"brightness":255,"click":"on"}'
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":89,"brightness":255,"click":"brightness_up"}'
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":92,"brightness":255,"click":"brightness_stop"}'
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":94,"brightness":255,"click":"brightness_down"}'
MQTT publish: topic 'zigbee2mqtt/IKEA_ONOFF_Switch_01', payload '{"battery":47,"linkquality":94,"brightness":255,"click":"brightness_stop"}'
```

Koenkk, 2019-02-10:
- Added a converter to ignore the `genPowerCfg` `devChange` message
- The `brightness` is probably in your `data/state.json`. Can you try removing this entry from `state.json`? Shouldn't appear afterwards anymore.

Curtisjk, 2019-02-10:
Thanks for this - I look forward to using it when I finally get my hands on one of the kits!

Attila-FIN, 2019-02-10:
@Koenkk , thanks! That was really my problem in the `data/state.json` file. I perhaps had that state stored there while playing with different converters. I removed the entry of the remote and now the correct messages are published.
Also unhandled 'genPowerCfg' messages are gone now.
It is really nice to have the battery level reported in every MQTT message, perhaps that cold be also implemented for the 5 button E1524 remote too, but that is off topic.

vke, 2019-02-10:
@Attila-FIN @Koenkk yes, E1524 remote is working exactly as this E1743 with battery. Just tried.

Koenkk, 2019-02-11:
Merged! will be supported in the dev branch within a few hours.

closed 2019-02-11 by Koenkk

patriot1889, 2019-02-12:
I feel like the long press clicks are the wrong way around?

"brightness_up" is called when holding down the off button.
"brightness_down" is called when holding the on button.

Surely it should be the other way around? Or is this based on information actually coming from the device?

patriot1889, 2019-02-14:
Ok. I thought everything was fine and dandy but I keep getting the following errors;
`PM Failed to configure IkeaSwitch1 (0xd0cf5efffed6ef5c) ('Error: Timed out after 10000 ms') (attempt #1)`

Any thoughts?

KidFoxXL, 2019-03-06:
I tried to connect the button but I can't get it to work.
I noticed, that not every button click is recognized and there is no long press click.
The MQTT payload has no "click" message.
`Mar 06 19:35:44 openHABianPi npm[505]:   zigbee2mqtt:info 3/6/2019, 7:35:44 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffed16ff8', payload '{"battery":100,"linkquality":42}'`

Am I doing something wrong?

cedricous, 2019-03-08:
Same for me, the MQTT payload has no "click" message. I use the edge version of the zigbee2mqtt on the Hass.io, put the parameter "zigbee_shepherd_devices": true, and put the devices.js from the zigbee-shepherd-converters github in the share/zigbee2mqtt directory. 
Maybe I forgot something ?

cedricous, 2019-03-08:
In fact the mqtt message is received only one time after the button is paired, after that even if I click on the button nothing appeard : Here the log of the paired

  zigbee2mqtt:info 3/8/2019, 4:54:57 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
  zigbee2mqtt:info 3/8/2019, 4:55:32 PM New device 'undefined' with address 0x000d6ffffe2f1753 connected!
  zigbee2mqtt:info 3/8/2019, 4:55:32 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6ffffe2f1753","meta":{}}'
  zigbee2mqtt:warn 3/8/2019, 4:55:32 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/8/2019, 4:55:32 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 3/8/2019, 4:55:33 PM Failed to configure 0x000d6ffffe2ce7ca (0x000d6ffffe2ce7ca) ('Error: Timed out after 10000 ms') (attempt #1)
  zigbee2mqtt:warn 3/8/2019, 4:55:33 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/click/config', payload '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","expire_after":1,"state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_click","unique_id":"0x000d6ffffe2f1753_click_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_battery","unique_id":"0x000d6ffffe2f1753_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/8/2019, 4:55:35 PM MQTT publish: topic 'homeassistant/sensor/0x000d6ffffe2f1753/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6ffffe2f1753","json_attributes_topic":"zigbee2mqtt/0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753_linkquality","unique_id":"0x000d6ffffe2f1753_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2f1753","name":"0x000d6ffffe2f1753","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/8/2019, 4:55:45 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":100,"linkquality":128}'
  zigbee2mqtt:warn 3/8/2019, 4:56:01 PM Failed to configure 0x000d6ffffe2f1753 (0x000d6ffffe2f1753) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
  zigbee2mqtt:warn 3/8/2019, 4:56:01 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 3/8/2019, 4:56:05 PM Device incoming...
  zigbee2mqtt:info 3/8/2019, 4:56:05 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 3/8/2019, 4:56:12 PM Successfully configured 0x000d6ffffe2f1753 (0x000d6ffffe2f1753)
  zigbee2mqtt:info 3/8/2019, 4:56:29 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2f1753', payload '{"battery":60,"linkquality":120}'