Koenkk/zigbee2mqtt #998: No pairing to Osram smart+ plug (v1.1.0.1)

michel72, 2019-02-02:
Today I reinstalled my network using the latest firmware for the CC2530 coordinator (09-01-2019) and zigbee2mqtt version 1.1.0. 

I first paired both my Osram smart+ plugs as the coordinator can not cover my entire house.

I had real issues pairing most of my (16) devices. Especially the Ikea GU10 bulb and the Xiaomi door/window sensors as well as the Aqara temperature sensors proved the be a real challenge.

I ended up removing the sensors and bulbs to bring them close to the coordinator to pair. All in all this took me the better part of the day. :(

I am now quit sure it is not possible to pair to the (osram) routers. Is this known issue I failed to find, or actually a bug?

universal, 2019-02-02:
Hmm... I'm on the same version and I've got one osram plug as well. It is connected and shows up as a router in the zigbee map, but the aqara temperature sensor didn't connect through it as well... If there is anything i can test, I'd be happy to.

Koenkk, 2019-02-03:
It depends on the Zigbee implementation of the routers. Some forwards joining device packets, some don't.

universal, 2019-02-03:
okay, then, iirc what i have read:

the plug will not work as a router with the aqara sensors, since they only send via the original pathway they were paired with :-) gonna try it now again, have to keep awake for the superbowl anyways...

michel72, 2019-02-04:
Keep us posted please!

michel72, 2019-02-05:
Though I was unable to pair the aqara devices to the routers, irt does seem that these enddevices can connect to a router. I have several links to both the coordinator asthe router. My zigbee map loos quit different (and a bit strange) since the update to v1.1.x

[image]

universal, 2019-02-05:
put the plug in another socket and turned it by 180 degrees... seems to be stable now... both sensors update through it now. so it might just be a little bit fiddly to figure out a setting that works. I read somewhere on reddit, that others had similar problems with the plug :-)

georg90, 2019-02-07:
Sorry to jump in this topic..

I have the OSRAM plugs and they pair fine, when I click the button the mqtt publish is received by all connected devices. However, when I try to publish an ON/OFF message to my device, it doesn't switch the plug..

What is the proper command to turn on/off the Osram plug?

I use the following:
```
topic: zigbee2mqtt/0x84182600000fafaa
payload: {"state":"ON"}
```

Any help is appreciated, if this is totally off topic, let me know and I'll delete the post

universal, 2019-02-08:
i'll have to check at home... ping my name if I haven't replied until next week :-)

closed 2019-04-16 by stale[bot]