Koenkk/zigbee2mqtt #1005: zigbee2mqtt crashing at start because of some new Livolo related code

milakov, 2019-02-03:
Hi, I am trying to pair Xiaomi doublу key switch, without much sucess. At some point it appeared in the list under unknown mnaufacturer name and I am repeatedly getting this error:
```
/app/lib/extension/pollLivoloSwitch.js:69
            .filter((d) => d.manufName.startsWith('LIVOLO')) // LIVOLO
```
zigbee2mqtt restarts and encounter the same error and so on. I have no Livolo devices connected.

Landrash, 2019-02-03:
Second that. Tested with DIY router and IKEA Lights.

Koenkk, 2019-02-03:
Can you post the complete exception?

khzd, 2019-02-03:
hi Koenkk
During this intesive test to implement LIVOLO, what about to use @ptvoinfo branch on github ? so it won't disturb your dev branch ?
cheers

milakov, 2019-02-03:
Here is the log (the device 0x00158d00027c1b4c in the beginning is correctly working one):
```
  zigbee2mqtt:info 2019-2-3 18:46:35 MQTT publish: topic 'homeassistant/sensor/0x00158d00027c1b4c/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00027c1b4c","json_attributes_topic":"zigbee2mqtt/0x00158d00027c1b4c","name":"0x00158d00027c1b4c_battery","unique_id":"0x00158d00027c1b4c_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00027c1b4c","name":"0x00158d00027c1b4c","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara double key wireless wall switch (WXKG02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
/app/lib/extension/pollLivoloSwitch.js:69
            .filter((d) => d.manufName.startsWith('LIVOLO')) // LIVOLO
                                       ^
TypeError: Cannot read property 'startsWith' of undefined
    at zigbee.getAllClients.filter (/app/lib/extension/pollLivoloSwitch.js:69:40)
    at Array.filter (<anonymous>)
    at PollLivoloSwitch.handleInterval (/app/lib/extension/pollLivoloSwitch.js:69:14)
    at Timeout.timer.setInterval [as _onTimeout] (/app/lib/extension/pollLivoloSwitch.js:23:45)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
npm
 
ERR!
 code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.1.0 start: `node index.js`
npm 
ERR! Exit status 1
npm ERR! 
npm
 ERR! Failed at the zigbee2mqtt@1.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR!
 A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-02-03T15_46_35_822Z-debug.log
2019-02-03T18:46:35: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-02-03T18:46:35: PM2 log: App [npm:0] starting in -fork mode-
2019-02-03T18:46:35: PM2 log: App [npm:0] online
```

Koenkk, 2019-02-03:
@milakov thanks, added an extra check so this shouldn't happen anymore.