Koenkk/zigbee2mqtt #1007: Xiaomi Gas sensor has wrong homeassistant config

rzhukov, 2019-02-03:
1. The device doesn't have battery, it mains powered, but mqtt dicsovery config provides battery sensor. https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L390
2. Density sensor configured with wrong json attribute `gas_density` https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L181, but converter generates `density` attr: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L962, so it always in "unavailable" state

Koenkk, 2019-02-03:
@rzhukov thanks for spotting this. Could you make a PR? For point 2, I think the converter should be updated to `gas_density`.

Koenkk, 2019-02-03:
Merged, thanks!

PR #1009 (2019-02-03): Fix Xiaomi Gas Leak Sensor Homeassistant mqtt discovery config
Fix #1007 (point 1)
Remove battery sensor
closed 2019-02-03 by Koenkk