Koenkk/zigbee2mqtt #1010: [QUESTION] reset zigbee2mqtt repository

yaniv1111, 2019-02-03:
Hi,
I have tested zigbee with my xiaomi hub and it works great, but I haven't enabled it security
1. is there a way to enabled it without repairing all devices
2. if not, how can i revert zigbee2mqtt configuration to starting point?

thanks

Yaniv

yaniv1111, 2019-02-03:
thank for the quick respond,

what is the advance configuration for setup ssl?
e.g
m2mqtt_ca.crt  : CA Certificate
m2mqtt_ca.key  : CA key pair (private, public)
m2mqtt_ca.srl  : CA serial number file
m2mqtt_srv.crt : server certificate
m2mqtt_srv.csr : certificate sign request, not needed any more
m2mqtt_srv.key : server key pair

https://mcuoneclipse.com/2017/04/14/enable-secure-communication-with-tls-and-the-mosquitto-broker/

thanks again
Yaniv

Koenkk, 2019-02-03:
I'm not sure about that, I assume the url has to be something like `mqtts://localhost:1883` (the underlying MQTT library supports this: https://www.npmjs.com/package/mqtt)

closed 2019-04-11 by stale[bot]

ben423423n32j14e, 2019-06-29:
Did you figure this out?

I've tried mqtts://localhost:1883 but can't get it to connect.

ben423423n32j14e, 2019-06-30:
I found a (not ideal) solution which was to set: require_certificate false in mosquitto.conf