Koenkk/zigbee2mqtt #1012: hue/saturation color mode not working

mrhumpen, 2019-02-03:
Hi,
 
I saw that #948 was added a few days ago and was eager to test. 
Unfortunately, it does not work for me.

Currently I am running zigb2mqtt 1.1.1 and trying to set the light (Ikea Trafdri  LED1624G9) with HSB values from OpenHAB.

MQTT msg:
`{"brightness":255,"state":"ON","color":{"hue":118,"saturation":93}}`

Log:
`2019-2-3 20:10:05 - info: Zigbee publish to device '0xd0cf5efffe1619eb', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null`

`2019-2-3 20:10:05 - info: MQTT publish: topic 'zigbee2mqtt/LED1624G9_45', payload '{"state":"ON","brightness":254,"last_seen":"2019-02-03T19:09:56.430Z","color_mode":1,"color":{"x":0.501,"y":0.414}}`

`2019-2-3 20:10:05 - info: Zigbee publish to device '0xd0cf5efffe1619eb', lightingColorCtrl - enhancedMoveToHueAndSaturation - {"transtime":0,"enhancehue":21480.916666666664,"saturation":236.22,"direction":0} - {"manufSpec":0,"disDefaultRsp":0} - null
`

The brightness setting is working perfectly fine, but the color of the bulb does not change at all.
If I send the message using RGB values the bulb changes the color as exprected.

mrhumpen, 2019-03-13:
@Koenkk 
I just came accross this issue and it turns out that your first guess is a hit :-)
#https://github.com/eclipse/smarthome/issues/4939

Would it make sense to handle this internally in zigbee2mqtt so that only color formats are being send that are actually supported by the bulb? Or even better, convert them internally and send a 
supported command to the bulb? 

If you dont have to pay attention to color format in the MQTT set command, it would make things easier.

Koenkk, 2019-03-14:
Yes, that would be an option, could you figure out how to convert this (color temperature -> xy)? (find a function on the internet?)

closed 2019-05-20 by stale[bot]

sjorge, 2019-08-11:
I've been using this in node-red, it seems to work fine if you pass in v = 100. I wish I could remember were I got this from but I have been hacking together multiple broken examples over the last month :(

```javascript
function hsv2rgb (h, s, v = 100) {
    var RGB_MAX = 255;
    var HUE_MAX = 360;
    var SV_MAX = 100;
    
    if (typeof h === 'object') {
        const args = h
        h = args.h; s = args.s; v = args.v;
    }

    h = (h % 360 + 360) % 360;
    h = (h === HUE_MAX) ? 1 : (h % HUE_MAX / parseFloat(HUE_MAX) * 6)
    s = (s === SV_MAX) ? 1 : (s % SV_MAX / parseFloat(SV_MAX))
    v = (v === SV_MAX) ? 1 : (v % SV_MAX / parseFloat(SV_MAX))

    var i = Math.floor(h)
    var f = h - i
    var p = v * (1 - s)
    var q = v * (1 - f * s)
    var t = v * (1 - (1 - f) * s)
    var mod = i % 6
    var r = [v, q, p, p, t, v][mod]
    var g = [t, v, v, q, p, p][mod]
    var b = [p, p, t, v, v, q][mod]

    return {
        r: Math.floor(r * RGB_MAX),
        g: Math.floor(g * RGB_MAX),
        b: Math.floor(b * RGB_MAX),
    }
}
```

h = 0, s = 100, v = 100 will result in 255, 0, 0 which is red and correct.

I'm using this to convert the output from a homekit service (Hue + Sat) to rgb to feed to an ikea bulb.