Koenkk/zigbee2mqtt #1014: Problems pairing a second Lightify Gardenstpot RGB

YamiFrankc, 2019-02-04:
Hello,

Right now im running the dev branch. I was able to pair my first Gardenspot without any problem, worked flawlessly. But now I am trying to join a second one and im having trouble doing so. I am following the instructions listed [here](https://www.belkin.com/us/support-article?articleNum=159343) to reset them and get in pairing mode. They do not flash 3 times, but they color cycle like mentioned [here](https://community.smartthings.com/t/osram-sylvania-gardenspot-pairing-issue/60008).
I dont really see anything in the logs, [https://pastebin.com/6D0Pvchh](https://pastebin.com/6D0Pvchh)

any tips?

Thanks

Koenkk, 2019-02-04:
Did you pair them to a other hub previously?

YamiFrankc, 2019-02-05:
Yes. They were paired to another previous instance of zigbee2mqtt using the same stick.
Since then I have moved to a new install using the dev branch a reflashed the stick

Koenkk, 2019-02-05:
@YamiFrankc are they still in the `database.db`?

YamiFrankc, 2019-02-05:
They arent

Koenkk, 2019-02-06:
When pairing and running with `DEBUG=* npm start`, do you see any logging appearing when trying to pair?

YamiFrankc, 2019-02-07:
not really, this is what i get: [pastebin](https://pastebin.com/6mYAxfLy)

Koenkk, 2019-02-07:
@YamiFrankc did you successfully pair this device to another hub?

YamiFrankc, 2019-02-07:
This device im having problems with is actually the one I made a post about before, so it was paired  with an older install of zigbe2mqtt with this same stick,
Right now I tried to pair other devices and they dont work. Noting appears in the logs with any device

Koenkk, 2019-02-07:
Did you reflash the stick in between?

YamiFrankc, 2019-02-07:
Yes

Koenkk, 2019-02-07:
Can you post your complete zigbee2mqtt startup log?

YamiFrankc, 2019-02-07:
Here: [pastebin](https://pastebin.com/rqVeHHcx)

Koenkk, 2019-02-07:
Is `permit_join: true` in `configuration.yaml`?

Your log shows:
`2/7/2019, 7:03:20 AM - info: Zigbee: disabling joining new devices.`

YamiFrankc, 2019-02-07:
Its not, but I use the HA automation that is in the wiki to enable joining. Been working perfectly until now

Koenkk, 2019-02-07:
Please set `permit_join: true` in `configuration.yaml`, from the log I cannot see that joining is enabled, and try again to pair devices.

YamiFrankc, 2019-02-07:
[Output log](https://pastebin.com/93VdGpE8)

Koenkk, 2019-02-07:
Are all the not paired Zigbee devices offline? As the devices of the old zigbee network (before flashing your stick) could still form a zigbee network. When trying to pair devices, they will prefer to reconnect to the network they were already paired to (the old network).

YamiFrankc, 2019-02-09:
All on the other devices paired were working correctly.
The problem fixed itself when I restarted the raspberry running zigbee2mqtt, I was able to pair devices again

closed 2019-02-09 by YamiFrankc