Koenkk/zigbee2mqtt #1015: No matching payload found for entity: 0xxxxxx_water_leak

milakov, 2019-02-04:
Hi, I started to see warnings in the main log with the recent dev branch:
```
No matching payload found for entity: 0x00158d00024a4e92_water_leak with state_topic: zigbee2mqtt/0x00158d00024a4e92
9:11 components/mqtt/binary_sensor.py (WARNING)
```
This is what I have at integration page:
[image]

milakov, 2019-02-04:
Start up log (messages for other devices removed):
```
2019-2-3 20:28:21 - info: Logging to directory: '/share/zigbee2mqtt/log/2019-02-03.20-28-21'
2019-2-3 20:28:27 - info: Starting zigbee2mqtt version 1.1.0 (commit #8a65827)
2019-2-3 20:28:27 - info: Starting zigbee-shepherd
2019-2-3 20:28:29 - info: zigbee-shepherd started
2019-2-3 20:28:29 - info: Coordinator firmware version: '20181024'
2019-2-3 20:28:29 - info: Currently 19 devices are joined:
2019-2-3 20:28:29 - info: 0x00158d00024a5018 (0x00158d00024a5018): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
2019-2-3 20:28:29 - info: 0x00158d00024a4e92 (0x00158d00024a4e92): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
2019-2-3 20:28:29 - info: Zigbee: disabling joining new devices.
2019-2-3 20:28:29 - info: Connecting to MQTT server at mqtt://192.168.1.170
2019-2-3 20:28:29 - info: zigbee-shepherd ready
2019-2-3 20:28:30 - info: Connected to MQTT server
2019-2-3 20:28:30 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-2-3 20:28:30 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a5018', payload '{"battery":100,"voltage":3015,"linkquality":26,"last_seen":"2019-02-03T16:44:02.730Z"}'
2019-2-3 20:28:30 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a4e92', payload '{"battery":97,"voltage":2995,"linkquality":28,"last_seen":"2019-02-03T16:48:29.006Z"}'
2019-2-3 20:28:30 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d00024a5018/water_leak/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.water_leak }}","device_class":"moisture","state_topic":"zigbee2mqtt/0x00158d00024a5018","json_attributes_topic":"zigbee2mqtt/0x00158d00024a5018","name":"0x00158d00024a5018_water_leak","unique_id":"0x00158d00024a5018_water_leak_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00024a5018","name":"0x00158d00024a5018","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-3 20:28:30 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00024a5018/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00024a5018","json_attributes_topic":"zigbee2mqtt/0x00158d00024a5018","name":"0x00158d00024a5018_battery","unique_id":"0x00158d00024a5018_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00024a5018","name":"0x00158d00024a5018","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-3 20:28:30 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d00024a4e92/water_leak/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.water_leak }}","device_class":"moisture","state_topic":"zigbee2mqtt/0x00158d00024a4e92","json_attributes_topic":"zigbee2mqtt/0x00158d00024a4e92","name":"0x00158d00024a4e92_water_leak","unique_id":"0x00158d00024a4e92_water_leak_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00024a4e92","name":"0x00158d00024a4e92","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-3 20:28:30 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00024a4e92/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00024a4e92","json_attributes_topic":"zigbee2mqtt/0x00158d00024a4e92","name":"0x00158d00024a4e92_battery","unique_id":"0x00158d00024a4e92_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00024a4e92","name":"0x00158d00024a4e92","sw_version":"Zigbee2mqtt 1.1.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-3 20:29:06 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a5018', payload '{"battery":100,"voltage":3015,"linkquality":26,"last_seen":"2019-02-03T16:44:02.730Z"}'
2019-2-3 20:29:06 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a4e92', payload '{"battery":97,"voltage":2995,"linkquality":28,"last_seen":"2019-02-03T16:48:29.006Z"}'
```

Triggers (Every hour the sensor wakes up to signal it is alive):
```
2019-2-4 09:58:34 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3015,"3":23,"4":5032,"5":503,"6":[0,1],"8":516,"10":45761,"100":0}}}' of device 'lumi.sensor_wleak.aq1' (0x00158d00024a5018)
2019-2-4 09:58:34 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a5018', payload '{"battery":100,"voltage":3015,"linkquality":86,"last_seen":"2019-02-04T06:58:34.293Z"}'
2019-2-4 10:01:48 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":28,"4":5032,"5":21,"6":[0,1],"8":516,"10":45761,"100":1}}}' of device 'lumi.sensor_wleak.aq1' (0x00158d00024a4e92)
2019-2-4 10:01:48 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00024a4e92', payload '{"battery":100,"voltage":3005,"linkquality":34,"last_seen":"2019-02-04T07:01:48.947Z"}'
```

The messages I see in the HA log occur exactly when those "I am alive" message are received.

Koenkk, 2019-02-04:
The problem is that the sensor expects the `water_leak` value to be set in the JSON, which is not the case: `{"battery":100,"voltage":3005,"linkquality":34,"last_seen":"2019-02-04T07:01:48.947Z"}`. I would suggest triggering the water sensor once, this will add the missing `water_leak` attribute which will be resend with every message (cached).

milakov, 2019-02-04:
Thanks a lot, this helped!

closed 2019-02-04 by milakov

Bonerserk, 2019-09-04:
> The problem is that the sensor expects the `water_leak` value to be set in the JSON, which is not the case: `{"battery":100,"voltage":3005,"linkquality":34,"last_seen":"2019-02-04T07:01:48.947Z"}`. [...]

I have the same problem with aqara water leak. What do you mean with "triggering the water sensor"?
Thanks

CodeFinder2, 2020-03-19:
@Bonerserk you can push the button on the water leak sensor (on the front where the water symbol is located). This sends its current state (water leakage detected or not) to z2m.

Bonerserk, 2020-03-20:
> @Bonerserk you can push the button on the water leak sensor (on the front where the water symbol is located). This sends its current state (water leakage detected or not) to z2m.

Thanks @CodeFinder2 ! That didn't resolve it! I tried to remove, re-register the sensor but nothing, the water_leak property in the json never appaired!
So I tried to put the sensor in the water and it apeared with true value!
Now water_leak  property always appeares, with false value!
Thanks again!

CodeFinder2, 2020-03-20:
Yes, surprisingly, I had the same issue with that sensor.