Koenkk/zigbee2mqtt #1019: Is it possible to change the MQTT discovery topic?

wsw70, 2019-02-04:
The [documentation ](https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html)mentions MQTT discovery messages:

> `homeassistant/[DEVICE_TYPE]/[DEVICE_ID]/[OBJECT_ID]/config`
> Only used when `homeassistant: true` in `configuration.yaml`. Required for Home Assistant MQTT discovery.

Is it possible to change the autodiscovery root topic (`homeassistant` above) to something else?

The `data/configuration.yaml` example does not have any special setting beside enabling the feature:

```
# Required: Home Assistant integration (MQTT discovery)
homeassistant: false
```

Koenkk, 2019-02-04:
No, as home assistant expects the messages to be send to these topics.

wsw70, 2019-02-04:
Well, this is indeed the default in HA, but can be changed with `discovery_prefix` (https://www.home-assistant.io/docs/mqtt/discovery/).

Since this is not settable, I am closing the question - thanks a lot for the super quick reply.

GreenVine, 2019-03-13:
Hi,

I'm re-opening this feature request because I host multiple Home Assistant instances and connect them with a single MQTT broker. To avoid collision, I prefer to assign each Home Assistant instance with a unique discovery prefix rather than the default `homeassistant` one.

Any chance we could overwrite it?

GreenVine, 2019-03-14:
@Koenkk Thanks for adding it to the next milestone.

For a quick hacky solution, would you mind to tell where the base topic `homeassistant` is defined? I tried to change the last parameter here without success: 
https://github.com/Koenkk/zigbee2mqtt/blob/e5ca977579d57e4eb021e80c01262e44b93c2804/lib/extension/homeassistant.js#L719

Koenkk, 2019-03-14:
That should be it.

GreenVine, 2019-03-14:
> That should be it.

I think I figured it out! In the [Docker documentation](https://www.zigbee2mqtt.io/information/docker.html), `homeassistant.js` is mounted to `/app/node_modules/lib/extension/homeassistant.js`. The correct path should be `/app/lib/extension/homeassistant.js` after I inspected the container data.

I have already submitted a PR to correct this issue: https://github.com/Koenkk/zigbee2mqtt.io/pull/47

Koenkk, 2019-03-15:
Implemented (in the dev branch), docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/configuration/configuration.md

closed 2019-03-15 by Koenkk