Koenkk/zigbee2mqtt #1020: Availability Timeout not working

timstanley1985, 2019-02-04:
Did you read the FAQ?
https://koenkk.github.io/zigbee2mqtt/information/FAQ.md

Unsupported device?
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://koenkk.github.io/zigbee2mqtt/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Hi, firstly this is a great piece of work. It took me 5 minutes with zero issues to get this set up and replace my existing xaiomi hub.  

It has been running flawlessly for one week except for one small point.  I have set availability_timeout to 3600 seconds but after 12 hours without a message from one of my aqara temperature sensors the sensor is still reported as online.  The expected behaviour in this situation would be that the sensor retorts unavailable after 1 hour

Bruceforce, 2019-02-05:
Please see #775
>Device availability is only checked for AC powered routers.

This is not a bug. Since your sensor is battery powered (==end device) it is desired behavior to not show availability status, because this would drain the battery unnecessarily.

Koenkk, 2019-02-05:
Updated documentation and added this as a feature request to #775.

closed 2019-02-05 by Koenkk