Koenkk/zigbee2mqtt #1021: CC2531 not found >> device was not an MTP device

renejorissen, 2019-02-04:
I am having a problem when connecting the CC2531 stick to my PI (Home Assistant). I don't see any ttyACM1 on my PI or any serial device (/dev/serial...). I do see the following error message in /var/log/messages.

> Feb  4 22:50:08 pi-hass kernel: [1904457.202216] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
> Feb  4 22:50:09 pi-hass kernel: [1904457.343959] usb 1-1.2: New USB device found, idVendor=0451, idProduct=16ae
> Feb  4 22:50:09 pi-hass kernel: [1904457.343984] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> Feb  4 22:50:09 pi-hass kernel: [1904457.343995] usb 1-1.2: Product: CC2531 USB Dongle
> Feb  4 22:50:09 pi-hass kernel: [1904457.344006] usb 1-1.2: Manufacturer: Texas Instruments
> Feb  4 22:50:09 pi-hass mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
> Feb  4 22:50:09 pi-hass mtp-probe: bus: 1, device: 6 was not an MTP device

curryhoden, 2019-02-11:
Same probleme here on iobroker, 

Feb 11 13:40:15 ioBroker-RasPi kernel: [ 7392.214705] usb 1-1.4: USB disconnect, device number 24
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.516670] usb 1-1.4: new full-speed USB device number 25 using dwc_otg
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.664394] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.664415] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.664426] usb 1-1.4: Product: TI CC2531 USB CDC
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.664437] usb 1-1.4: Manufacturer: Texas Instruments
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.664447] usb 1-1.4: SerialNumber: __0X00124B001936B43B
Feb 11 13:40:16 ioBroker-RasPi kernel: [ 7392.667178] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Feb 11 13:40:16 ioBroker-RasPi mtp-probe: checking bus 1, device 25: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Feb 11 13:40:16 ioBroker-RasPi mtp-probe: bus: 1, device: 25 was not an MTP device
Feb 11 13:41:17 ioBroker-RasPi kernel: [ 7453.910556] usb 1-1.4: USB disconnect, device number 25
Feb 11 13:41:17 ioBroker-RasPi kernel: [ 7454.216686] usb 1-1.4: new full-speed USB device number 26 using dwc_otg
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.364518] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.364537] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.364548] usb 1-1.4: Product: TI CC2531 USB CDC
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.364559] usb 1-1.4: Manufacturer: Texas Instruments
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.364570] usb 1-1.4: SerialNumber: __0X00124B001936B43B
Feb 11 13:41:18 ioBroker-RasPi kernel: [ 7454.367231] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Feb 11 13:41:18 ioBroker-RasPi mtp-probe: checking bus 1, device 26: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Feb 11 13:41:18 ioBroker-RasPi mtp-probe: bus: 1, device: 26 was not an MTP device

Any updates?

renejorissen, 2019-02-11:
I don't have an update. I flashed the device, but still no luck...... 

I hope somebody has the magic answer.

Koenkk, 2019-02-11:
What's the output of `ls -l /dev/serial/by-id`?

renejorissen, 2019-02-11:
I don't have that directory.

pi@raspberrypi:~$ ls -l /dev/serial
ls: cannot access '/dev/serial': No such file or directory

curryhoden, 2019-02-11:
ls -l /dev/serial/by-id/ shows me :
  "lrwxrwxrwx 1 root root 13 Feb 11 13:24 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936B43B-if00 -> ../../ttyACM0"

Koenkk, 2019-02-11:
@renejorissen that is very strange, AFAIK every Raspbian install should have this. Something fishy is going on.
 
@curryhoden yours look good, what does zigbee-shepherd log on startup?

curryhoden, 2019-02-11:
@Koenkk 
2019-02-10 12:40:25.673 - info: zigbee.0 starting. Version 0.9.1 in /opt/iobroker/node_modules/iobroker.zigbee, node: v8.15.0
2019-02-10 12:40:25.691 - info: zigbee.0 Start on port: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936B43B-if00 with panID 6754 channel 11
2019-02-10 12:40:25.692 - info: zigbee.0 Queue is: true
2019-02-10 12:40:25.842 - info: zigbee.0 Reset coordinator
2019-02-10 12:40:25.897 - info: zigbee.0 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

after 60seconds
2019-02-10 12:41:29.005 - error: zigbee.0 Error while starting zigbee-shepherd!. Error: request timeout

curryhoden, 2019-02-14:
Any ideas here?

Push for hope

closed 2019-04-22 by stale[bot]