Koenkk/zigbee2mqtt #1022: Ikea Tradfri LED1623G12 Won't Repair After Flashing 20190109 Firmware

kylezimmerman, 2019-02-05:
**Bug report:**
- **First make sure your on the latest version of zigbee2mqtt.**
  1.1.1
- **Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.**
20190109 
- **Provide a clear description of the problem.**
After flashing the latest version I have been able to re-pair everything except my Ikea Tradfri LED1623G12 bulbs. 
- **If applicable, provide steps how to reproduce the problem.**
I've followed the reset instructions (On/Off 6 times on for less than off) and every possible variation (same time one/off <1s, >1s,) with the stick <5cm, 5cm, >5cm, literally touching the bulb for over an hour and I cannot get a single log message to appear. The bulb sometimes slightly flickers for a fraction of a second on the 6th retry.
- **Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html**
There is nothing in the logs related to unknown devices or pairing.

JLFN, 2019-02-05:
you need to remove the item from database.

kylezimmerman, 2019-02-06:
I deleted the entire `database.db` and `state.json` files after flashing the stick.

kylezimmerman, 2019-02-06:
Some combination of re-flashing, restarting zigbee2mqtt, restarting my raspberry pi and re-pairing eventually allowed the bulbs to connect.

Hopefully won't have to re-flash for a while!

closed 2019-02-06 by kylezimmerman