Koenkk/zigbee2mqtt #1023: Guide: Virtual environment

orrpan, 2019-02-05:
If you want to run it in virtual environment:

```
# Clone zigbee2mqtt repository
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt

# Enter folder
cd /opt/zigbee2mqtt

# Install python env
python3 -m venv .

# Activate environment 
source /opt/zigbee2mqtt/bin/activate

# Install node environment
pip install nodeenv

# Init node environment
nodeenv -p -n 10.15.1

# Exit and enter environment to be sure
deactivate
source /opt/zigbee2mqtt/bin/activate

# Install dependencies
cd /opt/zigbee2mqtt
npm install

```

Configure and start the zigbee2mqtt as the installation guide (3 and 4) [Running Zigbee2mqtt](https://koenkk.github.io/zigbee2mqtt/getting_started/running_zigbee2mqtt.html) 

Configure systemd
```
# Create a systemctl configuration file for zigbee2mqtt
sudo nano /etc/systemd/system/zigbee2mqtt.service
```
Add the following to this file:
```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/bin/bash -c 'source /opt/zigbee2mqtt/bin/activate; /opt/zigbee2mqtt/bin/npm start'
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

Done.

JLFN, 2019-02-05:
You should make a pull request with this.

orrpan, 2019-02-05:
> You should make a pull request with this.

I'll make one tomorrow

closed 2019-02-06 by orrpan

UJStudio, 2020-12-08:
not working, try following modification:
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
WorkingDirectory=/opt/zigbee2mqtt
ExecStart=/bin/bash -c 'source /opt/zigbee2mqtt/bin/activate; npm start'
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target