Koenkk/zigbee2mqtt #1027: Stability issue with IKEA Tradfri

BadEendt, 2019-02-05:
Hi everybody,

I'm trying to get zigbee2mqtt to work properly, but I'm having serious stability issues.
I'm running the zigbee2mqtt HASS.io (V1.13 and home assistant 0.86.4) add-on version 1.1.1 with a CC2531 flashed with the 2019-01-09 firmware. I got 5 Tradfri 1000lm dimmable bulbs (LED1623G12) , a Trafdri dimmer and a Tradfri outlet hook up.

My problem is that if i try to switch multiple (like 4), zigbee2mqtt drops all connection's and stops functioning. I can fix this by restarting the add-on, re-plugging and resetting the stick, but the problem occurs multiple times a day so that is not a great fix.

I get these error's:
[code]
2019-2-5 19:40:42 - info: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe115554', payload '{"state":"OFF"}'
2019-2-5 19:40:42 - error: Zigbee publish to device '0x000b57fffeec799e', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17
2019-2-5 19:40:43 - info: Zigbee publish to device '0x90fd9ffffe159fe9', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-2-5 19:40:43 - info: Zigbee publish to device '0x000b57fffed83af6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-2-5 19:40:43 - error: Zigbee publish to device '0x90fd9ffffe159fe9', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17
2019-2-5 19:40:43 - error: Zigbee publish to device '0x000b57fffed83af6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 17
2019-2-5 19:40:48 - error: Zigbee publish to device '0xd0cf5efffe6d71a0', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
2019-2-5 19:40:48 - error: Zigbee publish to device '0x000b57fffed83af6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
[/code]

[code]
2019-2-5 19:41:12 - error: Zigbee publish to device '0x000b57fffe115554', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
[/code]
What i tried:
- Reflashing the cc2531 (tried multiple firmware versions)
- Repairing
- Changing the Zigbee channel
- Removing the Tradfri plug from the network (i suspected it was the problem, but it wasn't)
- Running the Edge extension 
 
I hope anyone can help me the solve this issue

Full debug code:
https://pastebin.com/Pa0rjb14

Regards,

Tijmen

BadEendt, 2019-02-05:
Running the latest egde add on i get a few of these errors as well:
`2019-2-5 21:05:36 - error: Zigbee publish to device '0xd0cf5efffe6d71a0', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.
` 
https://pastebin.com/XbDVjdNE

torbjorn2000, 2019-02-06:
I have the exact same problems since some days when I added more Ikea switches to my network. I have 10 lights and 6 switches now and these errors happen several times a day. 

I get the errors with the lights and switches, randomly. Also on a led driver. 

Running older firmware 20181024 but same version of home assistant and zigbee2mqtt.

BadEendt, 2019-02-06:
I already ordered a usb cable so i will try that as soon as it comes in.

What i don't understand is when i was using the ikea bridge i had no problems what so ever and on the network map i don't see any bad connection that isn't reachable via a router?

[image]

Koenkk, 2019-02-06:
@BadEendt network map looks indeed good. Can you try removing the TRADFRI control outlet from the network and check if the issue persists?

BadEendt, 2019-02-06:
I have tried that already. had it set up via bridge and removed it completely but i still had issues. but i dont know how the routing was set up back then so i could try it again if that is what you mean?

Koenkk, 2019-02-06:
Is you tradfri bridge still running?

BadEendt, 2019-02-06:
Nope, i removed all devices and unplugged it.

Koenkk, 2019-02-06:
@BadEendt how often do these errors happen? always or just randomly sometimes?

Koenkk, 2019-02-06:
@BadEendt the latest `-edge` has improved queueing which should make sending multiple commands (e.g. to multiple bulbs), more reliable.

BadEendt, 2019-02-06:
Well that seems to do the trick. I will test it for a couple of days if it keeps working correctly i will close the topic then

BadEendt, 2019-02-06:
And i got a other error....
`  zigbee2mqtt:error 2019-2-6 17:32:03 Zigbee publish to device '0x000b57fffed83af6', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 16
  zigbee2mqtt:error 2019-2-6 17:32:06 Zigbee publish to device '0x000b57fffeec799e', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout`

Koenkk, 2019-02-06:
How many commands in what timespan? Can you post the complete log?

BadEendt, 2019-02-06:
I think about 11 in a second or two, so it might not be a to realist scenario.

But here is the log file, it happend at 17:32:
https://pastebin.com/UietzdD1

torbjorn2000, 2019-02-06:
Hi again, as I said I'm having the same problem and is testing at home right now. Bought an USB extension cable now and keeping the USB thingie 1m from the raspberry pie. 

What I have experienced is also when sending many commands at once. I have scenarios for all lights on and off.  I have Ikea bulbs, outlets and led driver, a total of 22 devices that I turn on and off at once.

One error is that I get a lot of 205, 183 and 16 errors when bursting command. When trying after a few seconds with the failed ones, it works fine. 

Another error is that the devices goes completely "offline" at the burst of commands. The requests times out and it stays like that (`genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout`). Turning device off/on manually (cut power) does not remedy situation, but restarting zigbee2mqtt (and I had to reset USB stick as well, quite common now) made them work again.

Sending logs for the first problem ([log1.log](https://www.dropbox.com/s/2n4rp3pboovb1ku/log1.log?dl=0)) and the second ([log2.log](https://www.dropbox.com/s/25kfmoroe9aa3ca/log2.log?dl=0)).

I have a theory I will test: I'm always sending brightness and color temp, to all devices. Some of them complain that they cannot handle it, of course. Maybe that puts them in a state of error?

torbjorn2000, 2019-02-06:
Another thing my network graph look like a big wheel, where everything is connected to everything. The coordinator has at least 18 devices with arrows to it. Maybe it is a problem with the limit of 15, but since almost all my devices are routers, I would think it would just connect to them instead if coordinator gets full, even if it can reach it.

[image]
[image]

(couln't get it more high res, but I zoomed in on coordinator)

BadEendt, 2019-02-06:
sounds like we have the exact same problem. Are you running the latest edge version? 
That seems to help a bit in my situation, though it isn't completely solved

torbjorn2000, 2019-02-06:
No I'm running the stable version because of "stable"... ;)

Koenkk, 2019-02-07:
If you want to control a lot of bulbs at the same time, I would recommend using groups: https://koenkk.github.io/zigbee2mqtt/information/groups.html

trekker25, 2019-02-07:
i also recommend using groups.

I have 11 GU10 tradfri and 4 bulbs split in 4 groups.

I paired the devices with the old firmware with no group support. I tried to switch a group in HA (so not a group in zigbee2mqtt) with 8 spots. Zigbee Network crashed/stopped instant with recovering after a while.

I had a bad feeling about this as i built a new kitchen with 4 fysical switches out of sight (hidden in a cupboard) and placed 4 zigbee switches. I was horrified to replace the fysical switches back on a more normal location.

But the group support changed everything, always working perfectly for about more as a week stable. Also made sure: long usb cable on CC2531. And added more HUE bulbs to create a better path.

torbjorn2000, 2019-02-07:
Yes groups seems like a good solution. Was going there anyway just to make the lights go on and off simultaneously. But I have to reflash the stick and repair everything... :(

Will try with the delay as well!

Thanks for all help, and development of this! You are really doing a great job!

trekker25, 2019-02-07:
> Yes groups seems like a good solution. Was going there anyway just to make the lights go on and off simultaneously. But I have to reflash the stick and repair everything... :(
> 
> Will try with the delay as well!
> 
> Thanks for all help, and development of this! You are really doing a great job!

repair is not needed. Flash the new firmware with "retain address"

take your stick and PC/NUC/Pi to the attic, wrap the stick in allumium foil and boot up and make sure zigbee2mqtt starts running.  This prevents that the new flashed stick get's a different address.

(maybe a network cable) so you can check the logging of zigbee2mqtt. If in the logs you see the new firmware date (2019xxxxx) you can shutdown PC/Nuc/Pi.

Place back in normal positon and unwrap foil and boot up. Now all devices will reconnect without repair.

Same can also be done removing buls/routers from power (and maybe remove battery from sensors, not sure on that), both up with new firmware and after a few minutes give back power to your bulbs/routers/sensors.

torbjorn2000, 2019-02-07:
> repair is not needed. Flash the new firmware with "retain address"
>
> take your stick and PC/NUC/Pi to the attic, wrap the stick in allumium foil and boot up and make sure zigbee2mqtt starts running. [...]

Oh that's new! So if I understand correctly:

1. Flash stick normally
2. Start up zigbee2mqtt with flashed stick shielded from other Zigbee devices (with foil)
3. Verify the correct version of stick from the logs
4. Shut down and startup without the shielding

Is the crucial step that it is started up with new firmware without being able to see other Zigbee devices/networks?

And by "Flash the new firmware with retain address", you just mean the process, there is no flag when flashing?

Bruceforce, 2019-02-07:
Your first step should be the shielding of the stick I think. After that you can flash the stick and you are done.

>Is the crucial step that it is started up with new firmware without being able to see other Zigbee devices/networks?

Yes. Since I only have 1 battery powered device I can just switch off the fuse + remove the battery, so all devices are offline. Flash the stick with a notebook and achieve the same result (as suggested above)

torbjorn2000, 2019-02-07:
Good! I will try first with the delay just to see if it improves, then do the flashing. Will come back with report!

torbjorn2000, 2019-02-07:
> If groups is not a solution, you could try increasing this delay: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.js#L30 to e.g. ` 500` (let me know if that solves the problems).

I'm running zigbee2mqtt as an addon in hassio, so I don't really know where to find the zigbee.js file...

Bruceforce, 2019-02-07:
I don't know how exactly the hassio addons work but they seem to be simple docker containers.

So you probably could
```
sudo docker ps
```
to find out the container name (probably something with "zigbee2mqtt")

then
```
sudo docker exec -it <NAME_OF_CONTAINER> /bin/sh/
vi /app/lib/zigbee.js
#Change the value to 500 in line 30 and save
exit
sudo docker restart <NAME_OF_CONTAINER>
```

torbjorn2000, 2019-02-09:
I cannot access this from within Hassio... At least I think. I have no idea in what container I end up in when I SSH in.. Is it the underlaying OS or a Docker container? Feels like the Inception movie.. :D 

Will flash the stick tomorrow!

torbjorn2000, 2019-02-13:
A bit of an update:

- I flashed stick with latest 20190109 firmware. Tin foil didn't help, it could still control devices while wrapped in several layers. Maybe the unwrapped USB extension cord acted as antenna? Anyway, had to re-pair everything.
- After a stable 5 minutes, everything started to mess with me. Devices stopped taking commands (timeouts and missing acks etc). Had to restart+replug all the time. The 60 s "fix shephard" occured every start. After 5-6 restarts, it would no longer start up at all. Could not fix.
- I reflashed stick again. Paired only bulbs (IKEA) with stick and let the outlets be on Trådfri router. I suspected the outlets to be not super compatible.

Current status is that if I use groups to turn off/on the lights it works like a charm! But if I turn off the 10-15 bulbs one by one in a burts (from a group in HA), some lights will stop responding and a restart+replug is required. It seems that the burst of commands takes down the Zigbee network until restart.

One bulb is faulty I think, it may stop responding by random. Restart of bulb fixes that.

Conclusion: Groups are the solution to a large Ikea Trådfri network! Sending burst of commands may kill network requiring restart.

StyxyDog, 2019-02-14:
@torbjorn2000 I have the same problem, but instead of using groups, my automatons are in Node-RED and the service calls can be rate limited- I send one per second. 
In terms of repairing the Ikea bulbs, it is very time consuming to unsrew them all and take to the co-ordiantor. I have found using a second CC2531 configured as a router and powered up with a battery pack means I can re-pair without having to remove from the light fitting. I just hang the router from the lamp shade and flick the switch to reset the bulb.
[image]

torbjorn2000, 2019-02-14:
> @torbjorn2000 I have the same problem, but instead of using groups, my automatons are in Node-RED and the service calls can be rate limited- I send one per second.

Nice solution! Although it's a nice effect when all the lights turn on or off at the same time. But using groups in ZigBee, you have to have one group for every scenario you want to turn on or off.

> In terms of repairing the Ikea bulbs, it is very time consuming to unsrew them all and take to the co-ordiantor. I have found using a second CC2531 configured as a router and powered up with a [...]

That's for re-pairing I assume. My bulbs might end up in a state where they require restart, just by flicking the switch off and on again. But speaking of re-pairing, it often work with the bulb quite far away strangely enough. 

> [image]

Bruceforce, 2019-02-15:
> I have the same problem, but instead of using groups, my automatons are in Node-RED and the service calls can be rate limited- I send one per second.

If that works for you it should also work, if you increase the delay in https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.js#L30 as mentioned above. This would also help other users if you can find a good value and open a PR, since node red is not always applicable (for example if you want to control the devices via Alexa oder Google Assistant).

dyrkin, 2019-02-18:
I'm not sure that I'm right and understand everything correctly, but I think it is better not to increase the delay, but to change the logic in the zigbee-shepherd.

1. Add a queue for AfDataRequests
2. Process these request one by one.

**How to understand that the request is completed?**
Most global cluster commands have its own responses, for example ReadAttributesCommand have ReadAttributesResponse.
For most cluster local commands we can enable DefaultResponse, for example OnCommand doesn't have its own response, but device can respond with DefaultResponse when command execution is completed.

So, the strategy could be like this:
1. Take AfDataRequest from the queue.
2. Generate transaction ID and send this request.
3. Wait for AfDataConfirm with the same transaction id (it's is already implemented in the zigbee-shepherd)
4. 
   - For global commands with own responses. Wait for AfIncomingMessage with the same transaction id and extract command specific response
   - For local commands without responses. Wait for AfIncomingMessage with the same transaction id and extract DefaultResponse
5. Go to step one

Does this make any sense?

Koenkk, 2019-02-19:
@dyrkin this was something that was previously implemented, but as a `AfDataRequest` could fail or have a delay, all commands executed after that get delayed as well. Perhaps a per device queue could be a solution?

@BadEendt could you check if this is still an issue at all? Please try with the latest dev branch and the latest dev firmware (`20190218`)

BadEendt, 2019-02-20:
Alright, I will try the latest version. Is there a way to force the latest branch when running the edge extension or do I need to reinstall the plugin?

On a side note, last night the CC2531 crashed after an overload of commands (fist time in weeks), but then i had to re-flash it bring it back online instead of just re-plugging and resetting. I don't know if this issue is related though.

Koenkk, 2019-02-20:
Reinstalling should be enough I think.

BadEendt, 2019-02-20:
The problem still occurs. If i switch to many lights on and off a couple of times. It seems better in more "normal" use, the last couple of weeks.

https://pastebin.com/1RgAbq8k

Koenkk, 2019-02-20:
I could indeed reproduce this by turning on 11 TRADFRI GU10 spots at the same time. This somehow causes a 'DDOS' on the network.

For future reference, I think the behaviour of the scheduler should be as follows:
- Between each Zigbee command there is a delay of 200ms
- Only one zigbee command per device can be active at a time (a command is active when zigbee2mqtt hasn't received a response/timeout for the command)
- A maximum of 3 commands are active at the same time.

We should refine the above values of 3 and 200ms to check what the real boundaries are, I guess this also differs per setup.

andreasbrett, 2019-02-21:
I have the same issue for some weeks now. Let me know if I can help you out debugging. My network consists of 18 routers, 15 of them are TRADFRI bulbs (5x E27, 10x GU10). Those GU10s are harder to re-pair so there were times where I only paired the E27s to my network and that worked completely stable. Once I added the GU10s it became unstable again. Not sure though if the GU10s are causing issues (they are smaller and have less space for electronics; also their pairing process takes longer than the E27s) or if it's just because the network group quite a lot bigger. Those GU10s are also part of 2 lamps so switching those lamps will send a lot of zigbee messages as I'm not yet using groups (will look into that).

Koenkk, 2019-02-22:
Please try again with the latest dev branch and latest dev firmware (`20190222`) (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin)

With this, I achieved the following results:

I have a setup with a zigbee network of 40+ nodes. In my test, I controlled 9 different TRADFRI GU10 bulbs. I've made a script (MQTT) which turns these bulbs on and off (= 18 commands per run). I've executed this script 20 times without a delay (resulting in a burst of 360 MQTT commands which results in 360 zigbee commands). 

This ran for a few minutes, but eventually finished without a crash of the CC2531 or Zigbee2mqtt. The only errors I encountered were `205 no network route` errors, but these were sporadically, the same bulb would respond to the next on/off command (no network hangups). After this all finished, I could you all bulbs as normal.

Let me know about your results.

martinrosenauer, 2019-02-22:
@Koenkk, thanks, just flashed now and hope it'll be more stable on 20190222 firmware. Was on 20190218 which I flashed early this morning and unfortunately it stopped responding late afternoon again. 

I've reset the stick and only see some 205s now which I guess is acceptable considered your message. 

Is there any way we can detect a crash and have a notification from Home Assistant ?

Koenkk, 2019-02-22:
@martinrosenauer zigbee2mqtt will report as offline (all bulbs should be unavailable in home assistant).

martinrosenauer, 2019-02-22:
@Koenkk, ok - but in the scenario from this afternoon (https://pastebin.com/raw/QapGCYxt) it looks like some messages got through, however most of the bulbs didn't respond. (but zigbee2mqtt stayed up).

Koenkk, 2019-02-22:
@martinrosenauer could you try with https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/util/zigbeeQueue.js#L2 increasing that to 500?

martinrosenauer, 2019-02-22:
@Koenkk, trying - increased to 500 and restarted zigbee2mqtt on 
zigbee2mqtt version 1.1.1 (commit #1fda02a)
Coordinator firmware version: '20190222'

Will let you know if it's stable :) (probably won't know until tomorrow or the day after).

martinrosenauer, 2019-02-22:
@Koenkk, IKEA remote still works as expected, which is brilliant, however - how do I reflect the state of a light I control with it in HA ? (e.g. I turn it on by 'toggle' on the remote which is bound directly to a light - I can see the event in zigbee2mqtt but HA is obviously not aware of the change).

martinrosenauer, 2019-02-22:
@Koenkk, sorry to bring bad news - looks like it's still not working on 20190222 / #1fda02a and with 500 in delay. If you look at https://pastebin.com/raw/rKfkRCJd the light "traadfri_bryggers_lampe" (0x0017880110570b58) became unresponsive about an hour after starting up zigbee2mqtt. The light is physically right next to the CC2531.

djchen, 2019-02-23:
I've been having this issue for the past 2 days on 20190218

`[1905343.035202] usb 1-1: USB disconnect, device number 10
[1905354.045115] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[1905354.198865] usb 1-2: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[1905354.198870] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1905354.198873] usb 1-2: Product: TI CC2531 USB CDC
[1905354.198876] usb 1-2: Manufacturer: Texas Instruments
[1905354.198878] usb 1-2: SerialNumber: __0X00124B0019366410
[1905354.200609] cdc_acm 1-2:1.0: ttyACM0: USB ACM device`

Koenkk, 2019-02-23:
How big are your networks?

djchen, 2019-02-23:
3x Tradfri LED1622G12, 2x Tradfri Control Outlet, 1x Tradfri Dimmer, 1x Tradfri Remote, 3x Sengled E11-G13, 1x GE 45857GE, 1x Xiaomi WXKG01LM

djchen, 2019-02-23:
I downgraded to 20190109 and removed the Tradfri devices, so far no issues.
3x Sengled E11-G13, 1x GE 45857GE

Will try this out for a week and then try adding a couple Tradfri devices back in.

Koenkk, 2019-02-23:
So I've did some more testing today. I think in the older firmwares, the cached routes back to devices are not used properly (due to being expired). I've disabled the route expiry and this seems to improve things, hopefully this doesn't cause some nasty side effects. I've also bumped the routes to remember to 40. Meaning that the CC2531 now remembers routes to 40 routers.

New firmware can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin (`20190223`).

BEFORE bursting your network again, make sure that the CC2531 has a route to the device, otherwise you will get a lot of `205 no network route` errors again. It seems that during these bursts of command, the routes cannot be determined. 

To make sure the CC2531 has a route to each device, send a command (which has to succeed) to every device once (e.g. on/off).

martinrosenauer, 2019-02-24:
@koenkk, now upgraded firmware to latest (20190223) and ran update.sh to get the last updates from the dev branch. It seems stable except I have a single Hue that is now not responding. Tried turning off it's power and back on but same result. 

Pasted log on https://pastebin.com/raw/BLbqrWqc.

Anything you want me to try ? =) or any suggestions ? (delete and readd may be my next thought)

Thanks!

Koenkk, 2019-02-24:
@martinrosenauer just re-pair it

martinrosenauer, 2019-02-24:
@koenkk, already tried, giving the same result. Will try re-pair.

djchen, 2019-02-26:
> I downgraded to 20190109 and removed the Tradfri devices, so far no issues.
3x Sengled E11-G13, 1x GE 45857GE

Been running this without issue for 3 days. Issue with TRADFRI devices or was 20190218 broken?

Koenkk, 2019-02-26:
@djchen can you try with tradfri and `20190223`?

djchen, 2019-02-28:
I ran into an issue today where my GE 45857GE dropped out of the network for a period of time.

```
2/28/2019, 5:45:10 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:10 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:10 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:45:10 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:10 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:10 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:45:16 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:16 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:16 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:45:16 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:16 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:16 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:45:23 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:24 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:24 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:45:36 AM - info: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/28/2019, 5:45:36 AM - error: Zigbee publish to device '0x0..76', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2/28/2019, 5:45:36 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x0..76","type":"device","friendlyName":"Kitchen Lights"},"message":"{\"state\": \"OFF\"}"}}'
2/28/2019, 5:47:14 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":21}'
2/28/2019, 6:03:56 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":10}'
2/28/2019, 6:20:37 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":0}'
2/28/2019, 6:37:18 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":2}'
2/28/2019, 6:53:58 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":26}'
2/28/2019, 7:10:39 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":23}'
2/28/2019, 7:27:20 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":21}'
2/28/2019, 7:41:31 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"ON","linkquality":31}'
2/28/2019, 7:42:32 AM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights', payload '{"state":"OFF","linkquality":5}'
```

Maybe interference caused it to lose connection with the coordinator?
I'll try to pair a IKEA Tradfri Control Outlet in between to boost the signal.

BadEendt, 2019-03-01:
Hi, I'm using 20190223 now for a couple of days and it seems stable. No need for resetting and replugging. 
What i am noticing is when i issue multiple commands the response seems a bit slower and my Gledopto bulb sometimes causes error's but it will start working again after a minute or so.

[code]
3/1/2019, 10:47:26 AM - info: Zigbee publish to device '0x00124b001b503afa', genLevelCtrl - moveToLevelWithOnOff - {"level":127,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
3/1/2019, 10:47:27 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00124b001b503afa', payload '{"state":"ON","color_temp":370,"color":{"x":0.323,"y":0.329},"brightness":80}'
3/1/2019, 10:47:27 AM - info: Zigbee publish to device '0x00124b001b503afa', lightingColorCtrl - moveToColorTemp - {"colortemp":370,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
3/1/2019, 10:47:27 AM - error: Zigbee publish to device '0x00124b001b503afa', lightingColorCtrl - moveToColorTemp - {"colortemp":370,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11 failed with error Error: AF data request fails, status code: 233. MAC no ack.
3/1/2019, 10:47:27 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x00124b001b503afa","type":"device","friendlyName":"0x00124b001b503afa"},"message":"{\"state\": \"ON\", \"brightness\": 127, \"color_temp\": 370}"}}'
3/1/2019, 10:47:27 AM - info: Zigbee publish to device '0x00124b001b503afa', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - 11
[/code]

Full log:
https://pastebin.com/LT7Q6GNk

DeliriousMetro, 2019-03-11:
I am now having issues since updating the co-ordinator to firmware 20190223 from a release in January (logs deleted so can't tell which revision).

Phillips Hue bulbs connect to the stick and show the correct states, but changing the brightness or toggling the switch causes the issues mentioned here:

```
3/11/2019, 10:29:31 AM - debug: Received MQTT message on 'zigbee2mqtt/Phillips Hue (Hallway)/set' with data '{"state": "OFF"}'

3/11/2019, 10:29:31 AM - error: Zigbee publish to device '0x00178801102985d2', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.

3/11/2019, 10:29:31 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00178801102985d2","type":"device","friendlyName":"Phillips Hue (Stairs)"},"message":"{\"state\": \"OFF\"}"}}'
```

Koenkk, 2019-03-11:
@DeliriousMetro please try to re-pair the problematic device.

[22 more replies]

closed 2019-06-05 by stale[bot]