Koenkk/zigbee2mqtt #1032: Adding Yale Lock on Hass.io

CanuckianOz, 2019-02-06:
I'm having trouble setting up my Yale lock with Hass.io and zigbee2mqtt. I can pair the lock, but I need to add it as an unsupported device. I get up to the point of trying to run a test function and trigger some functions, such as locking and unlocking the door but nothing happens. The guide says I need to add a "configure:" section to the devices.js but I can't find any similar devices to base it on, and there's no errors or messages in the debug log to work through it. I'm stuck without knowing what's going wrong. Any ideas to work through it?

https://hastebin.com/ejopevunil.makefile

matisaul, 2019-02-06:
Hi !! I have the YMF40 lock from Yale. I could pair the lock too, but now I'm stuck on what doing for have some comunication with the lock. Could you tell me your steps to get that connection info? Thanks!

Koenkk, 2019-02-06:
@CanuckianOz we need to find out what to send, do you still have another gateway where the device worked with? We can use zigbee sniffing to find out what to send: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_sniff_zigbee_traffic.html

CanuckianOz, 2019-02-06:
@Koenkk I unfortunately don’t. I specifically went with this method to avoid spending a lot on another hub, but also because most of the hub options aren’t available in Australia. ZWave is on a different frequency here so most hubs are half useless. 

I do have a Xiaomi 1st gen gateway, but the lock only says “pairing successful”. It does not provide any confirmation on the hub side.

matisaul, 2019-02-06:
There is a github project for Smartthings hub... I think It's the same hub type that Yale uses.

(https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-lock.src/zigbee-lock.groovy)

I think that codes are:

```
private getDOORLOCK_CMD_LOCK_DOOR() { 0x00 }
private getDOORLOCK_CMD_UNLOCK_DOOR() { 0x01 }
private getDOORLOCK_CMD_USER_CODE_SET() { 0x05 }
private getDOORLOCK_CMD_USER_CODE_GET() { 0x06 }
private getDOORLOCK_CMD_CLEAR_USER_CODE() { 0x07 }
private getDOORLOCK_RESPONSE_OPERATION_EVENT() { 0x20 }
private getDOORLOCK_RESPONSE_PROGRAMMING_EVENT() { 0x21 }
private getPOWER_ATTR_BATTERY_PERCENTAGE_REMAINING() { 0x0021 }
private getDOORLOCK_ATTR_LOCKSTATE() { 0x0000 }
private getDOORLOCK_ATTR_NUM_PIN_USERS() { 0x0012 }
private getDOORLOCK_ATTR_MAX_PIN_LENGTH() { 0x0017 }
private getDOORLOCK_ATTR_MIN_PIN_LENGTH() { 0x0018 }
private getDOORLOCK_ATTR_SEND_PIN_OTA() { 0x0032 }
private getALARM_ATTR_ALARM_COUNT() { 0x0000 }
private getALARM_CMD_ALARM() { 0x00 }

```

This code are for specific locks model (Yale Touch Screen Lever Lock, Yale Push Button Deadbolt Lock,Yale Touch Screen Deadbolt Lock, Yale Push Button Lever Lock), but maybe works for all models.

CanuckianOz, 2019-02-07:
@matisaul That’s an excellent start, thank you. My lock is listed in that document. My initial thought is after having a 10-15min glance at the code is that I don’t know or understand the devices.js or zigbee-shepherd-converter structure and elements yet to translate it. I’ll dive in and see where I get stuck and ask some questions here. I think I’ve got line of sight to at least get it responding and start troubleshooting...

CanuckianOz, 2019-02-07:
@Koenkk sorry I'm not a developer and it's been a while since university programming. Can you help point me in the right direction per the below?

From @matisaul, we have these resources to help:

`private getDOORLOCK_CMD_LOCK_DOOR() { 0x00 }
private getDOORLOCK_CMD_UNLOCK_DOOR() { 0x01 }
private getDOORLOCK_CMD_USER_CODE_SET() { 0x05 }
private getDOORLOCK_CMD_USER_CODE_GET() { 0x06 }
private getDOORLOCK_CMD_CLEAR_USER_CODE() { 0x07 }
private getDOORLOCK_RESPONSE_OPERATION_EVENT() { 0x20 }
private getDOORLOCK_RESPONSE_PROGRAMMING_EVENT() { 0x21 }
private getPOWER_ATTR_BATTERY_PERCENTAGE_REMAINING() { 0x0021 }
private getDOORLOCK_ATTR_LOCKSTATE() { 0x0000 }
private getDOORLOCK_ATTR_NUM_PIN_USERS() { 0x0012 }
private getDOORLOCK_ATTR_MAX_PIN_LENGTH() { 0x0017 }
private getDOORLOCK_ATTR_MIN_PIN_LENGTH() { 0x0018 }
private getDOORLOCK_ATTR_SEND_PIN_OTA() { 0x0032 }
private getALARM_ATTR_ALARM_COUNT() { 0x0000 }
private getALARM_CMD_ALARM() { 0x00 }`

`fingerprint profileId: "0104", inClusters: "0000,0001,0003,0009,000A,0101,0020,0B05", outClusters: "000A,0019", manufacturer: "Yale", model: "YRD446 BLE TSDB", deviceJoinName: "Yale Assure Lock"`

Maybe these can be inserted into devices.js? Here's where I'm up to:
`       zigbeeModel: ['YRD446 BLE TSDB'],
        model: 'YRD426NRSC', 
        vendor: 'Yale', 
        description: 'Yale Assure Keyed Electronic Digital Deadbolt', 
        supports: 'locking and status', 
        fromZigbee: [], 
        toZigbee: [], 

All good up to here. I know I need to add a configure section:
`          configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);`
- what's ieeeAddr, shepherd, coordinator, callback? Some have all 4 variables in the function, others have 3. 
`              const cfg = {
                direction: 0, attrId: 33, dataType: 32, minRepIntval: 0, maxRepIntval: repInterval.MAX, repChange: 0,
            };`
- what is "direction: 0, attrId: 33, dataType: 32, minRepIntval: 0, maxRepIntval: repInterval.MAX, repChange: 0" and how do I determine these values?
`               (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.foundation('genPowerCfg', 'configReport', [cfg], foundationCfg, cb),`
- what's the difference between bind, foundation and report?

matisaul, 2019-02-08:
I found a interesting post about a smartthings hub connected to a Yale lock and HA.

https://image.ibb.co/gSXzPa/Front-Door-Lock-ST.jpg

On that picture, there is a log from ST talking with a yale lock.

Koenkk, 2019-02-08:
Let's do this in 2 steps:
1.  setup reporting for the device, this should allow us to have the unlock/locked status and battery percentage. 
2. allow to send lock/unlock commands.

For step 1.

```js
{
        zigbeeModel: ['YRD446 BLE TSDB'],
        model: 'YRD426NRSC', 
        vendor: 'Yale', 
        description: 'Yale Assure Keyed Electronic Digital Deadbolt', 
        supports: 'lock/unlock, battery', 
        fromZigbee: [], 
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 3, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

- Please check that on startup of zigbee2mqtt, `Device successfully configured..` is logged for this device, if not, try again and wakeup the device before starting zigbee2mqtt (by pressing a few buttons on it). 
- This should spam us (each 3 seconds), with the locked and battery state. (should produce `No converter available.` messages). Don't worry about the spam, we will fix this later (it's for debugging).
- If everything above is OK, when unlocking and locking, is the new state reported?

CanuckianOz, 2019-02-08:
@Koenkk yep! Device configured and now getting device configured and status changes. Lock status changes when I turn the lock (2 is unlocked, 1 is locked). Battery status is coming in but obviously it isn't changing.

Here's the log.

https://hastebin.com/asovozeyey.js

Koenkk, 2019-02-08:
@CanuckianOz is this a full battery?

CanuckianOz, 2019-02-08:
Sorry I wasn't clear. 
- It changes to lockstate 2 after I unlock. 
- changes to lockstate 1 after I lock
- I just changed to new batteries and the state stays at 200. We've had the lock for a couple months and should last for over a year.

Koenkk, 2019-02-09:
@CanuckianOz good, i've reduced the reporting interval, so spamming should be over, let's try to lock/unlock it.

Use the following files:
- `devices.js`: https://github.com/Koenkk/zigbee-shepherd-converters/blob/YRD426NRSC/devices.js
- `fromZigbee.js`: https://github.com/Koenkk/zigbee-shepherd-converters/blob/YRD426NRSC/converters/fromZigbee.js
- `toZigbee.js`: https://github.com/Koenkk/zigbee-shepherd-converters/blob/YRD426NRSC/converters/toZigbee.js

And send to `zigbee2mqtt/0x000d6f0010f9f93f/set` `lock` or `unlock`.

If that doens't unlock/lock the lock, try to add your pin here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/YRD426NRSC/converters/toZigbee.js#L1127 (e.g. 'pincodevalue': '1234',

matisaul, 2019-02-09:
> @CanuckianOz good, i've reduced the reporting interval, so spamming should be over, let's try to lock/unlock it.
>
> Use the following files:
>
> * `devices.js`: [...]

Hi @Koenkk ,

I'm using the Hassio add-on... To get this config working, I need to copy this 3 files into "/share/zigbee2mqtt/" folder? And use "zigbee-shepard-devices" : "true" option? Or I need to do something else?

matisaul, 2019-02-09:
When I try to add a device.js, I get this error:

``` zigbee2mqtt:info 2019-2-8 22:52:18 Logging to directory: '/share/zigbee2mqtt/log/2019-02-08.22-52-18'
  zigbee2mqtt:debug 2019-2-8 22:52:19 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
/zigbee2mqtt-1.1.1/node_modules/zigbee-shepherd-converters/devices.js:3
        model: 'YRD426NRSC', 
             ^
SyntaxError: Unexpected token :
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:617:28)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/zigbee2mqtt-1.1.1/node_modules/zigbee-shepherd-converters/index.js:3:17)
npm
 ERR! code
 ELIFECYCLE
npm 
ERR! errno 1
npm
```

And this is my device.js:

```
   {
        zigbeeModel: ['iZBModule01'],
        model: 'YRD426NRSC', 
        vendor: 'Yale', 
        description: 'Yale Assure Keyed Electronic Digital Deadbolt', 
        supports: 'lock/unlock, battery', 
        fromZigbee: [], 
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 3, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];
            execute(device, actions, callback);
        },
    }
```

matisaul, 2019-02-09:
Please, help me configuring this :(

I was looking into database.db, and I found this;

`{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00118c1895","nwkAddr":51588,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"iZBModule01","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genTime":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"closuresDoorLock":{"dir":{"value":1},"attrs":{}}}}},"_id":"14rCvKxH6MxiqK1Z"}
`

can I configure the lock with this data??

Koenkk, 2019-02-09:
You should replace the files with the ones from: https://github.com/Koenkk/zigbee2mqtt/issues/1032#issuecomment-461991343

Files can be found under `node_modules/zigbee-shepherd-converters`.

CanuckianOz, 2019-02-09:
@matisaul note, you're running on Hassio so you need to access the docker to change the toZigbee.js and fromZigbee.js in node_modules/zigbee-shepherd-converters to test. To be honest, it's a PITA unless you're comfortable with linux and the VI editor... I had to fumble around learning the commands, but I'm also just a hobbyist so you might find this easier than me. 

https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices_on_hassio.html

OR, I've just finished changing the files per @Koenkk 's changes so if it works we can have it included in the next update, but I don't know how often changes are released or quality control reqs to do that.

CanuckianOz, 2019-02-09:
@Koenkk I'm getting the following error when starting zigbee2mqtt:

    2019-02-09T23:22:03.335Z zigbee-shepherd-converters:devices Configured '(cb) => device.report('closuresDoorLock', 'lockState', 0, repInterval.HOUR, 0, cb)' with result 'Error: request unsuccess: 16'
  zigbee2mqtt:error 2019-2-10 09:22:03 Failed to configure 0x000d6f0010f9f93f (0x000d6f0010f9f93f) ('Error: request unsuccess: 16')

CanuckianOz, 2019-02-10:
@Koenkk Okay, so I've given up on trying to change in the Hassio host and zigbee2mqtt docker. Way too much effort and the changes were lost hours later. 

Instead, and I know this breaks good structure, I took your functions in fromZigbee.js and toZigbee.js and merged them into devices.js with the zigbee2mqtt hassio addon configured to refer to devices.js in shared/zigbee2mqtt folder. That works, and I can get the lock status and no errors in the log. 

However, it's not responding to lock commands. I tried adding the PIN and nothing happens. Here's what the log says. Interesting that it says "offline", when I'm clearly getting lock & unlock status updates. 

      zigbee2mqtt:debug 2019-2-10 20:22:07 Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":198}}' of device 'YRD446 BLE TSDB' (0x000d6f0010f9f93f)
      zigbee2mqtt:info 2019-2-10 20:22:07 MQTT publish: topic 'zigbee2mqtt/0x000d6f0010f9f93f', payload '{"battery":99,"linkquality":113,"device": 
 '{"ieeeAddr":"0x000d6f0010f9f93f","friendlyName":"0x000d6f0010f9f93f","type":"EndDevice","nwkAddr":62892,"manufId":4125,"manufName":"Yale","powerSource":"Battery","modelId":"YRD446 BLE TSDB","status":"offline"},"state":"UNLOCK"}'

Koenkk, 2019-02-10:
- Don't worry to much about the online/offline state (it's not working properly yet)
- Support for auto discovery can be added once this device is supported
- Can you post the log (with `log_level: debug`) when you try to unlock the device? https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html

CanuckianOz, 2019-02-10:
@Koenkk Boom! It works now. 

Thanks - I didn't diligently check the log. Because I moved the fromZigbee and toZigbee converters, I also had to include some dependencies in the devices.js. Specifically zclid and cfg. 

So it physically works. I can make mqtt service calls to integrate it into my home assistant no problem. One item to note - it doesn't require the PIN to operate. I'm not sure if it's possible to force this requirement, but I will include my own authentication setup. I guess the next step is integrating for device support? 

@matisaul if you aren't able to get yours working by my cryptic posts, let me know and I'll share my code and walk you through how to get it working.

matisaul, 2019-02-11:
> @Koenkk Boom! It works now.
>
> Thanks - I didn't diligently check the log. Because I moved the fromZigbee and toZigbee converters, I also had to include some dependencies in the devices.js. [...]

Wow, that would be great! Thanks! Please, if you could share files, and the path for using with hassio add-on! Thanks again !!

Koenkk, 2019-02-11:
Great, I will soon integrate this into the dev branch (with Home Assistant auto discovery support).

Koenkk, 2019-02-11:
Supported in the dev brach now! (with HA discovery support). For hassio use `-edge`.

matisaul, 2019-02-11:
@CanuckianOz Sorry for bother you, but could you share with me your code? Thanks again!

CanuckianOz, 2019-02-12:
@matisaul You can try with the hassio addon dev version zigbee2mqtt-edge from @Koenkk or use the below. 

Firstly, add the below to the zigbee2mqtt addon settings in hassio so it defaults to the configuration.yaml. I had issues with it allowing permit_join in the addon settings but it worked once I defaulted to the zigbee2mqtt configuration.yaml. 

    {
      "data_path": "/share/zigbee2mqtt",
      "devices": [],
      "network_key": [],
      "zigbee_shepherd_devices": true,
      "zigbee_shepherd_debug": true,
    }

Then write this into the configuration.yaml in share/zigbee2mqtt
https://hastebin.com/supiqobeju.coffeescript

Restart and make sure everything works. There should be a devices.js in the share/zigbee2mqtt folder. Replace the entire file with the below. I've just added the Yale lock functions and const cfg and const zclid definitions from the top of the tozigbee.js. 

https://hastebin.com/masejuzaxo.coffeescript

matisaul, 2019-02-12:
Wow, thanks @CanuckianOz  !! I'll test it tonight.

I forgot to ask you... Do you have this zigbee module?:

https://www.paris.cl/dw/image/v2/BCHW_PRD/on/demandware.static/-/Sites-cencosud-master-catalog/default/dw2e62a2ad/images/imagenes-productos/742/247579-0000-001.jpg?sw=1480&sh=2000&sm=fit

Just to know if we are talking about the same HW,

matisaul, 2019-02-12:
It works!!! I can Lock/Unlock the door, but sometimes I get some errors, and I don't know how to read the actual state of the lock:

```
2019-02-12T23:09:02.945Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:02.951Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:02.952Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:02.954Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:02.954Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2019-02-12T23:09:02.954Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 53
  zigbee2mqtt:info 2019-2-12 20:09:02 MQTT publish: topic 'zigbee2mqtt/0x000d6f00118c1895', payload '{"state":"UNLOCK"}'
2019-02-12T23:09:02.956Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 56
2019-02-12T23:09:02.963Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:02.976Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-02-12T23:09:03.787Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2019-02-12T23:09:03.788Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 56
2019-02-12T23:09:07.891Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:07.895Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:07.895Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:10.273Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:10.276Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:10.277Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
  zigbee2mqtt:debug 2019-2-12 20:09:13 Received MQTT message on 'zigbee2mqtt/0x000d6f00118c1895/set' with data 'LOCK'
  zigbee2mqtt:info 2019-2-12 20:09:13 Zigbee publish to device '0x000d6f00118c1895', closuresDoorLock - lockDoor - {"pincodevalue":""} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-02-12T23:09:13.543Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 57
2019-02-12T23:09:13.553Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
  zigbee2mqtt:info 2019-2-12 20:09:13 Zigbee publish to device '0x000d6f00118c1895', closuresDoorLock - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2019-02-12T23:09:13.756Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 58
2019-02-12T23:09:13.773Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-02-12T23:09:13.778Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2019-02-12T23:09:13.780Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 58
  zigbee2mqtt:error 2019-2-12 20:09:13 Zigbee publish to device '0x000d6f00118c1895', closuresDoorLock - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
  zigbee2mqtt:debug 2019-2-12 20:09:21 Received MQTT message on 'zigbee2mqtt/0x000d6f00118c1895/set' with data 'UNLOCK'
  zigbee2mqtt:info 2019-2-12 20:09:21 Zigbee publish to device '0x000d6f00118c1895', closuresDoorLock - unlockDoor - {"pincodevalue":""} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-02-12T23:09:21.404Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 59
2019-02-12T23:09:21.415Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
  zigbee2mqtt:info 2019-2-12 20:09:21 Zigbee publish to device '0x000d6f00118c1895', closuresDoorLock - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2019-02-12T23:09:21.614Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 60
2019-02-12T23:09:21.628Z zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2019-02-12T23:09:22.945Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:22.954Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:22.955Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:22.978Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2019-02-12T23:09:22.979Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 57
2019-02-12T23:09:22.985Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:22.987Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:22.988Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:22.990Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:22.992Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:22.992Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:09:22.993Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:09:22.995Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:09:22.996Z zigbee-shepherd:request REQ --> AF:dataRequest, transId: 61
  zigbee2mqtt:info 2019-2-12 20:09:22 MQTT publish: topic 'zigbee2mqtt/0x000d6f00118c1895', payload '{"state":"LOCK"}'
[... cut ...]
2019-02-12T23:11:07.293Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-12T23:11:07.295Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-12T23:11:07.296Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-12T23:11:09.776Z zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2019-02-12T23:11:09.779Z zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 66
  zigbee2mqtt:debug 2019-2-12 20:11:11 Saving state to file /share/zigbee2mqtt-edge/state.json

```

matisaul, 2019-02-13:
UPDATE: I get a message from the lock:

```
2019-2-13 10:09:31 - debug: Received zigbee message of type 'devChange' with data '{"cid":"closuresDoorLock","data":{"lockState":1}}' of device 'iZBModule01' (0x000d6f00118c1895)
2019-2-13 10:09:31 - warn: No converter available for 'YMF40' with cid 'closuresDoorLock', type 'devChange' and data '{"cid":"closuresDoorLock","data":{"lockState":1}}'
``` 

So, I changed the lockstatus on converters (I appears that lockstate === 1 is LOCK:
```
const converters = {
  YMF40_lockstatus: {
      cid: 'closuresDoorLock',
      type: 'devChange',
      convert: (model, msg, publish, options) => {
          return {state: msg.data.data.lockState === 1 ? 'LOCK' : 'UNLOCK'};
      },
  },
```

And now I can read the status!

```
2019-2-13 10:16:31 - debug: Received zigbee message of type 'devChange' with data '{"cid":"closuresDoorLock","data":{"lockState":1}}' of device 'iZBModule01' (0x000d6f00118c1895)
2019-2-13 10:16:31 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00118c1895', payload '{"state":"LOCK"}'
```

matisaul, 2019-02-15:
Nope... I don't get updated states from the lock. The only way to know the state, is calling: "zigbee2mqtt/0x00000/get" 

I think that my error (AF data request fails, status code: 240. MAC transaction expired.) is causing problems to get actual status...

@Koenkk could you help me with this, please?

CanuckianOz, 2019-02-15:
> Wow, thanks @CanuckianOz !! I'll test it tonight.
>
> I forgot to ask you... Do you have this zigbee module?: [...]

Hey @matisaul yes that's the one! Standard zigbee module from Assa Abloy.

Koenkk, 2019-02-15:
@matisaul you seem to have a different model, @CanuckianOz has zigbee model `YRD426NRSC` you have `YMF40`. Please open a new issue so we can also support this device.

@CanuckianOz can you confirm that this lock is working correctly now? If that's the case, please close this issue.

CanuckianOz, 2019-02-17:
@Koenkk I was just going to mark this as closed but came across an issue. It mostly is and is receiving lock/unlock commands, but I noticed yesterday that it's no longer picking up the lock status changes in HASS (but changes are showing up in the log) and getting this "no converter available" message in the zigbee2mqtt-edge log when I turn the lock deadbolt. Thoughts?!

https://hastebin.com/qujaguriji.sql

matisaul, 2019-02-18:
> @matisaul you seem to have a different model, @CanuckianOz has zigbee model `YRD426NRSC` you have `YMF40`. Please open a new issue so we can also support this device.
> 
> @CanuckianOz can you confirm that this lock is working correctly now? If that's the case, please close this issue.

@Koenkk I opened a new topic about this #1118

Koenkk, 2019-02-20:
@CanuckianOz when not receiving status updates do you see anything in the log?

CanuckianOz, 2019-02-20:
@Koenkk I'm now running the standard zigbee2mqtt release as I can see the Yale Lock has been added. Here's the log when I reboot:

https://hastebin.com/idujuwusul.makefile

I noticed there's a mismatch with the zigbee model and model:

    zigbeeModel: ['YRD446 BLE TSDB'],
    model: 'YRD426NRSC',

Mine is definitely the YRD426NRSC.

I tried my hacked code, the -edge release and the standard release. Tried uninstalling both and removing the repository and re-adding. Just can't get it to recognize the lock again. 

There were different errors earlier this week with other .js files but I can't troubleshoot as I'm using Hass.io and can't edit the files. Let's see what's up with this one?

Koenkk, 2019-02-22:
@CanuckianOz are you sure you are running the latest `-edge` release, can you post your startup log?

CanuckianOz, 2019-02-22:
@Koenkk Yes. Uninstalled all and removed the repository before reinstalling the edge version and ensuring my config is right. 

Here ya go. Starts up all good then starts throwing errors around the `closuresDoorLock` function. 

https://hastebin.com/aforocomol.makefile

By the way, thanks for all your work and help!

Koenkk, 2019-02-22:
@CanuckianOz can you post the real beginning? The zigbee2mqtt version is logged there.

CanuckianOz, 2019-02-22:
@Koenkk d'oh this must be frustrating for you...

https://hastebin.com/vibucokelo.makefile

Koenkk, 2019-02-22:
Seems that with hassio, the commit version is not included, could you go inside the docker container (http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html) and post the contents of `node_modules/zigbee-shepherd-converters/devices.js` here?

CanuckianOz, 2019-02-22:
@Koenkk arg... I think the next line that I missed answers our question:

`  zigbee2mqtt:info 2/22/2019, 6:27:39 PM Coordinator firmware version: '20190109'

Which was before you revised. Any idea how I get Hassio to grab the latest?

Koenkk, 2019-02-22:
That is the coordinator firmware version, which is up-to-date.

Somehow the lock cannot be reached, can you try re-pairing the device?

CanuckianOz, 2019-02-22:
Just giving that a go now... The lock is unpaired but zigbee2mqtt is showing it is paired and I can't remove it. I can't re-pair the lock for whatever reason.

Koenkk, 2019-02-22:
- Stop zigbee2mqtt
- Remove all entries of this device from database.db
- start zigbee2mqtt
- pair

CanuckianOz, 2019-02-25:
@Koenkk Sorry, I worked a good chunk of my weekend to get this working again and finally got it. I tried your advice but it wasn't allowing *any* new devices to join, including some xiaomi zigbee sensors. An MQTT device was also publishing a constant lock command. So I copied my settings and uninstalled Mosquitto, Z2mqtt and re-flashed my CC2531 dongle.

Anyway, fresh install of everything including using the zigbee2mqtt-edge Hass.io add-on and it now works out of the box. Got the lock working in 2min once that was done. 

I'll close this one out, thanks buddy!

closed 2019-02-25 by CanuckianOz

kiwijunglist, 2019-02-26:
Hey.

I have Yale Assure Bluetooth Lock. YRD426NRSC. Supports optional zigbee or z-wave module.

Listed here (it's the one that also takes a key).
https://www.assaabloy.co.nz/en/local/nz/products/keyless-entry/electronic-digital-locking/assure-lock/

Is this lock working with zigbee2mqtt? I have the lock and am currently using zigbee2mqtt for all my Xiaomi gear. Is our safe to go ahead and buy the zigbee module?

PS. I don't use hass.io, I use home assistant in docker and zigbee2mqtt in a docker container as well.

Thanks!

Very excited to hear this is working!

CanuckianOz, 2019-02-26:
Hey, yep that’s the exact one I have and it’s now working on zigbee2mqtt (developer branch). You don’t need Hass.io, in fact using that made it harder for me to troubleshoot. 

I’m also currently using a xiaomi device on z2m for testing and will be transitioning over from the Xiaomi Gateway for the rest of the devices.

kiwijunglist, 2019-02-26:
Sweet this is brilliant news. I had been waiting for z wave support for this model, and was thinking of buying both a zwave stick and the zwave module. This will save me money.

Are you based in Australia?

CanuckianOz, 2019-02-26:
Yep I’m in Oz and that’s why I did it, to save money and not have another bridge to manage. Plus Z-wave modules are hard to find here and ANZ-compatible bridges on our frequency are even more so.

kiwijunglist, 2019-02-26:
Where did you buy the zigbee module in Australia from and how much was it.  They are $150 here.  I have a friend who regularly goes back and forth to Sydney, I might buy it in Au.

CanuckianOz, 2019-02-26:
I got mine from eBay for $80 here:

https://www.ebay.com.au/itm/Yale-Assure-Zigbee-Network-Module-integrate-your-Yale-Lock-to-home-auto-system/273430010317?epid=6023135796&hash=item3fa9b2d9cd:g:SuMAAOSwm~Fbhd4T

kiwijunglist, 2019-03-11:
@CanuckianOz

Are you sure that is the right one?

On the auction description it has YRD-ZBM-1 listed as the model number.

But I thought our locks needed yrd-zbm-2

Can you confirm,
Thanks.

kiwijunglist, 2019-03-11:
See here:

ZBM-2

https://www.joineryhardware.co.nz/shop-digital-entrance-door-locks.html?id=2507-yale-assure-zigbee-module

ZBM-1

https://www.joineryhardware.co.nz/shop-digital-entrance-door-locks.html?id=2754-yale-zigbee-network-module

CanuckianOz, 2019-03-11:
Yes it needs to be ZBM 2. The other one is Zwave. That’s the exact seller and and i got it from and it’s zigbee and works fine. I didn’t look at the model number... thats a typo.

kiwijunglist, 2019-04-01:
I get an error trying to operate the zigbee 2 mqtt lock using an automation in home assistant.

Here is the entity:
````lock.front_door | unlocked | battery: 95 linkquality: 49 battery_low: false state: UNLOCK friendly_name: Front Door````

Here is the automation:
````
  alias: 'Lock door if closed for > 60 min'
  trigger:
    - platform: state
      entity_id: binary_sensor.door_front
      to: off
      for:
        minutes: 60
  action:
    service: lock.lock
    data:
      entity_id: lock.front_door  
````

and here is the error when i try to validate my yaml files

````
Configuration invalid
Package lock setup failed. Component automations does not exist
````

kiwijunglist, 2019-04-01:
Thanks, i bought the module from here: https://www.ebay.com.au/itm/Yale-Assure-Zigbee-Network-Module-integrate-your-Yale-Lock-to-home-auto-system/273430010317?epid=6023135796&hash=item3fa9b2d9cd:g:SuMAAOSwm~Fbhd4T

And it works with my Yale YRD 426 Assure Bluetooth Lock.

Thank you!

hvddrift, 2019-04-29:
A touch late to the party. Another Aussie here.

I have added the Yale lock with the Zigbee module and it has picked up the device and I can Lock and Unlock via both MQTT and home assistant.

However, the state always shows as LOCKED.

Any suggestions?

kiwijunglist, 2019-04-29:
@hvddrift I noticed that when first added then after a bit it was solid. I think the mesh just has to get stronger or something.

CanuckianOz, 2019-04-29:
Me too - seems to come good after a while for some reason. @Koenkk is this a normal problem?

[22 more replies]