Koenkk/zigbee2mqtt #1034: permit_join via router

DrRob, 2019-02-06:
Hi,

I'm hoping this isn't a stupid question:

With reference to https://github.com/Koenkk/Z-Stack-firmware/issues/30 

will zigbee2mqtt send the "ZDO" message to any routers attached when permit_join is set to true, and something else (?) to undo it when permit_join is false, so that the whole network observes the permit_join state, not just the coordinator?

Thanks,
Rob.

Koenkk, 2019-02-06:
Yes, in https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L327 you see that when type `all` is used a permit join request is send to all routers and the coordinator (https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L343). Zigbee2mqtt uses `type` `all` (which is the default).

DrRob, 2019-02-07:
Thanks for not just answering but also teaching me something about the code.  11 out of 10.  👍

closed 2019-02-07 by DrRob