Koenkk/zigbee2mqtt #1037: Tradfri Linkquality

sargent-penguin, 2019-02-06:
The link quality of Tradfri bulbs is displayed on the lines in the Graphviz, but not send as a json value in the MQTT topic.

Could this be added as an attribute, to be able to monitor it in home assistant for example?

Koenkk, 2019-02-07:
TRADFRI bulbs normally don't produce any messages, therefore the linkquality cannot be acquired. Wit h the new reporting functionality they will. Docs: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md

frogale, 2019-02-15:
Can I ask for more information? Provided link is not working.. :-(

Koenkk, 2019-02-15:
@frogale https://github.com/Koenkk/zigbee2mqtt.io/blob/master/configuration/device_specific_configuration.md (note that this also requires the dev branch).

sargent-penguin, 2019-02-16:
Thanks for the info.
Just wondered, because the Osram Smart Plug sends its linkquality already, without the specific config.
Will this behaviour be changed for consistency?

Koenkk, 2019-02-16:
Thats because the osram plug sends zigbee messages to the coordinator, tradfri bulbs dont do this by default, but should do when using report.

sargent-penguin, 2019-02-16:
That makes sense!
Thanks for the insight and help!

closed 2019-02-16 by sargent-penguin