Koenkk/zigbee2mqtt #1039: Log too verbose ?

H3llK33p3r, 2019-02-06:
Hi just to give a quick feeling on the application. First thank you for this great project.

Am I the only one that consider that sending the JSON message in "zigbee2mqtt:**info** MQTT publish: topic" is too much verbose ? More over the deletion of previous log folders is done only when 10  directories have been created. For me, the project is not restarted very often so logs collected are huge. 

Can you consider to refactor some messages to put them in debug only for example (to avoir to fork this project ^^)  ? 

Thank you in advance for the response.

Koenkk, 2019-02-06:
The log is automatically rotated, per directory, max 3 files with a max file size of 10mb each are kept.

clockbrain, 2019-02-06:
Try it with log level of 'warn' which doesn't include the mqtt messages.

H3llK33p3r, 2019-02-06:
@Koenkk  ok thanks for this information so max 3 * 10 * 10 that's no so bad :) (I do not see the 10Mb part sorry) . 
@clockbrain : it's that I have done.

closed 2019-02-06 by H3llK33p3r