Koenkk/zigbee2mqtt #1040: Generic support for all ZLL devices - is it possible?

james-fry, 2019-02-06:
Did you read the FAQ?
https://koenkk.github.io/zigbee2mqtt/information/FAQ.md

Yes :)

Unsupported device?
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Indeed I do.

A question?
Read the documentation: https://koenkk.github.io/zigbee2mqtt/

Yes - its a new question I hope - I searched the issues and wiki.

Is it possible that zigbee2mqtt can be enhanced to provide generic support for any/all ZLL devices? i.e. its not necessary to update and add code to specifically support each and every one?
Currently Hue hub, Ikea tradfri hub, amazon echo, smarthings etc can all pair devices that are ZLL compliant without needing per-device type specific support to be added in their firmware.

I just purchased a zigbee ZLL dimmer that is compatible with the above hubs, but I will need to code support in Z2M. (Device with modelID 'HOMA1031' is not supported.)
I will attempt to add support for this device...

Koenkk, 2019-02-06:
No, in practice they almost all require a different configuration (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js). (this could perhaps be done for bulbs)

james-fry, 2019-02-06:
Thanks for quick response. I guessed this might be the answer :)
Although I assume ZLL (zigbee light link) does target bulbs (or bulb-like devices such as dimmers) only.

khzd, 2019-02-07:
Hi
With the succès of zigbee and Z2M, what will be the code with 1000 différents devices
I think may  think to an alternative to ModelId strategy construction 
Cheers

ryanbeaton, 2019-02-07:
How are the hubs doing it then?

Chrischi-, 2019-02-07:
Basically, each device submits an identify with available / supported clusters and Zigbee2MQTT writes them to the database.db. If the devices fulfill the implementation via ZCL, then you should have the entire cluster (according to ZCL) available in `toZigbee.js` and `fromZigbee.js`. With this you could then create / supplement the devices.js. But this is only one approach. Of course there are more things to think about. 

On the other hand Zigbee2MQTT is only a bridge and does not need this logic at all. 

Maybe someone could build an extra device wizard!?!

Only my thoughts.

james-fry, 2019-02-07:
I started looking to support the ZLL dimmer I have (which btw might happen more slowly now I see it works perfectly on my hue hub)

There are a number of archetypes for ZLL bulbs:
- on/off+brightness
- on/off+brightness + colour temp
- on/off+brightness + colour temp + rgb

I can see that these result in standardised approach in devices.js:
- on/off+brightness --> extend: generic.light_onoff_brightness
- on/off+brightness + colour temp --> extend: generic.light_onoff_brightness_colortemp
- on/off+brightness + colour temp + rgb --> extend: generic.light_onoff_brightness_colortemp_colorxy

I'm guessing that these three archetypes are determined from some pretty specific and common zigbee zll clusters, so is it not possible for there to be analysis when a devices is paired to identify ZLL clusters and automatically determine correct extension?

Koenkk, 2019-02-07:
@james-fry yes this can be done, but note that `zigbee-shepherd-converters` also serves as a central point where working devices are gathered (also the zigbee2mqtt supported devices documentation is based on this). What we could do is to auto generate the converter so that pull request can be created more easily.

closed 2019-04-15 by stale[bot]

bohtho, 2019-12-04:
With a check for generic archetypes in zigbee2mqtt a HomeKit/Homebridge converter would be easy to make on top of this.