Koenkk/zigbee2mqtt #1042: Xiaomi RTCGQ11LM modelID 'undefined'

efiniste, 2019-02-07:
I'm having problems adding Xiaomi presence sensors to my standalone zigbee2mqtt. I set it up last night so sniffer firmware is the latest (9th of Jan iirc) and zigbee2mqtt is showing V1.1.1.

I've paired 2 new presence sensors (RTCGQ11LM) now, both show the following error once paired:

 zigbee2mqtt:warn 2/7/2019, 12:09:22 PM Device with modelID 'undefined' is not supported.

I've paired a brand new smoke detector (JTYJ-GD-01LM/BW) and that looks to be working correctly. Battery state, alarm state both report properly.

I've paired a presence sensor (RTCGQ11LM) that was previously paired with the Xiaomi gateway and that's reporting properly too. Occupancy state and illuminance level both report in to the log.

I've tried deleting the 1st presence sensor from database.db and configuration.yaml and pairing it with the Xiaomi gateway. It showed and worked correctly in there. Then I deleted it from Xiaomi gw and paired again with zigbee2mqtt. Still showing the "Device with modelID 'undefined' is not supported" error.

I tried copying the database.db line for the working presence sensor and modifying it to match the id etc of the non-working sensor. Now it shows properly at zigbee2mqtt startup but still gives the error message above on events.

Can anyone point me in the direction of what to try next? zigbee2mqtt seems to be working fine but for whatever reason it won't identify these sensors. Is this a sensor firmware problem maybe?

Log is available here: https://pastebin.com/69kVczUc

smokesensor is the smoke detector - working fine
presenceoffice is the sensor that was originally paired with the Xiaomi gateway - working fine
presencehall is the sensor that has been between zigbee2mqtt and the Xiaomi gateway and had it's db entry doctored - throwing errors

You can see the errors on various lines but 62-66 are an example. I can't see a way to differentiate between the 2 faulty sensors that are paired in this log, they both give identical errors.

database.db is available here: https://pastebin.com/8kDHPM83

id5 is presenceoffice
id6 is presencehall (originally id2 before I deleted it and paired with Xiaomi GW)
id4 is the presence detector that's had no messing about. Just out of the box and paired with zigbee2mqtt 

Thanks,

Steve

efiniste, 2019-02-07:
That deleted a working sensor and when I re-paired it it's come back with the same errors as the faulty sensors.

I've deleted the new lines added the id5 line back in.

efiniste, 2019-02-07:
Is there a link to how the db file is formatted? There are 2 lines for each end device which appear to be identical.
I've added a few other RTCGQ11LM sensors to the network now, all of which were previously paired with the Xiaomi gw and all of which have configured correctly.

I've added 1 more brand new out of the box RTCGQ11LM and it's failed exactly like the others.

What am I missing here?

efiniste, 2019-02-07:
Also, kzhd, what do you mean by, "It doesn't have modelID!!!"? I can see modelID in the line you've quoted, "modelId":"lumi.sensor_motion.aq2",

subzero79, 2019-02-08:
I guess this maybe one of the most desired improvements of z2m, faster pairing. Using the xiaomi gw takes 10 seconds to pair a device. Is normal that some xiaomi devices take up to 10 minutes to pair, by pressing the button every two seconds and also being close to the coordinator or a router. Keep trying, the undefined msg is normal at first.

clockbrain, 2019-02-08:
@efiniste There isn't any structure to the database, its just one row per device. zigbee2mqtt uses zigbee-shepherd which uses  [objectbox](https://www.npmjs.com/package/objectbox) which in turn uses [NeDB](https://www.npmjs.com/package/nedb).

From the NeDB page regarding persistence: 

> Under the hood, NeDB's persistence uses an append-only format, meaning that all updates and deletes actually result in lines added at the end of the datafile, for performance reasons. The database is automatically compacted (i.e. put back in the one-line-per-document format) every time you load each database within your application.

So you will find one row per device before the index line
```
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
```

and potentially multiple rows per device after that line. I can't see anywhere in the code that it does a compact on shutdown so I think the last lines will be the latest in a stopped system.

I'd avoid messing with the database unless you have to purge a device that won't go away in which case you would delete all lines with the relevant id.

khzd, 2019-02-08:
Hi
By experience if after $$index.. in the DB, i use to clean all duplicate lines manually , keeping the more complete one, show-stopper is modelid.
And repair again.
Some time I take off the battery or try with another one, can work
This is only a workaround of pairing nightmare :(

efiniste, 2019-02-08:
Thanks for the info everyone. I got one of the sensors working today so it does look like it's just down to patience.

Steve

closed 2019-02-08 by Koenkk