Koenkk/zigbee2mqtt #1043: Choose the "right" channels

h4nc, 2019-02-07:
I'd like to know how you guys choose the right channel for zigbee2mqtt.

Currently I have wifi router with two repeaters choosing their channel in auto-mode. Currently they use 11 and 6. 

I also have a hue bridge (I still have to have this one because my philips tv is connected to one hue bulb, I don't think zigbee2mqtt can do that) and the zigbee2mqtt coordinator.

I don't know which channel the hue bridge uses. Zigbee2mqtt is set to channel 11.

Will those interfere with each other. I also have some sonoff devices (wifi) that work most of the same but sometimes for a few minutes they don't. If they fail longer I have to restart my repeaters.
I'm starting to wonder if than I "wrong" channel is chosen and after restarting it picks a good one again. 

Do you set the channel static? Which one would you recommend? I can only see 2 other networks, so there is not much going on from the neighbours side.

Koenkk, 2019-02-07:
This is the best explanation I've seen so far: https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

h4nc, 2019-02-07:
I saw this one already, but how do you do it? Do you set up static channels. Or do you go the auto-setup way?

Don't know if I should change that? Also is it even possible to change the channel from the hue bridge?

Koenkk, 2019-02-07:
Static channel for Wifi and Zigbee. Changing the channel of hue is possible: https://www2.meethue.com/en-us/support/bridge/connectivity/what-is-a-zigbee-channel-change-in-the-philips-hue-app

h4nc, 2019-02-07:
> Also, when there is no problem, don't worry about it.

Thats the thing, my sonoffs go offline sometimes, but I don't know the reason and so I thought it might be the channels.

djchen, 2019-02-21:
Is there a reason to not use channel 1 through 10?
I am looking at http://www.rfwireless-world.com/Terminology/zigbee-channels.html and it seems like those channels are on a much lower frequency and thus have better penetration and range.

Thanks

Koenkk, 2019-02-22:
@djchen AFAIK, the devices we typically use are 2.4ghz devices.

closed 2019-04-23 by Koenkk