Koenkk/zigbee2mqtt #1044: HS3SA

raptorblue, 2019-02-07:
Hello,

First of all, I would like to thank you for the efforts you put in developing zigbee2mqtt!

Unfortunately I've got a problem with a new device:
I've tried to integrate a HEIMAN HS3SA smoke detector but was unable to complete the setup. Pairing works and the device is listed in the network map as well. But I do not receive any messages from it. Also, during start-up, the following message appears:

**zigbee2mqtt:error 2/7/2019, 3:16:58 PM Failed to configure 0x00158d0001dee7a2 (0x00158d0001dee7a2) ('Error: AF data request fails, status code: 240. MAC transaction expired.')**

Range is also not the issued, since it does not work if the distance is 2 rooms or 1 meter.

**I've put more infos here:**
**Log:** http://pbot.icip.de/9ab20dec7594187b3616e5339676459b
**Config:** http://pbot.icip.de/d01832aec367e328bbc0e132622cbefa
**Database:** http://pbot.icip.de/f3c81fdfc6bc82d3850ac44fca94e444
**State:** http://pbot.icip.de/9c8587688b65291fa887e80319899596
_Please be aware that these files also contain information regarding another device (Aqara door sensor)._
If there is any further information I could provide, please let me know.

Any help is highly appreciated!

Thank you very much
RB

Kryzek, 2019-02-07:
I would say device is sleeping at startup when Z2M is trying to configure it:
2/7/2019, 3:16:50 PM - debug: Configuring 0x00158d0001dee7a2 (0x00158d0001dee7a2) ...
And
2/7/2019, 3:16:58 PM - error: Failed to configure 0x00158d0001dee7a2 (0x00158d0001dee7a2) ('Error: AF data request fails, status code: 240. MAC transaction expired.')

If there is a test button try to keep pressing it to keep device awake during Z2M startup and check if configuration goes through.

raptorblue, 2019-02-08:
Thank you @Kryzek,
Indeed pressing the test button during startup helped and z2m was able to configure the device:

 zigbee2mqtt:info 2/7/2019, 7:13:02 PM Succesfully configured 0x00158d0001dee7a2 (0x00158d0001dee7a2)

zigbee2mqtt:debug 2/7/2019, 7:13:03 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'SmokeSensor-N' (0x00158d0001dee7a2)

zigbee2mqtt:info 2/7/2019, 7:13:03 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001dee7a2', payload '{"smoke":false,"battery_low":false,"linkquality":15,"device":{"ieeeAddr":"0x00158d0001dee7a2","friendlyName":"0x00158d0001dee7a2","type":"EndDevice","nwkAddr":1811,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-N","status":"offline"}}

But the problem still exists when I restart z2m now and it runs into the "MAC transaction expired" again. Is there a way to tell z2m to configure a known device "offline" or something else I can do? Pressing the test button every time I have to restart z2m is not practical, especially considering the fact that I plan to deploy several of these smoke sensors.

Does anyone else use HEIMAN smoke sensors and can share their experience?

Thank you very much!

RB

Kryzek, 2019-02-08:
I believe that configuration is always the same for the device and it keeps it even if Z2M is restarted. So device should still be working after restarting the software. This is just what I’ve figured so take this with a grain of salt.  But it should be fine after the configuration has passed.

Koenkk, 2019-02-08:
@Kryzek is right

raptorblue, 2019-02-20:
Thank you for your feedback and apologies for my late reply.
I am still testing around with the sensor and already triggered an alarm with a cigarette. But nothing appeared in the log. After placing the sensor in my office (quite away from the zigbee stick) I get a message with link quality 0 from time to time.

2/19/2019, 7:06:45 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0001dee7a2', payload '{"smoke":false,"battery_low":false,"linkquality":0,"device":{"ieeeAddr":"0x00158d0001dee7a2","friendlyName":"0x00158d0001dee7a2","type":"EndDevice
","nwkAddr":1811,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-N","status":"online"}}'
Is this an actual message from the smoke detector or an automated status updated message generated by z2m?

Now I've placed the sensor closer to the zigbee stick to see what happens.
Anyone else running a HS3SA sensor with z2m and can share their experience?

Thanks a lot!

h4nc, 2019-02-26:
I got a HS1SA yesterday. It worked fine to connect it. Also testet it with testspray. Reportet to homeassistant just fine.

But I have two issues:
1) the test mode doesn't work. So when I press and hold the button for 3 seconds it should test and start to beep. It doesnt. Pressing the button once when there is a real alarm (testspray) turns the alarm off, this tells me the button actually works.

2). Does this device report the battery? @Koenkk where in the code can I find what states are catched by zigbee2mqtt. I already saw this, but can't find it anymore.

h4nc, 2019-03-29:
does someone know if the testmode works?

raptorblue, 2019-04-16:
Hi guys,

Sorry for the late reply. I did some more testing and now the test button work. I get the event also in fhem and it's fine now. Nevertheless the device seems to have quite a limited range (compared to my Aquara door sensors).

Thank you for your help. 

Regards,
RB

h4nc, 2019-04-16:
@raptorblue So you are saying that you get a state change in home assistant when you push the button.

And another question:
Does your smoke detector beep when you push the test button. Mine doesn’t, but the buttons seems to work because it cancels a real alarm triggered by a testspray

raptorblue, 2019-04-17:
FHEM, not home assistant - but yeah. But I have to press the button for a few seconds, then the detector starts screaming and the FHEM event is triggered.

h4nc, 2019-04-17:
Weird mine does't scream. Even when I press it over 10 seconds. Tried about 3 sec. 10 sec ... nothing.

But it doesn't seem like it is faulty.

raptorblue, 2019-04-17:
Your model is different than mine. Still, the test button should trigger an alarm. But maybe this behavior is intended to outline a connection problem with the (original) gateway. Maybe the vendor could help you but probably they will refuse to handle problems without the original gateway.

If I can help you with further testing please let me know.

h4nc, 2019-04-17:
What’s your model number? 

Mine is a HS3SA

raptorblue, 2019-04-17:
Mine is also a HS3SA but you mentioned HS1AS earlier

> I got a HS1SA yesterday. It worked fine to connect it. Also testet it with testspray. Reportet to homeassistant just fine.

h4nc, 2019-04-17:
You are right that is confusing. 

I bought the HS1SA-N and the seller also told me that it is a HS1SA, but within zigbee2mqtt it describes itself as a HS3SA.

closed 2019-06-23 by stale[bot]