Koenkk/zigbee2mqtt #1047: transition time to fast

JackGruber, 2019-02-07:
Did you read the FAQ?
https://koenkk.github.io/zigbee2mqtt/information/FAQ.md
Yes

Unsupported device?
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
No

A question?
Read the documentation: https://koenkk.github.io/zigbee2mqtt/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Hi,

the transition option is specified in seconds, but for me this is 1/3 of the specified time. What could be the reason for this? 
```
'{"color":{"r":255,"g":0,"b":0}'
'{"color":{"r":0,"g":255,"b":0},"transition":45}'
```
results in approx 15 seconds transition

Version: 1.1.0 (commit #cf7e0d1)
Coordinator firmware version: 20180815

Koenkk, 2019-02-08:
What bulb do you have?

JackGruber, 2019-02-08:
Gledopto GL-C-008 and GL-B-008Z
I'll try again with smaller values and get a feedback.

sehraf, 2019-02-08:
@Koenkk do all bulbs support transition time? While 5 seconds are working fine for my RGB GU Hues, it didn't work with my Ikea and and Hue plain white dim-able bulbs.

Koenkk, 2019-02-08:
@sehraf it's a feature of the device, so it depends on the zigbee implementation of the device.

JackGruber, 2019-02-08:
No, the same result with 5 seconds.

Koenkk, 2019-02-08:
@JackGruber so when you want 5 second, you need to send 15?

JackGruber, 2019-02-09:
Hm it is difficult to see the changes near the end of the transition,
but i think its every time the factor **3.3** afert some  testing.
I tested it from up to 2 minutes, with different times.

Koenkk, 2019-02-09:
@JackGruber thanks, should be fixed in the latest dev branch (transitions will by multiplied by 3.3).

JackGruber, 2019-02-13:
Looks fine :)

closed 2019-02-13 by JackGruber