Koenkk/zigbee2mqtt #1051: Broken CC2531?

schattenkrieger, 2019-02-09:
Just got my CC2531 together with the Cable today in the mail. i was planning to flash it with an arduino acccording to the documentation.
the arduino seems flashed fine and CCLib also seems to be working ok.
but when i try to get the info from the CC2531 to see if everything is set up correctly i get the following error:
```
ERROR: No chip found. Check your connection and/or wiring!
```
i could obsere the following: when i just connect the CC2531 Dongle i can find a TI USB device with lsusb and the green LED on the Dongle is lit.
As soon as i connect the Arduino the LED goes off and the TI device disappears.
When i plug in the Dongle with the Arduino allready connectet the LED just stays off
When i now issue 
```
python2 cc_info.py -p /dev/ttyUSB0 
```
the LED on the Dongle goes on and then goes off after a few moments and i get the above error.
i checked the connections a couple of times but cant find any obvious error.
When removing the shortcut between CC_DD_0 and CC_DD_1 and let the open cable just dangle in the air i get the following error:
```
ERROR: Short-circuit or wrong wiring detected. Check your connection and/or wiring!'''
```
if i remove the cable completely and leave pin 4 or 5 just witout any connection i get this error:
```
ERROR: The chip is not responding. Check your connection and/or wiring!
 ```
Is there any way to find out if the CC2531 is defective of if i made some error connecting the arduino?

schattenkrieger, 2019-02-09:
Just continued to try to figure out how to fix my problem, i connected the CC2531 Dongle to an USB Charger and not to a free USB Port on my machine. With this Setup it seems to be working a bit better but now i get this error:
```
ERROR: No driver found for your chip (chipID=0xe000)!
```

Koenkk, 2019-02-09:
I have little experience with flashing via Arduino, but the one time I tried I failed to do so (same errors as above). Personally I prefer the CC debugger.

taifunk, 2019-02-25:
I am experiencing same problems, not able to flash with Arduino.
Getting the same: `ERROR: No chip found. Check your connection and/or wiring!`
Altough I'm using this: https://www.aliexpress.com/item/RF-TO-USB-CC2530-CC2591-RF-switch-USB-transparent-serial-data-transmission-equipment/1996354384.html?spm=a2g0s.9042311.0.0.768a4c4d7yYgVg
I'm thinking also about getting CC debugger for flashing, maybe its easier.

closed 2019-05-03 by stale[bot]