Koenkk/zigbee2mqtt #1052: How many devices can i connect with 4 diy cc2530 routers

JLFN, 2019-02-09:
I looking FAQ for optimal network as i understand CC2530 diy routers give me max devices to add is that right? Because iam thinking of because Ikea Extender if thats not the case.

Koenkk, 2019-02-09:
@ptvoinfo Do you know how many devices can connect through the CC2531 router with your firmware? (what's the `NWK_MAX_DEVICE_LIST`?)

ptvoinfo, 2019-02-10:
`NWK_MAX_DEVICE_LIST` is 21

JLFN, 2019-02-10:
@Koenkk  does this mean teoretic i can have 15 devices redirect to the cordinator and 21 for each cc2531 router ?  15+21+21+21+21 is that really correct?

Koenkk, 2019-02-10:
11 to coordinator (15 - 4 routers) + 21 to each router = 95 devices

JLFN, 2019-02-13:
@Koenkk  @ptvoinfo Can you please add an FAQ that compares the different firmware. Thanks in advance.

Koenkk, 2019-02-13:
@JLFN done!

closed 2019-02-13 by Koenkk