Koenkk/zigbee2mqtt #1055: Problems with OSRAM ledstrip

marcgarciamarti, 2019-02-09:
Hello,

I have a supported OSRAM ledstrip (4052899926110 in the supported devices list). It works well sometimes, but it very frequently acts up on zigbee2mqtt. 

I'm experiencing some weird behaviours with it. For starters, Home Assistant front-end does not seem to sync the state with zigbee2mqtt meaning that I can trigger an automation using Node-red (my automation of choice platform), the light will react, but HA will sometimes not show the state accurately. I'm not sure I can pin-point the problem further but I'm willing to of course.

One other problem that I have with this device is that it sometimes makes zigbee2mqtt choke and takes it offline. See for example this log:  https://pastebin.com/bMW7fQha 

As you can see, a problem was encountered at 22:15 where HA or Node-red did not receive a response when either of both pulled the state on the ledstrip. Right at 23:30 my node-red automation commanded the strip to switch off but it didn't. I manually attempted to switch it off as well, and it did not work as pastebin illustrates.
 
Restarting zigbee2mqtt solves the issue and I can again interact with the ledstrip but after a few hours or maybe one day the same problem will be showing up again.

I'm running z2m 1.1.1.

marcgarciamarti, 2019-02-09:
Hi again,

after rebooting zigbee2mqtt, I managed to switch the ledstrip off but then my only ikea tradfri bulb started acting up too. See this pastebin: https://pastebin.com/sXGcMUBK 

Thanks

Koenkk, 2019-02-10:
- I see you are running an old firmware (`Coordinator firmware version: 'undefined'`). more recent versions shown the date of the firmware.
- Can you post your zigbee network map? (https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html)

marcgarciamarti, 2019-02-10:
Thanks @Koenkk !

is this the output you were expecting?

```
  zigbee2mqtt:info 2019-2-10 19:36:10 Starting network scan...
  zigbee2mqtt:info 2019-2-10 19:36:20 Network scan completed
  zigbee2mqtt:info 2019-2-10 19:36:20 MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b0012022476" [style="bold", label="{0x00124b0012022476|Coordinator|No model information available|online}"];
  "0x00158d000252b7a5" [style="rounded, dashed", label="{switchw1xiaomi|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"];
  "0x00158d000252b7a5" -> "0x00124b0012022476" [label="170"]
  "0x00158d000251f830" [style="rounded, dashed", label="{occupancys1xiaomi|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"];
  "0x00158d000251f830" -> "0x00124b0012022476" [label="170"]
  "0x00158d00026765c4" [style="rounded, dashed", label="{doors1xiaomi|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d00026765c4" -> "0x00124b0012022476" [label="170"]
  "0x00158d00022f35ca" [style="rounded", label="{plug1xiaomi|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
  "0x00158d00022f35ca" -> "0x00124b0012022476" [label="68"]
  "0x000b57fffeda5dae" [style="rounded", label="{light11000lmikea|Router|IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (LED1623G12)|offline}"];
  "0x8418260000caa3af" [style="rounded", label="{ledstrip1osram|Router|OSRAM Flex RGBW (4052899926110)|offline}"];
  "0x8418260000caa3af" -> "0x00124b0012022476" [style="dashed", label="0"]
  "0x00158d00027de6c3" [style="rounded, dashed", label="{cube1xiaomi|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
  "0x00158d00027de6c3" -> "0x00124b0012022476" [label="170"]
  "0x00158d000272b6f5" [style="rounded, dashed", label="{temperatures1xiaomi|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d000272b6f5" -> "0x00124b0012022476" [label="170"]
  "0x00158d0002725756" [style="rounded, dashed", label="{temperatures2xiaomi|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d0002725756" -> "0x00124b0012022476" [label="170"]
  "0x00158d0002721121" [style="rounded, dashed", label="{temperatures3xiaomi|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
}'
```

now on the coordinator version, I don't dare updating it to be honest... I have a few concerns with upgrading the coordinator like the need for re-pairing  the devices or if I could use a spare usb stick and flash the new version on it, and have my existing stick as a backup. Too much entropy already in my system.

marcgarciamarti, 2019-02-10:
see attached snapshot of what node-red reports when pulling the state on this device

[image]

Koenkk, 2019-02-10:
According to the network graph, the strength of the network is not good. Here are some tips and trick to improve it: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_improve_network_range.html

marcgarciamarti, 2019-02-10:
Thanks @Koenkk may I ask what it is that suggests the signal strength is poor?

Koenkk, 2019-02-10:
Some nodes are not connected (e.g. the tradfri bulb).

marcgarciamarti, 2019-02-10:
That's because I have a wall switch which is now preventing the energy from reaching the bulb

rgruebel, 2019-02-10:
I also have problems with this device.  On and off is usually no problem but changing color or dimming causes the same problems.  But the signal strength in my network is also not optimal.

marcgarciamarti, 2019-02-10:
All my other devices work well but not this one, and it sits across the room from the coordinator. I have xiaomi temp sensors sitting like 20m away from the coordinator with several walls in between with no problem. Seeing that at least one other person is having issues as well is interesting.

marcgarciamarti, 2019-02-11:
@Koenkk I've placed even closer the coordinator and the ledstrip and I'm still experiencing this. If, for a moment, we thought the device may not be fully compatible, what tests would you suggest to be done on it?

Koenkk, 2019-02-12:
@marcgarciamarti can you post your full log again when this fails (also the startup of zigbee2mqtt)?

marcgarciamarti, 2019-02-19:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2882119/log.txt)

@Koenkk I'm attaching it. I would like to provide some additional context that I think is relevant in order to get the whole picture.

I have an automation that switches on the light at sunset (see node-red snapshot)
[image]

This automation switched off the light yesterday evening at 23:30 as per my time-bound rule (as one can see on the time-stamp). That very same automation has attempted to switch on the light today at sunset but it was unable to do so, because prior to that, the automation checks the status of the LED and HA has reported the light as on (nobody has switched on the led after yesterday's automation). 

If I now go to home assistant front-end, the OSRAM led appears as off even though I manually switched it on today in the evening around 20:39 (you can see it in the logs) and I can promise the light is on (I'm using it right now!)

[image]

Koenkk, 2019-02-20:
The light state isn't updated because the on commands fails:

```
2019-2-19 20:39:42 - error: Zigbee publish to device '0x8418260000caa3af', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
```

Can you check if this still happens with the latest dev branch and latest dev firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin)? (doesn't require reflashing: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/what_does_and_doesnt_require_repairing.md).

marcgarciamarti, 2019-02-20:
@koenkk when you say it does not requiere reflashing do you actually mean re-pairing? I honestly am a bit wary about tampering with my USB stick because I did not flash mine initially...

Koenkk, 2019-02-20:
@marcgarciamarti yes, sorry, I mean re-pairing.

rgruebel, 2019-02-26:
> @marcgarciamarti yes, sorry, I mean re-pairing.

Have you already flashed the current firmware and noticed an improvement?

rgruebel, 2019-02-26:
@marcgarciamarti  Have you already flashed the current firmware and noticed an improvement?

marcgarciamarti, 2019-02-26:
@rgruebel I have not but knowing the problem still happens even after upgrading the stick puts me off.

Koenkk, 2019-02-27:
@rgruebel  Can you re-pair the device?

RyuzakiKK, 2019-03-17:
I'm currently using the last stable firmware and I never experienced these problems.

Every day I turn it on and off, and sometimes I also change the color, but it always worked.

marcgarciamarti, 2019-03-23:
> The light state isn't updated because the on commands fails:
>
> ```
> 2019-2-19 20:39:42 - error: Zigbee publish to device '0x8418260000caa3af', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - [...]

@Koenkk I finally got around to re-flashing the usb stick with the latest version and I'm afraid my problem still happens. This is the output that I see:

`core-ssh:/share/zigbee2mqtt/log/2019-03-23.14-24-19# cat log.txt
3/23/2019, 2:24:19 PM - info: Logging to directory: '/share/zigbee2mqtt/log/2019-03-23.14-24-19'
3/23/2019, 2:24:19 PM - debug: Removing old log directory '/share/zigbee2mqtt/log/2019-03-23.13-53-01'
3/23/2019, 2:24:20 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6755,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
3/23/2019, 2:24:20 PM - debug: Loaded state from file /share/zigbee2mqtt/state.json
3/23/2019, 2:24:20 PM - debug: Saving state to file /share/zigbee2mqtt/state.json
3/23/2019, 2:24:20 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #unknown)
3/23/2019, 2:24:20 PM - info: Starting zigbee-shepherd
3/23/2019, 2:24:22 PM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
3/23/2019, 2:25:22 PM - info: Starting zigbee-shepherd
3/23/2019, 2:25:30 PM - info: zigbee-shepherd started
3/23/2019, 2:25:30 PM - info: Coordinator firmware version: '20190223'
3/23/2019, 2:25:30 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a63","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0012022476","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1553347530,"joinTimeLeft":0}
3/23/2019, 2:25:30 PM - info: Currently 10 devices are joined:
3/23/2019, 2:25:30 PM - info: switchw1xiaomi (0x00158d000252b7a5): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
3/23/2019, 2:25:30 PM - info: occupancys1xiaomi (0x00158d000251f830): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
3/23/2019, 2:25:30 PM - info: doors1xiaomi (0x00158d00026765c4): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/23/2019, 2:25:30 PM - info: plug1xiaomi (0x00158d00022f35ca): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
3/23/2019, 2:25:30 PM - info: light11000lmikea (0x000b57fffeda5dae): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/23/2019, 2:25:30 PM - info: ledstrip1osram (0x8418260000caa3af): 4052899926110 - OSRAM Flex RGBW (Router)
3/23/2019, 2:25:30 PM - info: cube1xiaomi (0x00158d00027de6c3): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
3/23/2019, 2:25:30 PM - info: temperatures1xiaomi (0x00158d000272b6f5): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/23/2019, 2:25:30 PM - info: temperatures2xiaomi (0x00158d0002725756): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/23/2019, 2:25:30 PM - info: temperatures3xiaomi (0x00158d0002721121): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/23/2019, 2:25:30 PM - info: Zigbee: disabling joining new devices.
3/23/2019, 2:25:30 PM - info: Connecting to MQTT server at mqtt://core-mosquitto:1883
3/23/2019, 2:25:30 PM - info: zigbee-shepherd ready
3/23/2019, 2:25:30 PM - debug: Mounted the cieApp (epId 11)
3/23/2019, 2:25:31 PM - info: Connected to MQTT server
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/switchw1xiaomi', payload '{"battery":100,"voltage":3002,"linkquality":42}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/occupancys1xiaomi', payload '{"occupancy":true,"linkquality":36,"battery":100,"voltage":3005}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/doors1xiaomi', payload '{"contact":true,"linkquality":44,"battery":86,"voltage":2975}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/plug1xiaomi', payload '{"state":"OFF","power":0,"voltage":234.5,"consumption":0.82,"temperature":25,"linkquality":68}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/light11000lmikea', payload '{"state":"OFF","brightness":246,"linkquality":39}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/ledstrip1osram', payload '{"state":"ON","brightness":254,"color_temp":370,"color":{"x":0.323,"y":0.329},"linkquality":15}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":91,"voltage":2985,"linkquality":47}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/temperatures1xiaomi', payload '{"temperature":23.64,"linkquality":15,"humidity":43.59,"pressure":1002,"battery":86,"voltage":2975}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/temperatures2xiaomi', payload '{"temperature":22.28,"linkquality":42,"humidity":48.29,"pressure":1003.9,"battery":100,"voltage":3005}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/temperatures3xiaomi', payload '{"temperature":21.42,"linkquality":55,"humidity":56.27,"pressure":1002,"battery":100,"voltage":3005}'
3/23/2019, 2:25:31 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":false}'
[... cut ...] e","friendlyName":"ledstrip1osram"},"message":"{\"state\": \"ON\", \"color\": {\"x\": 0.323, \"y\": 0.329}}"}}'
3/23/2019, 2:28:51 PM - info: Zigbee publish to device '0x8418260000caa3af', lightingColorCtrl - moveToColor - {"transtime":0,"colorx":21168,"colory":21561} - {"manufSpec":0,"disDefaultRsp":0} - null
3/23/2019, 2:28:52 PM - info: Zigbee publish to device '0x8418260000caa3af', lightingColorCtrl - read - [{"attrId":3},{"attrId":4},{"attrId":7}] - {"manufSpec":0,"disDefaultRsp":0} - null`

marcgarciamarti, 2019-03-23:
> @rgruebel I have not but knowing the problem still happens even after upgrading the stick puts me off.

@rgruebel any update? I just upgraded the stick today and encountered an error just by changing the light color

marcgarciamarti, 2019-03-23:
> I'm currently using the last stable firmware and I never experienced these problems.
> 
> Every day I turn it on and off, and sometimes I also change the color, but it always worked.

Can you confirm your device is recognized like this too? 
`ledstrip1osram (0x8418260000caa3af): 4052899926110 - OSRAM Flex RGBW (Router)`

marcgarciamarti, 2019-03-23:
> @rgruebel Can you re-pair the device?

@Koenkk note that I have not re-paired the OSRAM device after upgrading my usb stick. Let me know if you think I should do it.

Thanks for your hard work @Koenkk !

Koenkk, 2019-03-23:
Does it work when keeping it close to the coordinator? (also try re-pairing)

RyuzakiKK, 2019-03-23:
@marcgarciamarti yes, I have the same one `0x7cb03eaa0a000ef5|Router|OSRAM Flex RGBW (4052899926110)|online`

I just tried to turn it on and off multiple times from HA and it always worked https://paste.ubuntu.com/p/RWpvHkqJCJ/

marcgarciamarti, 2019-03-23:
@RyuzakiKK I also fired a bunch of ON and OFF actions and it worked like a charm (see the first part of this): https://hastebin.com/dekarafawu.makefile 

I think that upgrading the USB stick did address the on/off problem pretty good so far. 

However, when I swich the color it collapses (see tha last part of the same log file): https://hastebin.com/dekarafawu.makefile

marcgarciamarti, 2019-03-23:
> Does it work when keeping it close to the coordinator? (also try re-pairing)

Connectivity is good (the device sits very closely to the coordinator). Switching it ON and OFF seems to work well now. Switching colours isn't: https://hastebin.com/dekarafawu.makefile

RyuzakiKK, 2019-03-23:
I tried some more color changes and a couple of times I got an error https://paste.ubuntu.com/p/r2kv2rPvsd/

But keep in mind that my led strip is not near the coordinator so, at least in my case, it can be just a packet loss due to the poor connection link.

marcgarciamarti, 2019-03-23:
Is the syntax a bit different? (Sorry but I’m viewing this on my cell phone)

RyuzakiKK, 2019-03-23:
@marcgarciamarti mmm you are right. You get an error "205. No network route" and then a timeout.
On the other hand I get a "233. MAC no ack.", probably related to the not perfect connection quality.

marcgarciamarti, 2019-03-25:
@Koenkk could you kindly review this log? https://paste.ubuntu.com/p/NS6NVmVt4V/

After flashing the newest firmware on the stick three days ago, the OSRAM ledstripe worked well (ON/OFF - colour generates issues) for two days in a row. Unfortunatelly, today in the evening it refused to turn itself on based on my "on at Dusk".

The ledstripe is scheduled to turn on when it gets Dusk, but today it just didn't (you can check the log file today between 6PM and 8PM and you will see that no event was fired). If you review the Node-red automation I pasted right above, you will see that Node-Red checks the status, and if it is already on, the flow will simply die right there. 

And that is what happened today. Node-red checked the entity state and it did not proceed because, somehow, the entity in HA was registered as "still ON":

[image]

When I got home today around 8:30PM, the light was off but it appeared as on in the front-end. That's why I switched it off (it was already off in the real world), and I switched it no right after:

`3/25/2019, 8:37:42 PM - debug: Successfully pinged 0x00158d00022f35ca
3/25/2019, 8:38:27 PM - debug: Received MQTT message on 'zigbee2mqtt/ledstrip1osram/set' with data '{"state": "OFF"}'
3/25/2019, 8:38:28 PM - info: Zigbee publish to device '0x8418260000caa3af', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/25/2019, 8:38:28 PM - info: MQTT publish: topic 'zigbee2mqtt/ledstrip1osram', payload '{"state":"OFF","brightness":250,"color_temp":181,"color":{"x":0.14,"y":0.043},"linkquality":15}'
3/25/2019, 8:38:32 PM - debug: Received MQTT message on 'zigbee2mqtt/ledstrip1osram/set' with data '{"state": "ON"}'
3/25/2019, 8:38:32 PM - info: Zigbee publish to device '0x8418260000caa3af', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/25/2019, 8:38:33 PM - info: MQTT publish: topic 'zigbee2mqtt/ledstrip1osram', payload '{"state":"ON","brightness":250,"color_temp":181,"color":{"x":0.14,"y":0.043},"linkquality":15}'
3/25/2019, 8:38:42 PM - debug: Ping 0x00158d00022f35ca`

I would swear the problem starts here:

`3/24/2019, 11:29:42 PM - debug: Ping 0x00158d00022f35ca
3/24/2019, 11:29:42 PM - debug: Successfully pinged 0x00158d00022f35ca
3/24/2019, 11:30:00 PM - debug: Received MQTT message on 'zigbee2mqtt/ledstrip1osram/set' with data '{"state": "OFF"}'
3/24/2019, 11:30:00 PM - info: Zigbee publish to device '0x8418260000caa3af', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
3/24/2019, 11:30:30 PM - error: Zigbee publish to device '0x8418260000caa3af', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
3/24/2019, 11:30:30 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: Timed out after 30000 ms","meta":{"entity":{"ID":"0x8418260000caa3af","type":"device","friendlyName":"ledstrip1osram"},"message":"{\"state\": \"OFF\"}"}}'
3/24/2019, 11:30:41 PM - debug: Ping 0x00158d00022f35ca
3/24/2019, 11:30:42 PM - debug: Successfully pinged 0x00158d00022f35ca`

I may have a faulty ledstripe because I see a poor link quality, even if the ledstripe is close to where my Raspberry is located. I wonder if I may have a faulty device.

Koenkk, 2019-03-26:
@marcgarciamarti  It would be interesting to see a zigbee sniffing log when this error happens (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

marcgarciamarti, 2019-04-03:
@Koenkk my bad. I realize now that I need a new stick. Will order one tonight and will test it when I get it. In the meantime I can confirm that the newest firmware on the stick does work much better.

marcgarciamarti, 2019-05-11:
Hello @Koenkk , I finally got around to flashing the sniffer firmware on a brand new USB stick. I've followed all the steps contained in the tutorial referred by you. For the life of me I can't seem to be able to sniff usb traffic neither on my W10 nor on my Mac OS 10.13. I've spent like 3h in total with no success. This is probably not solely a zigbee2mqtt problem but rather how to use a Computer to capture traffic comming in throught the USB interface but I'm just not certain Wireshark is actually "seeing" the traffic inputing in the OS network stack I guess. 

Can you give me a hand even if, again, this is not per-se a zigbee2mqtt problem? 

See, if I open up Wireshark (snapshots are taken on a Mac OS but I could jump on a W10 if needed) I see these interfaces:
[image]

Apparently one needs a Nightly Builds in order to have Wireshark access the USB traffic: https://stackoverflow.com/questions/13827224/sniff-usb-traffic-on-a-mac 

but when I try to bring up the interface I see this:

```
Marcs-Air:~ marc$ sudo ifconfig XHC20 up
Password:
ifconfig: interface XHC20 does not exist

``` 

Note: all the time I see the USB stick's green led flashing, which I guess means the firmware is not the problem and is set to promiscuous mode

marcgarciamarti, 2019-05-11:
> Hello @Koenkk , I finally got around to flashing the sniffer firmware on a brand new USB stick. I've followed all the steps contained in the tutorial referred by you. For the life of me I can't seem [...]

Managed to solve the issues. I missread some part of the instructions. My bad! :(

@Koenkk could you please review the files I generated? I really hope they frame what the issue is. Cought the OSRAM ledstrip acting up on me and I hope I managed to ilustrate the behaviour. 

I uploaded 3 files here - a quick sniffing of the network and 2 specific files containing the command sent out from the coordinator and the response sent from the ledstrip: https://www.dropbox.com/sh/2x4z0zxwltvo44s/AACSLW-hnX3aZ7NOo9JiXASna?dl=0

Koenkk, 2019-05-14:
From the logs I don't see anything strange. 

- Could you try if it also happens with the latest firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin
- I also have a device which shows similar behaviour (30000 ms error), need to investigate what is going on here, but probably similar.

marcgarciamarti, 2019-05-18:
@koenkk, prior to upgrading my usb dongle, I just cloned it. I think I will use the spare one for a while and see if the behavior is still the same. In case the results are the same, I will upgrade my original dingle. Will report back.

Koenkk, 2019-05-21:
Can you check with the latest dev branch? I've done some fixes that caused my `30000 ms` errors.

marcgarciamarti, 2019-05-21:
I would swear that with my cloned stick it works more reliable but can definitely test the dev branch. I’m currently traveling and won’t be able to test it during the following 10 days or so. Can you indicate where I should pull the dev version from? I assume you refer to z2m http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html and not to the usb stick firmware right?

Koenkk, 2019-05-21:
Yes you can follow that link

marcgarciamarti, 2019-06-02:
@Koenkk just a quick update: so far I have not tampered with my setup anymore because the replacement dongle that I got, and used for when cloning my production dongle, seems to be working much much better than the original one that I used. So far, it has never failed on me that I've noticed. Will stick with the current dongle for a while. I know this sounds weird (it does at least to me) but it actually works.

Koenkk, 2019-06-03:
@marcgarciamarti good to hear, can this be closed?

marcgarciamarti, 2019-06-03:
@koenkk it sure can be closed. At lest my environment is working well.

closed 2019-06-03 by Koenkk