Koenkk/zigbee2mqtt #1057: Alt Firmware Flashing -> use CCLoader instead of CCLib

Domochip, 2019-02-10:
I tried CCLib to flash firmware, but got too much problems and flashing time took many hours

I discovered that using CCLoader that works very fine and flash took a few minutes...
https://github.com/RedBearLab/CCLoader

Koenkk, 2019-02-10:
Great, I think there is still a usecase for CCLib as CCLoader only seems to support windows.

Could you make a pr to update the documentation? (https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/information/alternative_flashing_methods.md#via-arduinoesp8266)

momo90-2, 2019-02-13:
Is ccloader working for a cc2531 or cc2530 stick?
Is it necessary to convert the hex file via hex2bin or can the bin file be used which is provided in the zip archives here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator?

momo90-2, 2019-02-15:
I have testet CCloader but it didnt work for me (although it should work according to the datasheet of cc254X and cc253X). What i did:

1. used srec_cat.exe to convert the .hex file from the archive from here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin to a bin file. srec_cat.exe warned that the hex is not in ascending address order and reordered it.
2. connected the cc2531 like described here https://github.com/RedBearLab/CCLoader, checked and measured the connection about 5 times. Connection was ok.
3. tried ccloader.exe and ccloader_x86_64.exe. 
4. tried a Arduino and (without logic level converter) and a nodemcu (changed the pin definitions in the sketch)

ccloader connected to the arduino but, didnt find the chip. Error message "no chip detected", which means that ccloader cant communicate to the chip via DD. Apart from that ccloader said that the bin is corrupt. Havent tried the files from here https://github.com/mtornblad/zstack-1.2.2a.44539/tree/master/CC2531 yet. Maybe it is a power problem. Eventually i am going to power the cc2531 via usb instead of 3,3. Could also be the problem mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/78#issuecomment-450737978

Possible solution/to be tested:
1. using Arduino Uno with logic level converter
2. using hex from here https://github.com/mtornblad/zstack-1.2.2a.44539/tree/master/CC2531
3. power cc2531 via usb

Nomsplease, 2019-02-18:
I got mine to flash without issue with CC_Loader, and it was much faster about 2-3 minutes.

I created [PR 9](https://github.com/Koenkk/zigbee2mqtt.io/pull/9) for this.

Koenkk, 2019-02-18:
PR is merged, thanks all.

closed 2019-02-18 by Koenkk

AS137430, 2019-02-19:
@Nomsplease : you flashed the SBL .bin version of the firmware with CCLoader from [here](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator)? By the way, CCLoader has Linux/Mac [version ](https://github.com/RedBearLab/CCLoader/pull/20)as well. This [fork](https://github.com/jeanniquini/CCLoader) works for ESP8266.

Nomsplease, 2019-02-19:
@AS137430 Yes that is the firmware in the updated documentation. I tested this with a Uno/Windows. If you can get it to work on Linux/Max and an ESP reliably please update that documentation with any other steps necessary. It never hurts to have more options. :)

AS137430, 2019-02-20:
@Nomsplease: Thanks for your clarifications on the bin file used. I already flashed my first CC2531 with the first Arduino/ESP method. But I am waiting for delivery of the second CC2531 (which is taking a while..)  and will definitely try the CCLoader method since it is much faster and report back if I am successful with the ESP8266 on Linux.

krikk, 2019-02-20:
i tried to flash my cc2531 with this new method (useing ESP8266, wemos D1 r2) and it seemed to work:

	D:\ccloader>CCLoader_x86_64.exe 3 CC2531ZNP-Prod.bin 0
	Copyright (c) 2013 RedBearLab.com
	CCLoader_x86_64.exe version 0.5
	Comport : COM3
	Bin file: CC2531ZNP-Prod.bin
	Device  : Default (e.g. UNO)

	Comport open!
	<Baud:115200> <data:8> <parity:none> <stopbit:1> <DTR:off> <RTS:off>

	File open!
	Block total: 468

	Enable transmission...
	Request sent already!
	/********************************************************************/
	* If there is no respond last for 3s, please press "Ctrl+C" to exit!
	* And pay attention to :
	*   1. The connection between computer and Arduino;
	*   2. The connection between Arduino and CC2540;
	*   3. Whether the device you using is Leonardo or not;
	*   4. Other unexpected errors.
	/********************************************************************/

	Waiting for respond from Arduino...

	Uploading firmware...

	1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53  54  55  56  57  58  59  60  61  62  63  64  65  66  67  68  69  70  71  72  73  74  75  76  77  78  79  80  81  82  83  84  85  86  87  88  89  90  91  92  93  94  95  96  97  98  99  100  101  102  103  104  105  106  107  108  109  110  111  112  113  114  115  116  117  118  119  120  121  122  123  124  125  126  127  128  129  130  131  132  133  134  135  136  137  138  139  140  141  142  143  144  145  146  147  148  149  150  151  152  153  154  155  156  157  158  159  160  161  162  163  164  165  166  167  168  169  170  171  172  173  174  175  176  177  178  179  180  181  182  183  184  185  186  187  188  189  190  191  192  193  194  195  196  197  198  199  200  201  202  203  204  205  206  207  208  209  210  211  212  213  214  215  216  217  218  219  220  221  222  223  224  225  226  227  228  229  230  231  232  233  234  235  236  237  238  239  240  241  242  243  244  245  246  247  248  249  250  251  252  253  254  255  256  257  258  259  260  261  262  263  264  265  266  267  268  269  270  271  272  273  274  275  276  277  278  279  280  281  282  283  284  285  286  287  288  289  290  291  292  293  294  295  296  297  298  299  300  301  302  303  304  305  306  307  308  309  310  311  312  313  314  315  316  317  318  319  320  321  322  323  324  325  326  327  328  329  330  331  332  333  334  335  336  337  338  339  340  341  342  343  344  345  346  347  348  349  350  351  352  353  354  355  356  357  358  359  360  361  362  363  364  365  366  367  368  369  370  371  372  373  374  375  376  377  378  379  380  381  382  383  384  385  386  387  388  389  390  391  392  393  394  395  396  397  398  399  400  401  402  403  404  405  406  407  408  409  410  411  412  413  414  415  416  417  418  419  420  421  422  423  424  425  426  427  428  429  430  431  432  433  434  435  436  437  438  439  440  441  442  443  444  445  446  447  448  449  450  451  452  453  454  455  456  457  458  459  460  461  462  463  464  465  466  467  468  Upload successfully!
	File closed!
	Comport closed!

but after "flashing" the stick seems to be dead... the led does no longer flash and i do not get a device if i connect it to the pc/raspberrypi :(

AS137430, 2019-02-22:
> but after "flashing" the stick seems to be dead... the led does no longer flash and i do not get a device if i connect it to the pc/raspberrypi :(

@Nomsplease : Did you have any similar issue? 
#320 
Is the SBL bin firmware supposed to be flashed only through SerialBootLoader??

kianusch, 2019-03-02:
After playing around a bit, a few thoughts ...

Using the bin file from zigbee2mqtt repository will not work.
But you can convert the hex file to a bin file - under linux the command is:

$ objcopy --gap-fill 0xFF --pad-to 0x040000 -I ihex CC2531ZNP-Prod.hex -O binary CC2531ZNP-Prod.bin

$ ls -lt CC2531ZNP-Prod.*
-rw-rw-r-- 1 kianusch kianusch 262144 Mar  2 13:59 CC2531ZNP-Prod.bin
-rw-r--r-- 1 kianusch kianusch 697099 Jan 29 19:23 CC2531ZNP-Prod.hex

objcopy is part of binutils package.

The resulting file can be passed to CCLoader for flashing:

$ /opt/CCLoader/SourceCode/Linux$ ./a.out  /dev/ttyACM0 /tmp/CC2531ZNP-Prod.bin 1
Comport open:
Device: Leonardo

Baud:115200 data:8 parity:none stopbit:1 DTR:on RTS:off
File open success!
Warning: file size isn't the integer multiples of 512, last bytes will be set to 0xFF
Block total: 512
Enable transmission...
Request sent already! Waiting for respond...
Begin programming...
1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53  54  55  56  57  58  59  60  61  62  63  64  65  66  67  68  69  70  71  72  73  74  75  76  77  78  79  80  81  82  83  84  85  86  87  88  89  90  91  92  93  94  95  96  97  98  99  100  101  102  103  104  105  106  107  108  109  110  111  112  113  114  115  116  117  118  119  120  121  122  123  124  125  126  127  128  129  130  131  132  133  134  135  136  137  138  139  140  141  142  143  144  145  146  147  148  149  150  151  152  153  154  155  156  157  158  159  160  161  162  163  164  165  166  167  168  169  170  171  172  173  174  175  176  177  178  179  180  181  182  183  184  185  186  187  188  189  190  191  192  193  194  195  196  197  198  199  200  201  202  203  204  205  206  207  208  209  210  211  212  213  214  215  216  217  218  219  220  221  222  223  224  225  226  227  228  229  230  231  232  233  234  235  236  237  238  239  240  241  242  243  244  245  246  247  248  249  250  251  252  253  254  255  256  257  258  259  260  261  262  263  264  265  266  267  268  269  270  271  272  273  274  275  276  277  278  279  280  281  282  283  284  285  286  287  288  289  290  291  292  293  294  295  296  297  298  299  300  301  302  303  304  305  306  307  308  309  310  311  312  313  314  315  316  317  318  319  320  321  322  323  324  325  326  327  328  329  330  331  332  333  334  335  336  337  338  339  340  341  342  343  344  345  346  347  348  349  350  351  352  353  354  355  356  357  358  359  360  361  362  363  364  365  366  367  368  369  370  371  372  373  374  375  376  377  378  379  380  381  382  383  384  385  386  387  388  389  390  391  392  393  394  395  396  397  398  399  400  401  402  403  404  405  406  407  408  409  410  411  412  413  414  415  416  417  418  419  420  421  422  423  424  425  426  427  428  429  430  431  432  433  434  435  436  437  438  439  440  441  442  443  444  445  446  447  448  449  450  451  452  453  454  455  456  457  458  459  460  461  462  463  464  465  466  467  468  469  470  471  472  473  474  475  476  477  478  479  480  481  482  483  484  485  486  487  488  489  490  491  492  493  494  495  496  497  498  499  500  501  502  503  504  505  506  507  508  509  510  511  512  Program successfully!
File closed!
Comport closed!

If CCLoader gets stuck at "Request sent already! Waiting for respond..." press CTRL-C and try  changing the last parameter from 1 to 0 ... (also check your wiring)

$ /opt/CCLoader/SourceCode/Linux$ ./a.out  /dev/ttyACM0 /tmp/CC2531ZNP-Prod.bin 0

This will set DTR to "off" during initialization phase.

Also - I adapted my CCLoader.inc and changed the pinout to have the pins in a row:

int DD = 2;
int DC = 3;
int RESET = 4;

This makes wiring more easy.

[image]

[image]

Hope this helps.

AS137430, 2019-03-02:
@kianusch: Thanks for the detailed instructions especially the objcopy command. Is that an Arduino Pro Micro you are using? You are using GND and pins 2-4 from what I can make out for the picture. How is CC2531 powered?

kianusch, 2019-03-02:
Yep - this is a Pro-Micro -I know - I soldered the pins on the wrong side :) 
But Arduino UNO works as well.

And - yes:

Arduino -- CC2531
- GND -- GND
- Pin 2 -- DD
- Pin 3 -- DC
- Pin 4 -- RESET

I  powered the CC2531 via USB - after plugging in the Arduino, I plugged in the CC2531.  But you just have to the correct (Arduino)  device for CCLoader .

3.3V via Arduino UNO to pin 9 on CC2531 did not work for me - don't know why.

AS137430, 2019-03-03:
@kianusch : Thanks. Will you raise a PR to update the wiki [https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html](url)?
@Koenkk : There are now more details on the CCLoader method for flashing. Please take a look, thanks.

kianusch, 2019-03-03:
Sorry - I have no idea how to raise a PR.
I actually don't now what you mean by PR. :/
Priority?

AS137430, 2019-03-03:
@kianusch : A Github Pull Request (PR). Similar to [https://github.com/Koenkk/zigbee2mqtt.io/pull/9](url).

AS137430, 2019-03-06:
If you need Windows version of objcopy, you can try MinGW's [binutils](https://osdn.net/projects/mingw/releases/p15608) includes objcopy.exe

AS137430, 2019-03-07:
For Mac users:
[https://brewinstall.org/install-binutils-on-mac-with-brew](binutils)
[https://github.com/hoiberg/CCLoaderX/releases](CCLoaderX)

turboproc, 2019-03-08:
For those with limited time available: I can confirm that the CCLoader route works in a flash (;-)). Spend a day on the CCLib route but couldn;t get to reliable uploading, although the communication through the chain worked as such.

ptath, 2019-03-22:
> After playing around a bit, a few thoughts ...

> $ /opt/CCLoader/SourceCode/Linux$ ./a.out /dev/ttyACM0 /tmp/CC2531ZNP-Prod.bin 1

Trying to do this (nodemcu v3), but still 

```
MBP:CCLoader-master ptath$ ./a.out /dev/cu.wchusbserial640 CC2531ZNP-Prod.bin 1
Opening comport...
Comport open!
Read portsettings...
Changing portsettings...
Baud:115200 data:8 parity:none stopbit:1 DTR:off RTS:off
File open success!
Block total: 512
Waiting for Arduino setup...
Remain: 5
Remain: 4
Remain: 3
Remain: 2
Remain: 1
Enable transmission...
Request sent already!Waiting for respond...
Begin programming...
^C
```

If pins changed to 

```
// Debug control pins & the indicate LED
int DD = 3;
int DC = 4;
int RESET = 5;
int LED = 13;
```

I get another result:

```
Enable transmission...
Request sent already!Waiting for respond...
Begin programming...
1  2  3  4  5  6  7  8  9  Verify failed!
Program successfully!
File closed!
Comport closed!
Exit: 0
```

Any ideas please?

kianusch, 2019-03-23:
I haven't tried programming via ESP myself ... but you might have a problem with the pin numbering ... 

Maybe this helps:

https://github.com/esp8266/Arduino/issues/584

kianusch, 2019-03-23:
@ptath 

[image]

AS137430, 2019-03-25:
[Does this help?](https://github.com/Koenkk/zigbee2mqtt/issues/78#issuecomment-475919087)

jackwilsdon, 2019-07-21:
I can confirm that I needed to use @kianusch's `objdump` command to get CCLoader to flash correctly on macOS - trying the normal `.bin` from the repository causes the device to not work, flashing the objdump'd bin works fine.

Hedda, 2020-06-08:
FYI, also see https://github.com/Koenkk/zigbee2mqtt/issues/3694 for another alternative method of firmware flashing.