Koenkk/zigbee2mqtt #1059: Network has no routes, nothing no longer works

philhawthorne, 2019-02-10:
I've been having several stability issues with zigbee2mqtt, and have needed to reboot the Pi running zigbee2mqtt for control to be restored.

This evening, zigbee2mqtt randomly stopped being able to control devices again. I did the usual reboot of the Pi, but I am unable to get zigbee2mqtt to communicate with anything now.

After booting up, any request to control a device is met with `null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.`

Here is my new network map, which shows no routes to anything. Previously there was a nice healthy mesh going on.

```
digraph G {
node[shape=record];
  "0x00124b0014d9e3a4" [style="bold", label="{0x00124b0014d9e3a4|Coordinator|No model information available|online}"];
  "0x0017880100bfbb52" [style="rounded", label="{Study Lamp|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x0017880102e05fa5" [style="rounded", label="{Plant Light|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x001788010463fa8f" [style="rounded", label="{Wardrobe Light|Router|Philips Hue White Single bulb B22 (8718696449691)|offline}"];
  "0xf0d1b80000009792" [style="rounded", label="{Ensuite Sink|Router|Sylvania Smart Home adjustable white MR16 LED bulb (74282)|offline}"];
  "0xf0d1b80000009f45" [style="rounded", label="{Ensuite Shower|Router|Sylvania Smart Home adjustable white MR16 LED bulb (74282)|offline}"];
  "0x0017880100b623e9" [style="rounded", label="{Office Spotlight|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x00158d00012904a9" [style="rounded, dashed", label="{Bathroom Switch|EndDevice|Xiaomi Aqara single key wireless wall switch (WXKG03LM)|online}"];
  "0x001788010361e509" [style="rounded", label="{Sconce Left|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x0017880103407970" [style="rounded", label="{Sconce Right|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x00158d0001a1fbb1" [style="rounded, dashed", label="{Coffee Table|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x0017880110460d3f" [style="rounded", label="{Living Room Lamp Bottom|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x0017880100ecee59" [style="rounded", label="{Living Room Lamp Top|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x00158d0001a360cd" [style="rounded, dashed", label="{Kitchen Wall|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d000236f9b5" [style="rounded, dashed", label="{Dishwasher Door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d0001a1f92d" [style="rounded, dashed", label="{Bedroom Wall|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x0017880100ecec5d" [style="rounded", label="{Phils Lamp|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x0017880100ecef80" [style="rounded", label="{Helens Lamp|Router|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|offline}"];
  "0x00158d000213168b" [style="rounded, dashed", label="{Switch Helen|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x0017880100c51807" [style="rounded", label="{Bookcase Right Bloom|Router|Philips Hue Bloom (7299760PH)|offline}"];
  "0x0017880100c26f60" [style="rounded", label="{Master Bedroom Bloom|Router|Philips Hue Bloom (7299760PH)|offline}"];
  "0x00158d000163b9e1" [style="rounded, dashed", label="{Switch Phil|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00178801011bf6d9" [style="rounded", label="{Kitchen Lightstrip|Router|Philips Hue white and color ambiance LightStrip plus (915005106701)|offline}"];
  "0x00158d0001dbd396" [style="rounded, dashed", label="{Ensuite Door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d000236fa31" [style="rounded, dashed", label="{Wardobe Door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d00015ef2f4" [style="rounded, dashed", label="{Switch Bedroom|EndDevice|Xiaomi Aqara single key wireless wall switch (WXKG03LM)|online}"];
  "0xf0d1b80000009f42" [style="rounded", label="{Office South|Router|Sylvania Smart Home adjustable white MR16 LED bulb (74282)|offline}"];
  "0xf0d1b800000096eb" [style="rounded", label="{Office North|Router|Sylvania Smart Home adjustable white MR16 LED bulb (74282)|offline}"];
  "0x00158d0002366cc0" [style="rounded", label="{Wax Melter|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
  "0x0017880100c085af" [style="rounded", label="{Bookcase Left Bloom|Router|Philips Hue Bloom (7299760PH)|offline}"];
  "0x00158d0001dbc903" [style="rounded, dashed", label="{Bathroom Door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d0001ab2037" [style="rounded, dashed", label="{Toilet Door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"];
  "0x00158d00023332b2" [style="rounded, dashed", label="{Ensuite Shower Water|EndDevice|Xiaomi Aqara water leak sensor (SJCGQ11LM)|online}"];
  "0xf0d1b80000009d3f" [style="rounded", label="{Entry|Router|Sylvania Smart Home adjustable white MR16 LED bulb (74282)|offline}"];
  "0x00158d0002365b11" [style="rounded", label="{Rangehood|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
}
```

I've tried several things to try and recover

- Turned off all powered zigbee devices (routers), then started zigbee2mqtt
- Unplugged the CC2531 device from the Pi
- Plugged the USB device into a different port on the Pi
- Pressed all the buttons on the CC2531 hoping it would fix it
- Shutdown the Pi, and then started from a cold boot
- Upgraded to the latest `dev` branch of zigbee2mqtt, but still no luck
- Placed some foil over the CC253X radio to shield the radio during startup.

Unfortunately it seems as though the logs of when this first happened have been cycled, so I don't have them now. However it seems as though the CC253X device just stops being able to communicate with anything. As mentioned usually a reboot of the Pi is enough to fix it, but now I can't get zigbee2mqtt tom communicate with anything.

The last time this happened, I had to re-flash my USB stick, and then re-pair all my devices again. I'm hoping there might be some ideas I can try here to recover my network, before having to nuke it and start again. I'd also like to not have to do this again in the future, so if there's anything I can do to help track this down, please let me know.

The router firmware I've been using (with group support) is CC2531ZNP-Prod_20181224. I've got some logs from my startup here: https://hastebin.com/oriqohayod.makefile

Any help or suggestions are appreciated!

Koenkk, 2019-02-10:
My guess is that your panid has been changed due to the crash. Ive also experienced this a few days ago, need to investigate why. You can reflash your stick, if you manage to get your old panid back (probably 0x1a62) everything should work again. See https://koenkk.github.io/zigbee2mqtt/information/flashing_without_re-pairing.html

mhelff, 2019-02-10:
@Koenkk couldnt the right panid be set through configuration (advanced/pan_id) to avoid these problems when reflashing?

Koenkk, 2019-02-10:
@mhelff even when configured, the pan_id is not honoured when the pan_id is already used by an other zigbee network.

philhawthorne, 2019-02-10:
Damn but that actually might make sense.

When reflashing I'm guessing there is a way to reset the pan ID back to its
original?

Koenkk, 2019-02-10:
Reflashing is even not needed. 

- First change the `pan_id` to a random one, set in configuration e.g `pan_id`: 0x1a67. Start zigbee2mqtt.
- Stop zigbee2mqtt
- Now change the `pan_id` back to the `pan_id` you want it to use (e.g. `0x1a62`), and start zigbee2mqtt, but make sure you are out of the range of your network! (like described in https://koenkk.github.io/zigbee2mqtt/information/flashing_without_re-pairing.html)

philhawthorne, 2019-02-10:
Oh really? Ok great I will give that a try later!

Appreciate the help, and zigbee2mqtt!

philhawthorne, 2019-02-11:
@Koenkk thanks for your help. Manually setting the `pan_id` has brought everything back online! Cheers

closed 2019-02-11 by philhawthorne