Koenkk/zigbee2mqtt #1060: Is it best to pair to coordinator or router?

b4dpxl, 2019-02-10:
I started my zigbee2mqtt network with just a Xiaomi WXKG02LM switch for testing. Happy with this I have now added 2 of my Tradfri bulbs to the network. These are showing up as a router, but the switch is still connected to the coordinator, although I'd expect it to get better signal from one of the bulbs.

My question is 2-fold. When adding a new device, e.g. I want to add additional switches downstairs, away from the coordinator but near to one of the bulbs, is is best to pair to the bulb nearest them instead of the coordinator?

Secondly, if that bulb/router is unavailable, will the devices find another one to communicate with? Or jump to a router with a better signal?

Apologies if I've missed something in the docs. My network looks like this at the moment:

[image]

Thanks

clockbrain, 2019-02-11:
End devices tend to stay with the coordinator or router they were initially paired with. They don't roam to where the best signal is. In theory they should switch to a different router if they one they paired with becomes unavailable but in practice this doesn't work so well, notably for Xioami devices.

As to whether it is best to pair with coordinator or router - depends on your planned network. Pairing to coordinator is simplest but you may hit the firmware number of devices limit in which case add routers and shift (re-pair) end devices to them. Routers can give you greater range if wherever you want the end device to be is far from the coordinator. A consideration is whether your router is something that may unexpectedly drop off the network, e.g. a Hue bulb that another family member tends to switch off at times!

P.S. looks from your network map that you aren't running the latest version of zigbee2mqtt.

b4dpxl, 2019-02-11:
Thanks @clockbrain. Fortunately my bulbs should be safe from accidental switch off; the Xiaomi switches I'm using are in part to control the Ikea bulbs that will act as routers, and I've printed mounts so they cover the real switches. 

Yeah, I need to update.

Vout, 2019-02-12:
> As to whether it is best to pair with coordinator or router - depends on your planned network. Pairing to coordinator is simplest but you may hit the firmware number of devices limit in which case [...]

Thing is that 98 out of 100 times devices won't pair through an router. I tried but it didn't work out, i had to pair all my battery powered devices through my coordinator and then had to relocate them because pairing through an router didn't work.

Currently i have 10 bulbs, 5 switches and 2 motions sensors and 2 power plugs connected through Zigbee2Mqtt, probably i will hitt the device limit pretty soon because of the need to pair battery powered devices on the coordinator.

Maybe something to look at, the possibility to pair devices to an router instead  of pairing with an coordinator only.

nanosonde, 2019-02-15:
@clockbrain 
Concerning roaming and Xiaomi: I have bought 4 Aqara Temp./Hum./Press. sensors recently.
My routers: Ikea Tradfri Repeater and Ikea Tradfri Driver 10W. 
The mentioned sensors happily move around those routers and even to the coordinator when signal becomes to weak.
The devices initially paired to the Tradfri Driver.

marc-gist, 2019-02-26:
> End devices tend to stay with the coordinator or router they were initially paired with. They don't roam to where the best signal is. In theory they should switch to a different router if they one [...]

How do you force a pair with a Router rather than the coordinator? and how do you remove Xiaomi devices from the Coordinator to make room for a router?

closed 2019-05-04 by stale[bot]