Koenkk/zigbee2mqtt #1062: Gledopto GL-B-007Z and GL-B-008Z don't respect "transition" parameter

milakov, 2019-02-10:
Hi, whatever parameter I specify for "transition" it is not used, both bulbs are switched on smoothly ~0.2ms and swicthed off sharp, with visibly no transition.

Koenkk, 2019-02-10:
Can you post your log when executing this transition? (with `log_level: debug` https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html)

milakov, 2019-02-10:
Here it is, sorry for not posting this in my initial post.

Switching off:
```
  zigbee2mqtt:debug 2019-2-10 22:20:06 Received MQTT message on 'zigbee2mqtt/0x00124b001b50416b/set' with data '{"state": "OFF", "transition": 500}'
  zigbee2mqtt:info 2019-2-10 22:20:06 Zigbee publish to device '0x00124b001b50416b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 11
  zigbee2mqtt:info 2019-2-10 22:20:06 MQTT publish: topic 'zigbee2mqtt/0x00124b001b50416b', payload '{"state":"OFF","color_temp":278,"color":{"x":0.323,"y":0.329},"last_seen":"2019-02-09T12:51:40.592Z","brightness":255}'
```

Switching on:
```
  zigbee2mqtt:debug 2019-2-10 22:20:43 Received MQTT message on 'zigbee2mqtt/0x00124b001b50416b/set' with data '{"state": "ON", "transition": 1000}'
  zigbee2mqtt:info 2019-2-10 22:20:43 Zigbee publish to device '0x00124b001b50416b', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 11
  zigbee2mqtt:info 2019-2-10 22:20:43 MQTT publish: topic 'zigbee2mqtt/0x00124b001b50416b', payload '{"state":"ON","color_temp":278,"color":{"x":0.323,"y":0.329},"last_seen":"2019-02-09T12:51:40.592Z","brightness":255}'
```

Koenkk, 2019-02-10:
on/off doesn't support transition as this is binary. Transition can only be used with brightness, color and color temperature.

milakov, 2019-02-10:
Just for my education, where is this limitation? In Zigbee protocol, Gledopto implementation, or the current limitations of zigbee2mqtt? HA in general supports it, Yeelight wi-fi lamps are capable of turning on and off with transition (not perfect though).

Koenkk, 2019-02-10:
For brightness it is actually supported, send e.g.:

On:
```js
{
  "state": "on",
  "brightness": 255,
  "transition": 3
}
```

Off
```js
{
  "state": "off",
  "brightness": 0,
  "transition": 3
}
```

milakov, 2019-02-11:
There are 2 issues here:

1) What would be the way to switch on the lamp to the target brightness? Calling ```light.turn_on``` with the following does nothing:
```
{
  "entity_id": "light.gledopto_desklamp",
  "brightness":"255"
}
```
The lamp stays switched off.

And if I try to appy transition to change brgightness:
```
{
  "entity_id": "light.gledopto_desklamp",
  "brightness":"255",
  "transition":"3"
}
```
it has no effect:
```
  zigbee2mqtt:debug 2019-2-11 17:57:49 Received MQTT message on 'zigbee2mqtt/0x00124b001b50c0a6/set' with data '{"state": "ON", "transition": 3, "brightness": 255}'
  zigbee2mqtt:debug 2019-2-11 17:57:49 Skipping 'state' because of 'brightness'
  zigbee2mqtt:info 2019-2-11 17:57:49 Zigbee publish to device '0x00124b001b50c0a6', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":98.99999999999999} - {"manufSpec":0,"disDefaultRsp":0} - 11
```
The transition certainly takes < 3 seconds.

Koenkk, 2019-02-11:
To switch on the bulb to the target brightness you should send over MQTT:
```js
{
  "brightness": 100,
  "transition": 20
}
```

This will result in an `moveToLevelWithOnOff` which is the Zigbee command to turn on a light with a given brightness. The `transition` is also supported. So the above will turn the light from on the off with a transition to `100` brightness in 20 seconds.

When the bulb is off, and sending over MQTT:

```js
{
  "brightness": 100
}
```
Does the bulb turn on? If not, this indicates that this device has no support for `moveToLevelWithOnOff`.

milakov, 2019-02-11:
```
  zigbee2mqtt:debug 2019-2-11 22:36:17 Received MQTT message on 'zigbee2mqtt/0x00124b001b50c0a6/set' with data '{
  "brightness": 100
}'
  zigbee2mqtt:info 2019-2-11 22:36:17 Zigbee publish to device '0x00124b001b50c0a6', genLevelCtrl - moveToLevelWithOnOff - {"level":100,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
```
And the bulb doesn't switch on. That seems to indicate that the device doesn't support moveToLevelWithOnOff?

But the transition doesn't work at all, even if changing brightness for the already switched on lamp:
```
  zigbee2mqtt:debug 2019-2-11 22:40:30 Received MQTT message on 'zigbee2mqtt/0x00124b001b50c0a6/set' with data '{
  "brightness": 10,
  "transition": 20
}'
  zigbee2mqtt:info 2019-2-11 22:40:30 Zigbee publish to device '0x00124b001b50c0a6', genLevelCtrl - moveToLevelWithOnOff - {"level":10,"transtime":660} - {"manufSpec":0,"disDefaultRsp":0} - 11
```
While I am asking for 20 seconds transition it does fraction of a second one.

Koenkk, 2019-02-12:
I guess that this device just doens't support transitions, however this is strage, as in #1047 there is a `GL-B-008Z` users which has this working.

milakov, 2019-02-12:
Nope, it doesn't work for me both with 007 and 008 bulbs. I will probably play with packet sniffer!

milakov, 2019-02-12:
Okay, transition time successfully goes to the bulb via Zigbee, and the bulb reports success, but 1) it doesn't respect transition time, and 2) moveToLevelWithOnOff does not switch on the lamp:
[image]
[image]
[image]
I am closing this issue, thanks for your help @Koenkk !

closed 2019-02-12 by milakov

kevincaradant, 2021-01-08:
Hi guys

Just to understand bacause all my GLEODOPTO bulbs x20 => GL-B-007Z => GL-B-001Z  => GL-S-007Z => GL-C-007-1ID have this issue. 
I'm not sure to understand if the transition time is available on the binary state ON/OFF.
The transition works if I ask it when the device is already turn on.

When I try to specify a brightness and transition time during TURN_ON action, my bulbs crashed.
I mean by that when I say "TURN ON" nothing happens and the bulb doesn't "turn on" but the state is "ON" on Z2M.
So what should I understand by that ? 

Thank you :)

lenny-duckowsky, 2021-01-15:
I'm experiencing the same problem.
Zigbee2Mqtt ver. 1.7.0.1 installed on Home Assistant 2020.12.1 and managing a USB CC2531.
I have two Gledopto dimmer switches (model GL-C-009) that work perfectly with in terms of ON and OFF but I cannot manage the brightness transition time as described.
By publishing the following mqtt message
`topic: zigbee2mqtt/0x00124b0021f2ab8c/set
payload: |-
  {
   "brightness": "220",
   "transition": "120"
  }`
the dimmer go to the desired brigthness level (220) but not in 120 seconds, it just fades to 220 in 1 second.
In the Zigbee2mqtt log I see the message received but the message send to the dimmer miss the transition value
`Zigbee2MQTT:debug 2021-01-15 09:18:29: Received MQTT message on 'zigbee2mqtt/0x00124b0021f2ab8c/set' with data '{ "brightness": "220",
  "transition": "120"
}'`
`Zigbee2MQTT:debug 2021-01-15 09:18:29: Publishing 'set' 'brightness' to 'Luci Acquario Spettro Alba/Tramonto'`
`Zigbee2MQTT:info  2021-01-15 09:18:29: MQTT publish: topic 'zigbee2mqtt/Luci Acquario Spettro Alba/Tramonto', payload '{"brightness":220,"linkquality":57,"state":"ON"}'`
`Zigbee2MQTT:debug 2021-01-15 09:18:29: Publishing 'set' 'transition' to 'Luci Acquario Spettro Alba/Tramonto'`

How is it possible?