Koenkk/zigbee2mqtt #1067: error starting zigbee2mqtt

sander816, 2019-02-11:
i just did a git pull on the latest zigbee2mqtt, because my aldi e14 were not working correct in domoticz:
using this guide:
https://koenkk.github.io/zigbee2mqtt/getting_started/running_zigbee2mqtt.html

now my zigbee2mqtt fails to start:

```
pi@rpi3:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.1.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2/11/2019, 9:03:42 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-02-11.09-03-42'
  zigbee2mqtt:info 2/11/2019, 9:03:43 AM Starting zigbee2mqtt version 1.1.1 (commit #5554579)
  zigbee2mqtt:info 2/11/2019, 9:03:43 AM Starting zigbee-shepherd
  zigbee2mqtt:info 2/11/2019, 9:03:46 AM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2/11/2019, 9:04:46 AM Starting zigbee-shepherd
  zigbee2mqtt:error 2/11/2019, 9:04:49 AM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2/11/2019, 9:04:49 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2/11/2019, 9:04:49 AM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}

----

2/11/2019, 9:07:50 AM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-02-11.09-07-50'
2/11/2019, 9:07:51 AM - info: Starting zigbee2mqtt version 1.1.1 (commit #5554579)
2/11/2019, 9:07:51 AM - info: Starting zigbee-shepherd
2/11/2019, 9:07:54 AM - info: Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2/11/2019, 9:08:54 AM - info: Starting zigbee-shepherd
2/11/2019, 9:08:57 AM - error: Error while starting zigbee-shepherd!
2/11/2019, 9:08:57 AM - error: Press the reset button on the stick (the one closest to the USB) and start again
2/11/2019, 9:08:57 AM - error: Failed to start Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:273:13)
    at CcZnp.emit (events.js:182:13)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
2/11/2019, 9:09:02 AM - error: Not connected to MQTT server!

```
mqtt server is running:

```
pi@rpi3:/opt/zigbee2mqtt/data/log$ service mosquitto status
● mosquitto.service - LSB: mosquitto MQTT v3.1 message broker
   Loaded: loaded (/etc/init.d/mosquitto; generated; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-11 08:38:08 CET; 31min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 362 ExecStart=/etc/init.d/mosquitto start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mosquitto.service
           └─435 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

Feb 11 08:38:07 rpi3 systemd[1]: Starting LSB: mosquitto MQTT v3.1 message broker...
Feb 11 08:38:08 rpi3 mosquitto[362]: Starting network daemon:: mosquitto.
Feb 11 08:38:08 rpi3 systemd[1]: Started LSB: mosquitto MQTT v3.1 message broker.
pi@rpi3:/opt/zigbee2mqtt/data/log$

```

i already pressed the button closest to the usb, no result

Koenkk, 2019-02-11:
- When downgrading does it work again?
- If not: stop zigbee2mqtt, re-plug stick, press the reset button (the one close to the USB port) green light should go off, start zigbee2mqtt

sander816, 2019-02-11:
when i press that button closest to the usbm, nothing happens with the green led

Koenkk, 2019-02-11:
When downgrading, does it work again?

sander816, 2019-02-11:
oddly zigbee2mqtt has started to work again? no idea what went wrong. Maybe my rpi3 is a bit slow in discovering the devices i have?

Koenkk, 2019-02-11:
This is indeed strange, it somehow failed to communicate with the CC2531.

closed 2019-02-11 by Koenkk

sander816, 2019-02-11:
error is back:
i cannot get it to run again

`pi@rpi3:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.1.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2/11/2019, 9:02:26 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-02-11.21-02-26'
  zigbee2mqtt:info 2/11/2019, 9:02:27 PM Starting zigbee2mqtt version 1.1.1 (commit #5554579)
  zigbee2mqtt:info 2/11/2019, 9:02:27 PM Starting zigbee-shepherd
  zigbee2mqtt:info 2/11/2019, 9:02:30 PM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

  zigbee2mqtt:info 2/11/2019, 9:03:30 PM Starting zigbee-shepherd
  zigbee2mqtt:error 2/11/2019, 9:03:33 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2/11/2019, 9:03:33 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2/11/2019, 9:03:33 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}

`

sander816, 2019-02-11:
after pressing ctrl-c i get this:

```

^[[A
^C  zigbee2mqtt:error 2/11/2019, 9:05:08 PM Not connected to MQTT server!
  zigbee2mqtt:error 2/11/2019, 9:05:08 PM Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
  zigbee2mqtt:info 2/11/2019, 9:05:08 PM zigbee-shepherd stopped
```

i think it means that mqtt server is not running?

but that is running fine
```
pi@rpi3:/opt/zigbee2mqtt $ service mosquitto status
● mosquitto.service - LSB: mosquitto MQTT v3.1 message broker
   Loaded: loaded (/etc/init.d/mosquitto; generated; vendor preset: enabled)
   Active: active (running) since Mon 2019-02-11 20:57:26 CET; 8min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 364 ExecStart=/etc/init.d/mosquitto start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mosquitto.service
           └─432 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

Feb 11 20:57:25 rpi3 systemd[1]: Starting LSB: mosquitto MQTT v3.1 message broker...
Feb 11 20:57:26 rpi3 mosquitto[364]: Starting network daemon:: mosquitto.
Feb 11 20:57:26 rpi3 systemd[1]: Started LSB: mosquitto MQTT v3.1 message broker.

```

Koenkk, 2019-02-12:
Due to the ctrl-c it disconnects perhaps it's some race condition, however this isn't the root cause of your issue, let's continue in #937