Koenkk/zigbee2mqtt #1069: Question: Benifit of zigbee2mqtt?

momo90-2, 2019-02-11:
Hello!

First of all please excuse my bad english!

I really tried to find an answer to my question before posting, but i didnt find any.

I have ordered a CC2531 usb stick and want to add several Xiaomi sensors. Maybe later I am going to order some tradfri, osram or other devices. I am using Openhab.

As far as I understood there are two possibilities after flashing the stick with the right firmware:

    use the OH zigbee binding
    use zigbee2mqtt

All of the mentioned devices should work with both variants. So, what ist the benifit of zigbee2mqtt?

Thanks for your help!

rgruebel, 2019-02-12:
for example if you switch from openhab to something else

momo90-2, 2019-02-12:
Well, that makes sense!

Thank You!

closed 2019-02-12 by Koenkk