Koenkk/zigbee2mqtt #1071: Adding new device: LivingWise (Orvibo) Dimmer Switch

dev-abo, 2019-02-12:
I'm trying to add LivingWise Smart Dimmer (https://www.primecables.ca/p-368972-lvs-zb500d-neutral-zigbee-smart-dimmerus-type-120v500va) as a new device, but having issues with receiving the status.  Also i could find the same model ID as Orvibo Smart Dimmer (https://www.orvibo.com/en/product/us_dimmerswitch.html). 

After adding the model to devices.js i was able to pair and control it successfully (On/Off/Set brightness), but cannot receive any status updates. 

devices.js:
```
	{
        zigbeeModel: ['abb71ca5fe1846f185cfbda554046cce'],
        model: 'abb71ca5fe1846f185cfbda554046cce',
        vendor: 'LivingWise',
        description: 'Dimmer',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: [],
        toZigbee: generic.light_onoff_brightness().toZigbee,
    },
```

Wireshark can catch the status updates, but none of them appears in zigbee2mqtt logs: [LivingWise.pcapng](https://github.com/dev-abo/sharedfiles/raw/master/LivingWise.pcapng)

I believe, i need to add `configure:` section to the devices.js, but none of the existing examples worked. 
I'd appreciate any help with this.

khzd, 2019-02-12:
Hi
Did you fill this section : fromZigbee: [], ?

dev-abo, 2019-02-12:
Hi, 

Yes, i did try to use `fromZigbee: generic.light_onoff_brightness.fromZigbee,`, but could see no incoming messages.

Koenkk, 2019-02-12:
Can you control the device? (by sending to `zigbee2mqtt/[DEVICE_ID]/set` `ON`)

dev-abo, 2019-02-12:
yes,
i'm able to control the switch: turn On/Off and set brightness, but do not get any status updates from it.

Koenkk, 2019-02-12:
Try with
```js
{
        zigbeeModel: ['abb71ca5fe1846f185cfbda554046cce'],
        model: 'abb71ca5fe1846f185cfbda554046cce',
        vendor: 'LivingWise',
        description: 'Dimmer',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: [],
        toZigbee: generic.light_onoff_brightness().toZigbee,
configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
},

dev-abo, 2019-02-13:
now i'm getting

` zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb)' with result 'Error: Timed out after 30000 ms' +30s`

Koenkk, 2019-02-13:
Can you post your `database.db`?

dev-abo, 2019-02-13:
Sure [Database.db](https://hastebin.com/okocadivux.json)

Koenkk, 2019-02-13:
While looking at your wireshark capture, I don't think this device needs a `configure` (sorry). Somehow the messages don't end up in zigbee2mqtt.

Could you replace the contents of `node_modules/zigbee-shepherd/lib/component/af.js` with the one from https://raw.githubusercontent.com/Koenkk/zigbee-shepherd/fix/lib/components/af.js (this has improved logging). Trigger some status updates and post the log here.

dev-abo, 2019-02-14:
Unfortunately, still no incoming messages. I can see additional debug info from my other devices (xiaomi), but nothing from the dimmer. 

I uploaded some outgoing messages (mqtt => zigbee) with additional info here [outgoing log](https://hastebin.com/xusiqaqulu.coffeescript), maybe this can give something.

Koenkk, 2019-02-14:
@dev-abo somehow that links give me an empty page.

I expect to receive messages when you manually toggle the switch, is this the case?

dev-abo, 2019-02-14:
Weird...the link is broken for me as well. I'll upload it again once I get home.

Yes, I'm trying to get status change when I physically press buttons (on/off/brightness).

dev-abo, 2019-02-14:
this link should work [out.log](https://raw.githubusercontent.com/dev-abo/sharedfiles/master/out.log)

Koenkk, 2019-02-15:
@dev-abo in the log you also toggle the switch from zigbee2mqtt, can you provide a log where you only toggle the switch manually (physically), otherwise I cannot distinguish the messages.

dev-abo, 2019-02-15:
@Koenkk that's correct, this is the issue i'm trying to fix. The attached log has only events triggered from mqtt, but i toggled the switch manually as well, but as you can see, none of the messages from switch appeared in the log. It looks like they've been skipped/filtered out for some reason.

Koenkk, 2019-02-15:
@dev-abo that means we have to dive deeper here.

When running with `DEBUG=* npm start`; do you see any direct output when you manually toggle the switch?

dev-abo, 2019-02-15:
@Koenkk i've tried `DEBUG=* npm start` as well. I do not think there was anything from that switch. However, i clearly saw output from my other devices. 
Is there something i can search for in the DEBUG output which can help?

Koenkk, 2019-02-15:
@dev-abo if you don't see anything directly, there is nothing to look into. I see that the device send it's report to coordinator endpoint `10`. let's see if we can change that.

Can you try with:

```js
{
        zigbeeModel: ['abb71ca5fe1846f185cfbda554046cce'],
        model: 'abb71ca5fe1846f185cfbda554046cce',
        vendor: 'LivingWise',
        description: 'Dimmer',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: [],
        toZigbee: generic.light_onoff_brightness().toZigbee,
       configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
            ];

            execute(device, actions, callback);
        },
},
```

dev-abo, 2019-02-16:
@Koenkk still no incoming messages.

The only message related to the dimmer is that it's configured successfully:

` zigbee2mqtt:info 2019-2-15 18:52:23 Succesfully configured zbDimmer1 (0x00124b001a4ddba6)`

Koenkk, 2019-02-16:
Can you sniff the traffic again?

dev-abo, 2019-02-16:
Here is a log from Wireshark [zb-dimmer.pcapng](https://github.com/dev-abo/sharedfiles/raw/master/zb-dimmer.pcapng).

I believe, the device i'm trying to connect has Source = 0x026e.
so, i can see
**On event**:
[image]
**Off event**:
[image]

and **brightness report**:
[image]

Koenkk, 2019-02-17:
can you also expand the network and application support layer data?

dev-abo, 2019-02-17:
- **On:**

[image]

- **Off:**
[image]

- **Brightness:**
[image]

Koenkk, 2019-02-17:
Thanks, can you try with the following CC2531 firmware: https://drive.google.com/open?id=10341IaglXLet65mZeEGd4jEI6P3_oAJG ?

dev-abo, 2019-02-17:
after re-pairing the dimmer i can see the warning message "No converter available for...", full log is here [log](https://pastebin.com/raw/27y3Y3HT). 

All my other devices stopped report their status, but i can still control them - should i re-pair them also?

Koenkk, 2019-02-17:
looks good! what do you mean with reporting their status? can you provide the log?

dev-abo, 2019-02-17:
There is nothing in the log. I just had to pair all my devices again. Now it works.

Koenkk, 2019-02-17:
Great, could you make a pr for the device?

dev-abo, 2019-02-17:
sure, i will. 
Let me finish with converters and adding some comments first, and then i'll make a pr.

closed 2019-02-18 by dev-abo

tradenet, 2019-03-16:
> sure, i will.
> Let me finish with converters and adding some comments first, and then i'll make a pr.

Will you be adding  more LivingWise (printcables) Orvibo devices? Thx.

dev-abo, 2019-03-16:
I do not have any other devices at the moment. In case I get one, I'll add it.

tradenet, 2019-03-16:
> I do not have any other orvibo devices at the moment. In case I get one, I'll add it.

OK thanks for adding the switch BTW.

tradenet, 2019-03-24:
> I do not have any other orvibo devices at the moment. In case I get one, I'll add it.

@dev-abo I've added the LivingWise door/window sensor. Works great.