Koenkk/zigbee2mqtt #1072: Fork documentation into a new repository

dgomes, 2019-02-12:
Looking at the commit history, most of it is related to documentation.

I believe the documentation has grown into a stage that it should live in it's own repo, else it's very difficult to track actual changes in zigbee2mqtt.

Bruceforce, 2019-02-12:
Or use a wiki in the main repo maybe?

ryanbeaton, 2019-02-12:
@Bruceforce It used to be in the wiki.  It was moved into the repo in November

dgomes, 2019-02-12:
it just needs to go into it's own repo... "zigbee2mqtt-doc"

Koenkk, 2019-02-12:
@dgomes completly agree, as this changes the url, ive registered http://www.zigbee2mqtt.io . Will point this to the zigbee2mqtt-docs repo.

Koenkk, 2019-02-12:
Done, docs are now in https://github.com/Koenkk/zigbee2mqtt.io, available under https://www.zigbee2mqtt.io/

closed 2019-02-12 by Koenkk