Koenkk/zigbee2mqtt #1074: New chipset from TI?

erazor666, 2019-02-12:
http://www.ti.com/product/cc1352p?HQS=epd-mcu-simp-cc1352p_zigbee-dspbs-pf-epd_wcsimpsub1ms_all_null_fp-wwe&DCM=yes&dclid=CNeY7_f7tOACFdjNsgodUXsDGw

I got this ad in my webbrowser probably some cookie from downloading tools of ti's website. I don't have much clue about reading datasheets but maybe someone has. If this is an improvement over the ones we are currently using.

Koenkk, 2019-02-12:
Looks interesting! Once it reaches aliexpress for cheap prices this will get interesting for us.

closed 2019-02-12 by Koenkk

odx, 2020-01-02:
Isn't this ebyte board what we are looking for?
https://www.aliexpress.com/item/4000321518715.html

The documentation should contain the information required to flash it with a JTAG adapter ("XDS100 downloader").
As there is no USB adapter onboard there is a serial adapter required to communicate. This should work according to the CC2530, correct?

How can we tell which pins to use for the serial RX/TX?