Koenkk/zigbee2mqtt #1075: Is it possible to update the MQTT topics and message structure?

antti365, 2019-02-12:
1. I make the control panel in the node-red dashboard and the zigbee2mqtt/bridge/log topic. In my opinion, I’m missing several options, is it possible to add them?
Something like this ...
- 1. device_renamed
- 2. device_binding
- 3. device_add_group, device_remove_group, device_remove_all_group

2. Also, if possible, I would like to see in the output of the topic zigbee2mqtt/bridge/log "devices": a list of all devices with a battery value for all devices that transmit this parameter.

3. And finally, at the start of the bridge, he could report in a certain topic about the state permit_join and log_level.

I would be very grateful for the help in adding these new options.

Koenkk, 2019-02-13:
1. Done
2. This can already be retrieved, subscribe to `zigbee2mqtt/+` and parse the `battery` out of the payloads
3. Done
4. Same as 2

antti365, 2019-02-13:
@Koenkk On the first three points everything is fine, thank you very much! 
On the fourth point, I probably incorrectly explained - is it possible to make it so that they can be changed via MCT, as well as, for example, friendly_name, on the fly?

antti365, 2019-02-17:
@Koenkk Are we closing this issue or can something be done on the last post?

Koenkk, 2019-02-17:
I will look into the last, should be doable.

Koenkk, 2019-02-18:
Added `last_seen`, adding more should be easy, just apply the same changes as in https://github.com/Koenkk/zigbee2mqtt/commit/26bed2d9983d6622e35f2a917212590455abf574.

antti365, 2019-02-18:
@Koenkk Great, thanks, I figured out where to add this. I think this can be closed Issue.

antti365, 2019-02-23:
@Koenkk Hurried up, changing last_seen via mqtt works correctly with the Ikea switch, but the disable command is not perceived as a dimmer.
Feb 23 16:15:45 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:15:45 PM Set last_seen to epoch
Feb 23 16:16:14 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:16:14 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":74,"linkquality":63,"elapsed":151492,"click":"off","last_seen":1550931374726}'
Feb 23 16:16:20 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:16:20 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Dimmer', payload '{"battery":74,"linkquality":0,"last_seen":1550931380967,"brightness":0,"elapsed":47858}'

Feb 23 16:16:48 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:16:48 PM Set last_seen to ISO_8601
Feb 23 16:16:51 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:16:51 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":74,"linkquality":63,"elapsed":36541,"click":"off","last_seen":"2019-02-23T14:16:51.267Z"}'
Feb 23 16:16:54 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:16:54 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Dimmer', payload '{"battery":74,"linkquality":31,"last_seen":"2019-02-23T14:16:54.081Z","brightness":0,"elapsed":17280}'

Feb 23 16:17:01 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:17:01 PM Set last_seen to disable
Feb 23 16:17:05 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:17:05 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":74,"linkquality":57,"elapsed":14471,"click":"off"}'
Feb 23 16:17:11 raspberrypi3 npm[32239]:   zigbee2mqtt:info 2/23/2019, 4:17:11 PM MQTT publish: topic 'zigbee2mqtt/Ikea_Dimmer', payload '{"battery":74,"linkquality":31,"last_seen":"2019-02-23T14:16:54.081Z","brightness":0,"elapsed":17110}'

Koenkk, 2019-02-23:
Can you check with the latest dev branch?

antti365, 2019-02-24:
@Koenkk Thank you very much, now everything works as it should!

antti365, 2019-03-07:
@Koenkk I tried to change the above files for the `elapsed` option. It did not work, in the logs the change record appears, in the configuration file the option appears in quotes for some reason (elapsed: ‘false’), and the sensors now send the same value all the time ...
Your help is needed. :)

Koenkk, 2019-03-07:
@antti365 please make a PR with the changes you made (doesn't have to be working), we will continue there.

closed 2019-03-07 by Koenkk