Koenkk/zigbee2mqtt #1076: Xiaomi Aqara Temperature sensor does not show battery level

nanosonde, 2019-02-12:
After setting up zigbee2mqtt on a complete new Raspberry Pi3 with another CC2530 device as coordinator, I have an issue with the above sensors. The former RPI is offline and also all routers.

The sensors used to show the battery voltage for example and also send the current values if I pressed the button shortly.
Now they do not show it anymore and also when pressing the button shortly there is no messages in the log showing the current value.

Instead I see the following on each button press:
```
 zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1s
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +8ms
  zigbee2mqtt:debug 2/12/2019, 12:36:51 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.weather"}}' of device 'lumi.weather' (0x00158d0002cxxxxx)
```

Any idea why this happens?
I have reduced the setup to only one sensor without router.
Did the pairing not happen properly? Do I have to factory reset the sensor somehow?
It happens with all four sensors that I own.

Koenkk, 2019-02-12:
Is this reported after a day?

nanosonde, 2019-02-12:
No. This is reportes on each short button press on the sensor device.

On the old setup (Rpi3+Cc2530) it reported all values with each short button press.
On the old setup the sensors were initially paired with an older version (<1.1.1) of zigbee2mqtt and the previous coordinator FW version (last from 2018)though. It was working fine even after upgrading to latest 1.1.1 version(FW was NOT upgraded) on the old setup.

nanosonde, 2019-02-12:
Just tried dev branch with latest FW on both CC2530 and CC2531. Same issue.
Always only the above message and no battery level.

WebSpider, 2019-02-12:
It reports battery level after a few hours, it needs the time to get the value.

nanosonde, 2019-02-12:
@WebSpider 
Ok, I was not aware of this. But what about the voltage? It should be available immediately, shouldn't it?

nanosonde, 2019-02-12:
What is also strange for alle sensors:
I do not see any message "New device......" when pairing.
I only see this a couple of times: "Device with modelID 'undefined' is not supported."
But then a few seconds later, it see first reports of temperature, humidity and pressure.
When I restart zigbee2mqtt I see that there is a new Xiaomi temp sensor paired.
Did the pairing not properly succeed?

WebSpider, 2019-02-12:
@nanosonde it is my experience it takes a bit of time. Be patient, the devices can be a bit slow :)

nanosonde, 2019-02-12:
@WebSpider Ok, I will give it some time. ;)
However, besides the not shown battery level I was used to see the current values on each short button press. This also does not happen anymore. Do you think that this is also related to "device needs some time"?

WebSpider, 2019-02-12:
I have seen it stop reporting battery values, but never temp/hum.

nanosonde, 2019-02-12:
@WebSpider I just meant the reports sent on short button presses on the sensor device.
You are right! After some time it starts sending the battery voltage and the level automatically together with the other values.

AND the best: now it also sends the sensor values on each short button press again!

```
 zigbee2mqtt:debug 2/12/2019, 10:39:56 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.weather"}}' of device 'lumi.weather' (0x00158d0002c1xxxx)
  zigbee2mqtt:debug 2/12/2019, 10:39:57 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":2965,"4":17320,"5":24,"6":[0,0],"10":0,"100":2359,"101":4782,"102":102916}}}' of device 'lumi.weather' (0x00158d0002c1xxxx)
  zigbee2mqtt:info 2/12/2019, 10:39:57 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c1xxxx', payload '{"temperature":23.59,"linkquality":78,"humidity":47.82,"pressure":1029.16,"battery":31,"voltage":2965}'
```

Closing issue. One just has to be patient!

closed 2019-02-12 by nanosonde

ryanbeaton, 2019-02-12:
Just an FYI because of a quirk in Home Assistant all values must be sent in a MQTT message or it wipes out the values it did not receive for the device. For this reason Zigbee2MQTT caches values and combines them into one message.

So once the battery value has been received once it can then add that to the message on all subsequent messages. That is why now you get the battery on every button push.

nanosonde, 2019-02-13:
@ryanbeaton Ok, thanks for this additional hint concerning the caching.

However, once the device itself has the battery voltage available it also starts to send it along with the other values on each short button push.

See the logs:
```
zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +10ms
  zigbee2mqtt:debug 2/13/2019, 8:41:36 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":2975,"4":17320,"5":24,"6":[0,524288],"10":0,"100":2344,"101":6204,"102":103007}}}' of device 'lumi.weather' (0x00158d0002c1xxxx)
  zigbee2mqtt:info 2/13/2019, 8:41:36 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c1xxxx', payload '{"battery":37,"voltage":2975,"linkquality":23}'
  zigbee2mqtt:info 2/13/2019, 8:41:36 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c1xxxx', payload '{"temperature":23.44,"humidity":62.04,"pressure":1030.07,"linkquality":23}'
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +17ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  zigbee2mqtt:debug 2/13/2019, 8:41:36 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,2975,null,null,17320,24,[0,524288],null,null,null,0]}}' of device 'lumi.weather' (0x00158d0002c1xxxx)
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +8ms
  zigbee2mqtt:debug 2/13/2019, 8:41:36 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.weather"}}' of device 'lumi.weather' (0x00158d0002c1xxxx)
```